ct-Open-Source/tuya-convert #176: IP address of device

optilude, 2019-04-16:
I'm trying to adapt this set of scripts to work in an OpenWRT environment (because that's the device I have to do this with). I've got most of the moving parts in place (see https://github.com/optilude/tuya-convert/blob/master/INSTRUCTIONS.md)

However, there is one thing I don't really understand: the script assumes the device ends up on `10.42.42.42` (i.e. in the all the `curl` commands).

How is that? I can't find any references to that IP address outside of the docs. It's outside the range of the DHCP server set up by `dnsmasq` in the generated configuration.

Grateful for any insights.

kueblc, 2019-04-16:
This is defined [here](https://github.com/vtrust-de/esp8266-ota-flash-convert/blob/master/src/esp8266-ota-flash-convert.ino#L22). This decision is fairly arbitrary, though I am working to generalize this to take any IP, so that multiple devices can be flashed at the same time.

optilude, 2019-04-16:
Thank you for clarifying. Also helps confirm this would not work on a different subnet.

Two more questions if I may:

1) There are two invocations of `ip route add 255.255.255.255 dev wlan0`, one in start_flash.sh and one in setup_ap.sh. I assume only one is needed, but I’d also like to understand why it’s needed at all.

2) In setup_ap.sh there are iptables commands that (if I’ve understood correctly) flush tables and set up NAT from wlan0 to eth0. Assuming the DNS “capture” of the Tuya domains via dnsmasq is set up, is there any reason this wouldn’t work with a bridged interface? Or does it have to use NAT from an isolated network?

Martin

closed 2019-04-21 by optilude