ct-Open-Source/tuya-convert #177: Flash not working for FCMILA bulb

X-Light, 2019-04-17:
I am trying to flash my FCMILA bulbs unsuccessfully. Not sure about the exact model, but took the following pictures: [image]
[image]
[image]

From the logs:

**smarthack-smartconfig.log**
```
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```

**smarthack-wifi.log**
```
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:01:08:63 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 RADIUS: starting accounting session BBA786E3ECE880A0
wlan0: STA 84:0d:8e:8e:1f:24 WPA: pairwise key handshake completed (RSN)
wlan0: STA 4c:74:bf:c2:aa:08 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 4c:74:bf:c2:aa:08
wlan0: STA 4c:74:bf:c2:aa:08 RADIUS: starting accounting session 68705B1680E8F741
wlan0: STA 4c:74:bf:c2:aa:08 WPA: pairwise key handshake completed (RSN)
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 RADIUS: starting accounting session 31E26D079F889165
wlan0: STA 84:0d:8e:8e:1f:24 WPA: pairwise key handshake completed (RSN)
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 RADIUS: starting accounting session E412E84773565014
wlan0: STA 84:0d:8e:8e:1f:24 WPA: pairwise key handshake completed (RSN)
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 84:0d:8e:8e:1f:24
wlan0: AP-STA-DISCONNECTED 4c:74:bf:c2:aa:08
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
```

**smarthack-web.log**
```
Listening on port 80

URI:/gw.json?a=s.gw.token.get&gwId=03765723840d8e8e1f24&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":2,"source":"ap","path":"broadcast"}}&t=8&v=3.0&sign=1100a6fb9963e6b7244d7446755d6e5e
[... 130 lines cut ...]
1555522787: Received PINGREQ from 03765723840d8e8e1f24
1555522787: Sending PINGRESP to 03765723840d8e8e1f24
1555522831: Received PINGREQ from 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555522831: Sending PINGRESP to 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555522847: Received PINGREQ from 03765723840d8e8e1f24
1555522847: Sending PINGRESP to 03765723840d8e8e1f24
1555522891: Received PINGREQ from 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555522891: Sending PINGRESP to 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555522907: Received PINGREQ from 03765723840d8e8e1f24
1555522907: Sending PINGRESP to 03765723840d8e8e1f24
1555522951: Received PINGREQ from 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555522951: Sending PINGRESP to 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555522967: Received PINGREQ from 03765723840d8e8e1f24
1555522967: Sending PINGRESP to 03765723840d8e8e1f24
1555523012: Received PINGREQ from 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555523012: Sending PINGRESP to 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555523045: Received PINGREQ from 03765723840d8e8e1f24
1555523045: Sending PINGRESP to 03765723840d8e8e1f24
1555523072: Received PINGREQ from 3219f414-bcd1-4fb6-bc37-8898058c3f6e
1555523072: Sending PINGRESP to 3219f414-bcd1-4fb6-bc37-8898058c3f6e
```

I have no problem configuring the device through the tuya app. After several unsuccessful attempts, I came to a video, stating that there might be some new firmware from tyua preventing the updates. Could this be the problem, or my devices are not compatible?

Is there other way to flash these?

X-Light, 2019-04-21:
hello?

ruievmonteiro, 2019-04-24:
I have the same bulb, but can't even go that far. Somehow the bulb IP Address gets ip address 10.42.42.31 and my device 10.42.42.13. Can someone help me to debug this?

From the logs:

**smarthack-smartconfig.log**

```
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```

**smarthack-wifi.log**

```
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:01:08:63 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 RADIUS: starting accounting session BBA786E3ECE880A0
wlan0: STA 84:0d:8e:8e:1f:24 WPA: pairwise key handshake completed (RSN)
wlan0: STA 4c:74:bf:c2:aa:08 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 4c:74:bf:c2:aa:08
wlan0: STA 4c:74:bf:c2:aa:08 RADIUS: starting accounting session 68705B1680E8F741
wlan0: STA 4c:74:bf:c2:aa:08 WPA: pairwise key handshake completed (RSN)
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 RADIUS: starting accounting session 31E26D079F889165
wlan0: STA 84:0d:8e:8e:1f:24 WPA: pairwise key handshake completed (RSN)
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:8e:1f:24 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:8e:1f:24
wlan0: STA 84:0d:8e:8e:1f:24 RADIUS: starting accounting session E412E84773565014
wlan0: STA 84:0d:8e:8e:1f:24 WPA: pairwise key handshake completed (RSN)
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 84:0d:8e:8e:1f:24
wlan0: AP-STA-DISCONNECTED 4c:74:bf:c2:aa:08
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
```

**smarthack-web.log**

```
Listening on port 80
^[[32m[I 190424 00:29:45 web:2246]^[[m^O 200 GET / (10.42.42.13) 6.75ms
```

**smarthack-mqtt.log**

```
1556065776: mosquitto version 1.5.7 starting
1556065776: Using default config.
1556065776: Opening ipv4 listen socket on port 1883.
1556065776: Opening ipv6 listen socket on port 1883.
```

kueblc, 2019-04-25:
Unfortunately, sounds like you are running a newer firmware where this has been patched. At present, we have no guarantee as to when or if we will have a fix.

Are you able to flash other devices?

See if the device is broadcasting anything on UDP port 6666 or 6667 ie, by running `nc -u -l -k -p 6667` while connected to the same network as the device. You may need to register the device to get data from this.

X-Light, 2019-04-30:
Today I tried to flash them once again and it was successful. The only difference was that I did not connected a device to the vtrust-flash network, which was needed according to several video tutorials I was following. My bulbs got flashed without issues and I am running tasmota firmware on both of them now. 
Thank you for your great work!

closed 2019-04-30 by X-Light