ct-Open-Source/tuya-convert #179: Teckin SP22 - bricked?

gmcbrien, 2019-04-19:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

gmcbrien, 2019-04-19:
Hi, I successfully flashed 2 Teckin Sp22 devices today - all fine. I followed the same procedure for number 3. After the curl http://10.42.42.42/flash3 the device restarted but just goes immediately to a flash LED. Removing the plug and putting it back in again, also immediately goes to flashing LED.

Attached are the logs (I cleverly deleted the ones from the actual flashing.... so I could get clean logs for a new attempt.... doh!).

Any advice how to get out of this, or do I simply kiss it goodbye?

Thanks a lot,
Grant

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3098791/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3098792/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3098793/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3098794/smarthack-wifi.log)

DutchmanNL, 2019-05-06:
same issue here with 2 teckin, almost same behavior 👎 

- flashed successfully
- Tasmota AP comes up
- entered wifi credentials
- device keeps flashing, not visible in wifi network and no AP to enter different values

SirRedZ, 2019-05-06:
It is very common to typo the password, and therefor lose access to the device.

Try taking your smartphone, setup a hotspot (no password) with the exact same name you keyed in to Tasmota- With any luck you did not typo the SSID and you will notice the device log-on to your phone's hotspot and from there you can go to the IP address displayed in your phone's hotspot screen (no longer 192.168.4.1) and recover the device. I strongly urge you once in the device, after reboot, goto Console and type Wificonfig 2. This will allow the device to setup the 'sonoff-xxxx' after 180 seconds if it fails to connect to your wifi. This could be considered a security threat, but it you can negate that by using a web password later on.

Note: After wificonfig 2 - the sonoff will reboot and setup as sonoff-xxxx --- Do nothing, wait 5 minutes -- it will jump back on your network if configured correctly.

Good luck!

[-=SrZ=-]

digiblur, 2019-05-06:
I wish the Tuya Convert script would pull down a bin like the one that I compiled of 6.5.0 with WiFi Manager as default, we would stop seeing all these threads like this.

SirRedZ, 2019-05-07:
Agreed- Have you contacted Kueblc and offered to include your compiled bin/update when needed?

It sure would make a difference - assuming I am correct with my response above about typo's/wifi for this situation.

[-=SrZ=-]

gmcbrien, 2019-05-07:
Thanks for responding SirRedZ - I also had your idea that maybe I fat-fingered the SSID / password. The question is, I would have no way of knowing if I did a typo on the SSID or the password, or even if WiFi is running at all on the device.
I also like digiblur's idea - sounds like this should be the standard image that's bundled with Tuya Convert :)

digiblur, 2019-05-08:
@SirRedZ my bins with generic as the device default and wifi manager as default are always in this folder for them to pull I won't link the direct bin right now as that could change in the future as I add versions as Tasmota development progresses.  @kueblc

probonopd, 2019-05-17:
@digiblur thanks for providing those.

After flashing your 2.5.0 to thee factory-new SP22, the devices boot into Tasmota, open the AP, but quickly thereafter they reboot. This makes it next to impossible to scan for WLAN and enter credentials. I suspect that either something is crashing or some watchdog timer is resetting the device. Is this a known issue?

Also, when I select "Teckin" in Tasmota, then I can switch on/off the relais, but the blue LED always blinks instead of going solid blue when the relais is on and off when the relais is off. I think it never gets red. Is this also know and is there a workaround? (BlitzWolf does NOT work for my batch of Teckin SP22 devices, btw.)

SirRedZ, 2019-05-17:
@probonopd  Looks like the Teckin SP22 is not yet in the Tasmota Template Database.

You can try and load this template ```https://blakadder.github.io/templates/teckin-sp23.html``` and see how that goes.

The fast reboots is definitely an issue, and I am unsure why that is happening. Perhaps the GPIO configs are wrong and that would do it. Login and quickly go to CONSOLE > and type ```reset 5``` and ```enter``` and hopfully that will erase the bad info stored and reboot. Re-setup as brand new, and try the new template I have provided, or create your own by choosing Module 'Generic Type 18' and start figuring out what GPIO is for what.

You might find this helpful.

```https://github.com/arendst/Sonoff-Tasmota/issues/3950```

[-=SrZ=-]

probonopd, 2019-05-17:
Do I risk bricking when playing around with templates and GPIOs? I don't want to pry open my devices.

probonopd, 2019-05-18:
Looks like __Teckin (52)__ is working for me. Thank you.

SirRedZ, 2019-05-18:
Thanks for the update!

[-=SrZ=-]

rwit-ormer, 2019-05-26:
@gmcbrien Did you get any luck of getting it working again? I've also a SP22 that looks like its bricked. The only life I see is that the light is flashing blue. It does not respond to the button, whatever combinations I use. The only option I have left is to open it and then flash it.

sebastianklein96, 2019-05-26:
@rwit-ormer did you already flash with Tuya Convert? If you did and it has the Sonoff / Tasmota firmware, pushing the button four times in quick succession should put your device in wifi recovery mode. It'll open the original sonoff-xxxx wifi AP again. Connect to that, navigate to 192.168.4.1 and you can put in new information.

rwit-ormer, 2019-05-26:
It does not respond to the button, whatever combinations I use.

SirRedZ, 2019-05-26:
Only items (typically sonoff designs) that have the button physically tied to GPIO 0 and GND will offer this reboot functionality.
@rwit-ormer did the Tuya-Convert connect and begin the flashing process and seemingly connect, backup and then you selected FLASH3 to flash with sonoff?

Is it possible you converted and are not aware of the new 'sonoff-xxxx' AP that is now available?

If I understand what happened, you failed the backup file, and the device just sat there. If you try running tuya-convert again,-- start the dots.... THEN plug in the module,  no buttons need to be pushed, just let it come online and it should connect and recover. I am not sure if you have previously seen my help document - please try some of the steps listed here.
[tuya-convert-help.pdf](https://github.com/ct-Open-Source/tuya-convert/files/3220287/tuya-convert-help.pdf)
 
[-=SrZ=-]

baldfox, 2019-09-21:
For those of you looking to flash teckin SP22, here's what I did to get mine working with OTA convert 2!
some pointers:
1) to put it in pairing mode, i plugged it in (the light was off) and I held it down for about 6 seconds... the light was blinking blue and off, not red and blue (I tried many times with red and blue flashing but it wouldn't work for some reason).

2) I followed this video - https://www.youtube.com/watch?v=6_IZEa9JcC8
I also manually installed libssl-dev

3) I left my original tuya convert folder and just opened a new folder on the raspberry pi...

4) I cloned this repository into it:
https://github.com/M4dmartig4n/tuya-convert

Follow the instructions in the vid...

5) The lights went out on the teckin SP22 and i thought it had messed up, but it just worked. You need to be patient.

Then I proceeded to flash using the curl commands...
Hopefully this helps, it's not the cleanest instructions, but it worked for me. My bricked SP22 is now in Home Assistant :)

kueblc, 2019-09-30:
Closing due to age. If you still have issues, please try with the latest `tuya-convert` or the `new-api` branch at #279, opening a new issue if you encounter any problems.

closed 2019-09-30 by kueblc