ct-Open-Source/tuya-convert #180: I can not log into the Wifi ssid: vtrust-flash

cdrom1028, 2019-04-19:
when I get to this screen:
###########################################################
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   The wpa-password is flashmeifyoucan
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue
###########################################################

I do see the ssid: vtrust-flash but I cannot authenticate. when i type in flashmeifyoucan as the password I get a message stating that password is incorrect. 

when I tail at the log file smarthack-wifi.log I see the following:
###########################################################
root@kali:/etc/wpa_supplicant/tuya-convert/scripts# tail smarthack-wifi.log 
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Failed to remove interface (ifidx=4)
###########################################################
this makes no sense because i see the ssid on iOS, Android and Win 10. 

Hardware : Raspberry Pi 3B+
OS : kali-linux-2018.4a-rpi3-nexmon-64

cdrom1028, 2019-04-20:
Changed approach and distro. Worked with no more issues!

closed 2019-04-20 by cdrom1028

wesmininger, 2019-07-05:
> Changed approach and distro. Worked with no more issues!

@cdrom1028 

????

Its great you came back to post you solved the problem. But please more info? You might well never have posted again and had the same effect!

I am having the same issue exactly. Ive tried w raspbian and the Kali linux image from this readme.
What did you do to get it to connect to the vtrust-flash wifi?