ct-Open-Source/tuya-convert #181: Is it possible to stock image after successful flash3?

optilude, 2019-04-21:
I've successfully flashed a two-outlet wall socket (https://www.amazon.co.uk/gp/product/B07GBQJSHK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1).

However, I can't find a Tasmota template that works for it. I wanted to revert to the stock firmware for now.

Is it possible to do so? The documentation talks about `curl http://10.42.42.42/undo` but that presumably only works before `flash3` when the device is actually at that IP address.

I have the backup firmware from `curl -JO http://10.42.42.42/backup`. Is it viable to use this? How?

AS137430, 2019-04-22:
Why not work out the template? https://github.com/ct-Open-Source/tuya-convert/issues/49#issuecomment-458635136

optilude, 2019-04-22:
Hi,

That link is really helpful! Would be great to see that in the docs somewhere.

It would also be really good to understand the ability to “undo”. When I first read the docs I assumed “/undo” would work forever. I am now thinking it’s not possible after “flash3” but I’m not sure.

As of right now I have figured out enough to get the device to work with tasmota. It was crashing repeatedly (rebooting into basic AP mode) but seems to be stable now.

Thanks!
Martin

Martin Aspeli

Why not work out the template? #49 (comment)<https://github.com/ct-Open-Source/tuya-convert/issues/49#issuecomment-458635136>

AS137430, 2019-04-22:
@optilude: The wiki is editable so feel free to add the entry for VIGICA 2.1A USB Charger. Tasmota template also allows new devices' submission so please contribute back! You can also close this issue

optilude, 2019-04-22:
Done!

closed 2019-04-22 by optilude