ct-Open-Source/tuya-convert #182: Issue with flashed plug - Wifi showing Vtrust-recovery?

Springs87, 2019-04-22:
Hi all,

I tried flashing a plug some time ago and for some reason it didn't flash completely.

I tried to sort it today and when i start the flash script it will pick up the device without me trying anything and try and pull the current firmware for a backup but it fails and then gives me the options to flash / restore etc.

I've tried flashing a file to it but it states that the device is not available on port 80.

as well i've noticed that there is a vtrust-recovery SSID showing up.

Is there anyway to recover this device?

kueblc, 2019-04-22:
Make sure you can access the `vtrust-flash` AP, then reconnect your device. You should then be able to `curl http://10.42.42.42/flash3` or whatever you'd like to do.

SirRedZ, 2019-04-23:
What model of plug is causing you issues? You can try some of the solutions I have prepared in this document, and let us know if any of them help. Note the document mentions ProxMox but most information is valid for normal tuya-convert processes.

[-=SrZ=-]
[tuya-convert-help.pdf](https://github.com/ct-Open-Source/tuya-convert/files/3105909/tuya-convert-help.pdf)

Springs87, 2019-04-23:
This plug thats causing the issue is an AISIRER plug. I've got another one of the same plug that i flashed yesterday without an issue.

Since i originally tried to flash this plug i've changed the device that i used to sit on the vtrust wifi. It's gone from a Android tablet to a Windows laptop. I've also flashed 2 other plugs yesterday with no issues and managed to recover one as well.

I've tried this plug again and i'm still getting timeout errors on screen. It is seeing that the device is connected.

"curl: (7) Failed to connect to 10.42.42.42 port 80: Connection timed out"

Does anyone know what the recovery SSID is and what it can be used for?

SirRedZ, 2019-04-23:
My experience is the plug has been placed in to 'Stage 1' and is now waiting for firmware to be pushed to it. If I am not mistaken, you DO NOT NEED a device to connect to Vtrust at this point, the hack has already been performed - you might find success rapidly proceeding through the steps (as if you had connected to vtrust) the device will come online, backup, and allow flashing. You will have to delete the failed backup file matching this device (it uses the chipset ID) - best to rename it, and keep it. 

Did the backup 'freeze' on you? Did you try plugging the device in to a different outlet -- perhaps a picky power supply?

[-=SrZ=-]

Springs87, 2019-04-23:
I've tried 2 different outlets and get the same result.

It did freeze when i did the initial backup some time ago and since then i've re-installed the OS so i don't have the original backup file.

SirRedZ, 2019-04-24:
I had that exact issue with SP20 plugs, some were picky -- I was able to recover all of them following the tips I have provided in the help document. I was able to use a UPS device to recover a stubborn unit, I blamed dirty power/bad powerbars+filtering.

Springs87, 2019-04-30:
I've tried everything in the help document with no change.

I have moved it to a different power strip and still the same. 

The only difference between what i'm doing and the document is that i'm doing it through a RPI3 and not Proxmox.

I'll give it another go on a different power strip but otherwise i'll have to pull it apart and see if i can easily get to the pins

SirRedZ, 2019-04-30:
I am sorry that it didn't help!

The key thing I keep hearing about is the device being used to connect to Vtrust really does matter. Some people have reported 2 devices not working (that might have worked previously) and trying a 3rd then Boom. they get it. Not sure why this happens, but I have heard about it numerous times.

Good luck!

[-=SrZ=-]

kueblc, 2019-09-28:
Closing due to inactivity, feel free to open a new issue if you have any concerns.

closed 2019-09-28 by kueblc