ct-Open-Source/tuya-convert #183: Failed to create interface mon.wlan0: -95 (Operation not supported)

ezcell, 2019-04-23:
Hi,
Using a brand new RPI3 Rasbian lite setup, I cant seem to get the AP running stable.
I get an endless amount of dots waiting for devices to connect, and this in the wifi log: 
/etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver

Unfortunately, linux isnt my strong suit, so I don't know how to troubleshoot- it's using the built in wifi from the RPI 3B so I doubt it is a driver issue. 
Any suggestions?

kueblc, 2019-04-23:
Please post the full log, that one error is benign and doesn't tell me anything

demian85, 2019-04-24:
I'm having the same issue, please check my comment: https://github.com/ct-Open-Source/tuya-convert/issues/172#issuecomment-486006685

demian85, 2019-04-24:
Can someone please clarify if by using a Raspberry Pi wlan0 can act as an AP and keep an SSH connection to my notebook at the same time? Should I use eth0 instead and wlan0 only as an AP?

Thanks.

SirRedZ, 2019-04-24:
the SSH connection will work if you are connected to ethernet for this connection. the Wifi adapter must be available to have it's state changed, and be brought up into AP mode for the tuya-convert process. You didn't specify your PI version, anything without EMBEDDED Wifi may need some tweeks to get working. Try running IFCONFIG and IWCONFIG (if installed) to get the name of the interface. If you need to install, ```apt install wicd``` will likely help with that.

[-=SrZ=-]

demian85, 2019-04-24:
I have a raspberry pi 3B+. How can I make the wlan0 device available?
Should I disconnect from wifi or disable it somehow?

SirRedZ, 2019-04-24:
For tuya-convert, you MUST connect to Internet via Ethernet (not optional). The wifi adapter is used to create hotspot. If you were previously connecting via Wifi, that is the problem. Just plug in to your router (modify wpa supplicant as needed to use ethernet) and you can SSH in and it should resolve the issues.

[-=SrZ=-]

ezcell, 2019-04-24:
OK I managed to get it running by first killing off wpa supplicant. 
sudo wpa_cli terminate
I managed to convert a bunch of tuya devices, a few did not work but I'm assuming it is because they might be the new firmware that others on this board have mentioned (https). Looks like I'll have to break these open and try a serial connection.
Than you SOO MUCH for all  your hard work!!
I'll keep this thread open for others to comment if this worked for them or not, might be something to consider adding to the bash script if it is just that simple.

SirRedZ, 2019-04-24:
That's great news! Here is a document of trouble-shooting steps to try for difficult modules. (see attachment)
[tuya-convert-help.pdf](https://github.com/ct-Open-Source/tuya-convert/files/3113690/tuya-convert-help.pdf)

Lets see if this helps you.

[-=SrZ=-]

kueblc, 2019-09-28:
Closing due to inactivity, feel free to open a new issue if you have any concerns.

For those that missed it, support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc