ct-Open-Source/tuya-convert #185: SmartLife 2CH wall switch (LYASI) not flashing  - Firmware 1.0.7 MCU 1.0.7

mzanetti, 2019-04-25:
Trying to flash a LYASI/SmartLife 2CH switch.

It takes the WiFi credentials and connects to the flasher's WiFi, however, doesn't proceed with contacting the webserver.

Firmware Version 1.0.7, MCU version 1.0.7 too.

The device's MAC address is a4:cf:12:a3:5f:a8

```pi@raspberrypi:~/tuya-convert $ cat scripts/smarthack-smartconfig.log 
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.```

```pi@raspberrypi:~/tuya-convert $ cat scripts/smarthack-wifi.log 
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:5f:74:43 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 7c:d9:5c:a7:a8:ad IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 7c:d9:5c:a7:a8:ad
wlan0: STA 7c:d9:5c:a7:a8:ad RADIUS: starting accounting session 2F04A665E809A27A
wlan0: STA 7c:d9:5c:a7:a8:ad WPA: pairwise key handshake completed (RSN)
wlan0: STA 84:c7:ea:30:3a:5c IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:c7:ea:30:3a:5c
wlan0: STA 84:c7:ea:30:3a:5c RADIUS: starting accounting session 95339341EC0DC704
wlan0: STA 84:c7:ea:30:3a:5c WPA: pairwise key handshake completed (RSN)

wlan0: STA a4:cf:12:a3:5f:a8 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED a4:cf:12:a3:5f:a8
wlan0: STA a4:cf:12:a3:5f:a8 RADIUS: starting accounting session 421B8F99DD5E78D8
wlan0: STA a4:cf:12:a3:5f:a8 WPA: pairwise key handshake completed (RSN)
```
```pi@raspberrypi:~/tuya-convert $ cat scripts/smarthack-web.log 
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
```
Note: The webserver is working. It shows the Hello world website.

```pi@raspberrypi:~/tuya-convert $ cat scripts/smarthack-mqtt.log 
^C1556219727: mosquitto version 1.4.10 terminating
1556219730: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1556219730: Using default config.
1556219730: Opening ipv4 listen socket on port 1883.
1556219730: Opening ipv6 listen socket on port 1883.
```
From other issues I read that there is a newer version upcoming which is said to work with firmware 2.0 (or also 1.0.5 for some devices?). Could this be related to the same and there is a high chance it'll start working soon or would you recommend to blow off the dust from the soldering iron?

Also, if you think any other logs would be helpful, I'd be happy to provide those.

kueblc, 2019-04-25:
It seems likely that you have the new firmware (version number is irrelevant as its per product). See if the device is broadcasting anything on UDP port 6666 or 6667 ie, by running `nc -u -l -k -p 6667`.

We are working on this but no guarantees when or if we will be able to release a workaround.

mzanetti, 2019-04-25:
Hmm... actually I cannot seem to catch anything from the device on those ports...

mzanetti, 2019-06-11:
For people ending up here, this might be of interest while the new Firmware can't be OTA'd still:
https://www.hackster.io/michael_zanetti/smartlife-tuya-wifi-light-switch-with-tasmota-and-nymea-09a7a6

thirug010, 2019-07-06:
Please take look at Tuya-VT, it provides the virtually tasmota features on tuya devices using tuyapi. [Virtual Tasmota for Tuya Api Devices](https://github.com/thirug010/Virtual-Tasmota-for-Tuya-api-devices). 

[image]

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc