ct-Open-Source/tuya-convert #186: Another device unable to convert

jziolkowski, 2019-04-27:
Along with Blakadder from Tasmota discord I've bought these:
https://www.aliexpress.com/item/Tuya-Smart-life-app-Control-WiFi-Light-86-EU-Button-Switch-Support-Alexa-Google-Home/32996551323.html?spm=a2g0s.9042311.0.0.5b4b4c4dkzgJNJ

3-gang and 2-gang versions. While I bricked mine 3-gang version during the convert, Blakadder reports that it undergo the process smoothly. However, I've unsuccessfully tried to tuya-convert the 2-gang version. Three different computers, 3 different systems (including live boot MX). The symptoms are similar to other reported: despite enabling the pairing procedure, the device doesn't connect to the vtrust-flash AP and timeouts. I've not connected this device with their application. Since it's trivial to flash it using standard methods, I don't want to try convert anymore, however I'm attaching the original firmware backup for you to analyze.
[fwbackup_ds102_2relays.zip](https://github.com/ct-Open-Source/tuya-convert/files/3123684/fwbackup_ds102_2relays.zip)

kueblc, 2019-04-29:
Thanks for the report and the backup file :+1: 

I am curious if you have any more information on the device that bricked.

jziolkowski, 2019-04-29:
Well it was just my error. It was a while since I used tuya-convert, and forgot that I should watch output in screen instances, instead of waiting for the output on the main terminal. And thinking that the device is doing nothing, I cut the power. No, it wasn't doing nothing :) And it stopped responding. Luckily I managed to flash it via serial and works flawlessly.

jziolkowski, 2019-04-29:
Also I will be getting more of these devices, so I'll let you know if any issues appear.

kueblc, 2019-04-29:
Excellent, thanks! I'll close this now, but feel free to keep adding to your report as needed.

closed 2019-04-29 by kueblc

jamietang68, 2019-06-05:
Hi, 

I followed the steps and previously able to flash the devices few months back (universal remote control).
Same with this just different branding : https://www.amazon.com/YTF-Conditioner-Controller-Compatible-Smartphone/dp/B07H46LJZK

I just bought a few new 1, but it doesnt seems to respond. How do i know if the firmware cant be flashed anymore? the logs below:

pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-mqtt.log
1559774655: Opening ipv4 listen socket on port 1883.
1559774655: Opening ipv6 listen socket on port 1883.
1559775620: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1559775620: Using default config.
1559775620: Opening ipv4 listen socket on port 1883.
1559775620: Opening ipv6 listen socket on port 1883.
1559776009: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1559776009: Using default config.
1559776009: Opening ipv4 listen socket on port 1883.
1559776009: Opening ipv6 listen socket on port 1883.

pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-web.log
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80

pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-wifi.log
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:dd:80:08 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA 04:4b:ed:79:f4:35 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 04:4b:ed:79:f4:35
wlan0: STA 04:4b:ed:79:f4:35 RADIUS: starting accounting session 886C15EAECDC8FAB
wlan0: STA 04:4b:ed:79:f4:35 WPA: pairwise key handshake completed (RSN)
wlan0: STA 04:4b:ed:79:f4:35 WPA: group key handshake completed (RSN)

pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-smartconfig.log
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.