ct-Open-Source/tuya-convert #187: Dnsmasq conflict with systemd-resolved on Lubuntu 18.04

fred9176, 2019-04-29:
Hi,

When launching start_flash.sh on Lubuntu 18.04, I have an error in the log file :
Dnsmasq cannot strat because address is already in use.

And effectively, the port 53 is already used by systemd-resolved.

To solve this problem, I had to edit scripts/setup.asp and add the following line to /etc/dnsmasq.conf, juste after interface=$WLAN 👍
bind-interfaces
listen-address=10.42.42.1

This allowes Dnsmasq to start on a different IP than systemd-resolved.

Fred

kueblc, 2019-04-29:
Great solution, thanks! I'll implement this when I have time, or if you'd like to make a pull request I'll merge it.

ridale, 2019-04-30:
Tested this on Ubuntu 18.04 and flashed a tuya switch without any other modifications.
This also closes #152 
The only other issue on Ubuntu 18.04 is that there is no remove_ap.sh script to remove the unmanaged line from NetworkManager.conf to re-enable wifi. If I get time I'll write a teardown script or update the stop_flash script to return the system to its initial state.

fsaris, 2019-05-05:
Tnx for this fix. 

After adding those 2 lines I was able to flash my NEO Coolcam plugs without any further issues on my Ubuntu 18.04 system.

poerlemans, 2019-05-05:
Some clarification might be helpful.
The file ~/tuya-convert/scripts/setup_ap.sh has to be modified.
After the comment "# Interface to bind to " you will find the description given in the first post.
With a Lubuntu setup I flashed one Avatar smart plug after modification of setup_ap.sh-file.

jcollie, 2019-06-17:
You can also add the following to `/etc/systemd/resolved.conf` to prevent `systemd-resolved` from listening on port 53.

    DNSStubListener=no

Test-Automation-Engineer, 2019-07-17:
> Tested this on Ubuntu 18.04 and flashed a tuya switch without any other modifications.
> This also closes #152 and #157
> The only other issue on Ubuntu 18.04 is that NetworkManager.conf is not being restored by setup-ap.sh to re-enable wifi, i will check to see if the iptable entries are being removed as well.

I'm having the same issue with Ubuntu 18.04. Reboot doesn't restore my wifi-settings. Do you have a solution? Thanks

kueblc, 2019-07-17:
@Test-Automation-Engineer should hopefully be fixed in #225

fsaris, 2019-07-19:
Check of you have a file `etc/NetworkManager/NetworkManager.conf.backup` if not remove/backup `etc/NetworkManager/NetworkManager.conf` and restart network manager again.

That did the trick for me, default there was no conf file for me

Test-Automation-Engineer, 2019-07-20:
That worked after rebooting the networkmanager service. Thank you!

kueblc, 2019-09-26:
See #225

kueblc, 2019-09-28:
If anyone is able to test #225 we can get this fix merged

DonStefano, 2019-10-02:
I was able to flash 4 Teckin SP22 with the above described changes to setup_ap.sh. Also I had to copy back the backup of the NetworkManager configuration file.

kueblc, 2019-10-04:
Closing as resolved by #225. Please provide feedback there. Thanks!

closed 2019-10-04 by kueblc