ct-Open-Source/tuya-convert #188: Start flash - never finds device, just sits with dots running across the screen

joelrader, 2019-04-29:
Trying to flash a Tessan dimmer, but the flash script fails to find the device. I believe the issue is on the Raspberry Pi side however and not on the Tessan device. How I set up my Pi 3b:

1. Fresh install of Raspian Lite
2. apt-get update && apt-get dist-upgrade && apt install git
3. Git clone tuya-convert
4. Install pre-reqs, run flash
5. Connect smartphone to vtrust-flash
6. Put dimmer into pairing mode (red flashing), press enter on flash script
7. Watch dots on screen (no change after 10min)

I find that my smartphone gets constantly disconnected after I resume the flash script. Is that supposed to happen? I've also tried doing a 'sudo killall wpa_supplicant' without any improvement.

Logs:

```
pi@raspberrypi:~/tuya-convert $ tail -f ./scripts/*.log
==> ./scripts/smarthack-mqtt.log <==
^C1556553878: mosquitto version 1.4.10 terminating
1556553881: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1556553881: Using default config.
1556553881: Opening ipv4 listen socket on port 1883.
1556553881: Opening ipv6 listen socket on port 1883.
1556554332: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1556554332: Using default config.
1556554332: Opening ipv4 listen socket on port 1883.
1556554332: Opening ipv6 listen socket on port 1883.
^C1556554573: mosquitto version 1.4.10 terminating
```

```
==> ./scripts/smarthack-smartconfig.log <==
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
^CTraceback (most recent call last):
  File "./smartconfig/main.py", line 23, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 55, in smartconfig
    sock.send_broadcast(broadcast_head)
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 35, in send_broadcast
    sleep(self._gap)
KeyboardInterrupt
```

```
==> ./scripts/smarthack-web.log <==
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
```

```
==> ./scripts/smarthack-wifi.log <==
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 84:0d:8e:a2:fc:08
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:6e:8f:08 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 84:0d:8e:a2:fc:08
wlan0: STA 84:0d:8e:a2:fc:08 RADIUS: starting accounting session E2649651D617E6FA
wlan0: STA 84:0d:8e:a2:fc:08 WPA: pairwise key handshake completed (RSN)
```

```
==> ./scripts/smarthack-web.log <==
Listening on port 80
```

```
==> ./scripts/smarthack-mqtt.log <==
1556554958: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1556554958: Using default config.
1556554958: Opening ipv4 listen socket on port 1883.
1556554958: Opening ipv6 listen socket on port 1883.
```

```
==> ./scripts/smarthack-wifi.log <==
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED d0:04:01:2d:71:14
wlan0: STA d0:04:01:2d:71:14 RADIUS: starting accounting session 3293222915E5AF5C
wlan0: STA d0:04:01:2d:71:14 WPA: pairwise key handshake completed (RSN)
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 84:0d:8e:a2:fc:08
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA 84:0d:8e:a2:fc:08 IEEE 802.11: disassociated
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED d0:04:01:2d:71:14
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
wlan0: STA d0:04:01:2d:71:14 IEEE 802.11: disassociated
```

```
==> ./scripts/smarthack-smartconfig.log <==
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
```

kueblc, 2019-04-29:
>     7\. Watch dots on screen (no change after 10min)

Should take 10 seconds tops

> I find that my smartphone gets constantly disconnected after I resume the flash script. Is that supposed to happen?

Yes, `start_flash` calls `stop_flash` first, which closes the AP.

Thanks for posting your logs. It looks like you have two device mac addresses in your wifi log, is one of them the device in question?

joelrader, 2019-04-29:
Yes, if I pair the dimmer switch with the Smart Life app, it shows up:
84:0d:8e:a2:fc:08

kueblc, 2019-04-29:
So it would seem that at some point, smartconfig succeeded and paired your device to the AP. Since there are no HTTP requests logged, my best guess would be that unfortunately the firmware on this device has been patched.

joelrader, 2019-04-29:
Well that's a bummer. Can you still update the firmware if you solder wires directly to the chip? It's pretty easy to open up. Haven't tried that yet.

kueblc, 2019-04-29:
Believe me, I know! Yes you can certainly still flash the old fashion way.

cyberdemon79, 2019-04-29:
Hi there,
the new firmware includes mbedtls support, the former http requests are now done via https with an psk_aes_128_cbc_sha_256 handshake and server certificate verification. You can see this if you run wireshark and filter for requests on port 443.
I am currently analysing the firmware, trying to find a way of flashing the device without opening it (sacrificed one of them for dumping the firmware using the testpoints and inappropriate soldering skills :)). Will report back if I am successful.
Greetings Cyberdemon

joelrader, 2019-04-29:
cyberdemon79 - That would be excellent if you can get it to work! I'll contribute to your sacrificial device fund :-)

kueblc, 2019-04-29:
@cyberdemon79 We are working on this :) I can be reached by my gmail account of the same name.

I'll close this issue now but you can still comment or reopen if the situation changes.

closed 2019-04-29 by kueblc

joelrader, 2019-05-03:
I was able to successfully flash the dimmer over serial. Had to hold down a button while connecting the serial cable, but went pretty quickly. Spent WAY less time flashing via serial than trying to fiddle with OTA. Another dimmer I bought I couldn't flash via OTA either. I think firmware updates rolled out pretty quick.

meingraham, 2019-05-03:
@joelrader yes... unless the device assembly precludes getting to the serial interface pins. The compactness of many of these, and even those that aren't compact, are sometimes laid out in a manner that makes getting to the contacts unfeasible because the amount of disassembly required.

Unfortunately, it appears that yet another supplier (Tuya now, Itead before) is plugging up their firmware to block an OTA mechanism.