ct-Open-Source/tuya-convert #189: Esicoo Wi-Fi Smart Plug Outlet..no success

jesusvalverde, 2019-04-30:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2019-04-30:
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

jesusvalverde, 2019-04-30:
Esicoo Wi-Fi Smart Plug Outlet [http://www.esicoo.com/index.php?route=product/product&path=60&product_id=94] 
firmware version :1.0.2 [Last Version]

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3129652/smarthack-web.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3129653/smarthack-smartconfig.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3129654/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3129651/smarthack-mqtt.log)

kueblc, 2019-04-30:
Can you confirm that the device is in pairing mode before running the script?

jesusvalverde, 2019-04-30:
Yes ...The pulg makes a Wifi network named SmartLife-XXXX

kueblc, 2019-04-30:
That's AP mode (blinking slowly), you need EZ config mode (blinking fast).

jesusvalverde, 2019-04-30:
Also done that...Gonna done both modes and seend Logs.

kueblc, 2019-04-30:
No need to try AP mode, it won't work

jesusvalverde, 2019-04-30:
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3132388/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3132389/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3132390/smarthack-wifi.log)

No luck...
-Android phone as "men in the middle"
-Pi3b+ [wifi and cable]
-Plug on EZ mode blinking like 2 times/sec

The same config worked with other plug [Diferent brand]

kueblc, 2019-04-30:
I see two devices in your WiFi log, is one of those the device you are trying to flash?

jesusvalverde, 2019-04-30:
ec:fa:bc:22:5e:1a is the one  [Mac number match the app]

Found a model number under YX-WS01 for the Esicoo Wi-Fi Smart Plug Outlet

kueblc, 2019-04-30:
Given the fact that you've been able to flash a different plug, and that the pairing process completed successfully (device connected to wifi) it looks like this device firmware has been patched and unfortunately will not work with tuya-convert.

jesusvalverde, 2019-04-30:
Ok...too bad, but at least we can and the YX-WS01 Esicoo Wi-Fi Smart Plug Outlet as not compatible Over the Air.

closed 2019-04-30 by jesusvalverde

dachshund-digital, 2021-05-08:
I just got some of these Esicoo plugs and they do NOT establish a SmartLife AP, but provide some weird name.  Moreover, although the appears to be ESP based, they can't use the typical tuya smart app, but only seem to work with the 'cloud intelligence' app.  One more thing, the setup with the CI app, is inconsistent.  Where I can get a tuya smart plug to connect 90% on the first try, these new plugs I can't get more than 2 of the 4 to connect at all, and that took many retries.

geoffoxholm, 2021-06-29:
I'm having the same experience as @dachshund-digital and @jesusvalverde - this device seems to have a firmware that does not allow EZ Config mode. Commenting here in hopes of saving others some time.

dachshund-digital, 2021-07-24:
Or... the IC has changed to RealTek not ESP based, which I have also seen.  Either way the PSK issue, or IC variant issues, have broken tuya-convert.

kueblc, 2021-07-28:
I have started work on support for AP config mode, please see #966 and feel free to leave feedback. @dachshund-digital @geoffoxholm

dachshund-digital, 2021-08-14:
This is still specific to the ESP based devices right?  Or will it work on the none ESP based devices?  Will AP mode side setup the PSK issue?