ct-Open-Source/tuya-convert #190: Can't get vtrust-flash wifi to start

ChrisN2501, 2019-05-08:
Hi, 

I'm attempting to flash a couple Teckin SP20 Smart Plugs using Tuya Convert. I'm using a Acer R11 running crouton but have not been successful in getting the vtrust-flash wifi to start. 

I've tried killing the wpa_supplicant before hand per other threads related to this issue to no avail... 

Here are the results of ifconfig 

> arc_wlan0 Link encap:Ethernet  HWaddr 22:e3:af:42:39:7f  
>           inet addr:100.115.92.13  Bcast:100.115.92.15  Mask:255.255.255.252
>           inet6 addr: fe80::f0e5:6ff:fe0d:c776/64 Scope:Link
>           UP BROADCAST MULTICAST  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:0 (0.0 B)  TX bytes:462 (462.0 B)
> 
> arcbr0    Link encap:Ethernet  HWaddr 1e:84:3c:70:68:82  
>           inet addr:100.115.92.1  Bcast:100.115.92.3  Mask:255.255.255.252
>           inet6 addr: fe80::50b5:6fff:feb9:bb3b/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:27519 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:91189 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:4055625 (4.0 MB)  TX bytes:115013840 (115.0 MB)
> 
> lo        Link encap:Local Loopback  
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope:Host
>           UP LOOPBACK RUNNING  MTU:65536  Metric:1
>           RX packets:333 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:333 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:25748 (25.7 KB)  TX bytes:25748 (25.7 KB)
> 
> veth_android Link encap:Ethernet  HWaddr 1e:84:3c:70:68:82  
>           inet6 addr: fe80::1c84:3cff:fe70:6882/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:27519 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:91167 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:4440891 (4.4 MB)  TX bytes:116288222 (116.2 MB)
> 
> veth_wlan0 Link encap:Ethernet  HWaddr 22:e3:af:42:39:7f  
>           UP BROADCAST MULTICAST  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
> 
> wlan0     Link encap:Ethernet  HWaddr a4:02:b9:9e:42:65  
>           inet addr:192.168.1.152  Bcast:192.168.1.255  Mask:255.255.255.0
>           inet6 addr: fe80::a602:b9ff:fe9e:4265/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:1653815 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:382243 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:2225160021 (2.2 GB)  TX bytes:65263781 (65.2 MB)

And here are the logs from running ./start_flash.sh

> ^C1557289520: mosquitto version 1.4.8 terminating
> ^CTraceback (most recent call last):
>   File "./fake-registration-server.py", line 129, in <module>
>     main()
>   File "./fake-registration-server.py", line 125, in main
>     tornado.ioloop.IOLoop.current().start()
>   File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
>     self.asyncio_loop.run_forever()
>   File "/usr/lib/python3.5/asyncio/base_events.py", line 345, in run_forever
>     self._run_once()
>   File "/usr/lib/python3.5/asyncio/base_events.py", line 1276, in _run_once
>     event_list = self._selector.select(timeout)
>   File "/usr/lib/python3.5/selectors.py", line 441, in select
>     fd_event_list = self._epoll.poll(timeout, max_ev)
> KeyboardInterrupt
> Backing up /etc/dnsmasq.conf...
> Writing dnsmasq config file...
> Creating new /etc/dnsmasq.conf...
> Writing hostapd config file...
> Configuring AP interface...
> Applying iptables rules...
> sudo: iptables: command not found
> sudo: iptables: command not found
> sudo: iptables: command not found
> sudo: iptables: command not found
> sudo: iptables: command not found
> sudo: iptables: command not found
> Starting DNSMASQ server...
> Starting AP on wlan0 in screen terminal...
> Configuration file: /etc/hostapd/hostapd.conf
> nl80211: Could not configure driver mode
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> nl80211 driver initialization failed.
> wlan0: interface state UNINITIALIZED->DISABLED
> wlan0: AP-DISABLED 
> hostapd_free_hapd_data: Interface wlan0 wasn't started
> sudo: iptables: command not found
> sudo: iptables: command not found
> sudo: iptables: command not found
> sudo: iptables: command not found
> Listening on port 80
> 1557289525: mosquitto version 1.4.8 (build date Wed, 13 Feb 2019 00:27:01 +0000) starting
> 1557289525: Using default config.
> 1557289525: Opening ipv4 listen socket on port 1883.
> 1557289525: Opening ipv6 listen socket on port 1883.

Does anyone have any tips?

EvertDekker, 2019-05-09:
Also worth trying an alternative branch based on @oblique's create_ap with git clone -b create-ap https://github.com/kueblc/tuya-convert.git. Note that you'll have to run install_prereq.sh again.

kueblc, 2019-05-14:
@ChrisN2501 have you been able to resolve this? Have you tried `create_ap` as @EvertDekker mentioned?

kueblc, 2019-05-14:
Please also provide information about your hardware and software configuration.

kueblc, 2019-07-02:
Closing due to inactivity, please let us know if you still need help

closed 2019-07-02 by kueblc