ct-Open-Source/tuya-convert #191: TypeError: IV is not meaningful for the ECB mode

capstan1, 2019-05-08:
I'm trying to flash an Blitzwolf BW-SHP2 bought last year. It was never connected to the tuya app. I'm trying to flash from an Ubuntu 16.04 machine using [this guide.](https://blog.quindorian.org/2019/02/home-assistant-10-wifi-energy-meter-with-esphome.html/).

However I'm stuck on 
```
Starting pairing procedure in screen
~/Desktop/androidtv/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
```
The device is never found, the LED keeps flashing rapidly.
The screenlog.0 shows:

```
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlp2s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp2s0 with hwaddr 78:92:9c:e3:62:37 and ssid "vtrust-flash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
Listening on port 80
1557308623: mosquitto version 1.4.8 (build date Wed, 13 Feb 2019 00:27:01 +0000) starting
1557308623: Using default config.
1557308623: Opening ipv4 listen socket on port 1883.
1557308623: Opening ipv6 listen socket on port 1883.
wlp2s0: STA 68:c4:4d:42:03:01 IEEE 802.11: authenticated
wlp2s0: STA 68:c4:4d:42:03:01 IEEE 802.11: associated (aid 1)
wlp2s0: AP-STA-CONNECTED 68:c4:4d:42:03:01
wlp2s0: STA 68:c4:4d:42:03:01 RADIUS: starting accounting session 5CD2A4CF-00000000
wlp2s0: STA 68:c4:4d:42:03:01 WPA: pairwise key handshake completed (RSN)
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/eric/Desktop/androidtv/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/eric/Desktop/androidtv/tuya-convert/scripts/smartconfig/multicast.py", line 14, in <module>
    aes = AES.new( "a3c6794oiu876t54", AES.MODE_ECB, '' )
  File "/home/eric/.local/lib/python3.5/site-packages/Crypto/Cipher/AES.py", line 232, in new
    return _create_cipher(sys.modules[__name__], key, mode, *args, **kwargs)
  File "/home/eric/.local/lib/python3.5/site-packages/Crypto/Cipher/__init__.py", line 77, in _create_cipher
    raise TypeError("IV is not meaningful for the ECB mode")
TypeError: IV is not meaningful for the ECB mode
```

kueblc, 2019-05-08:
Very strange that you should get this error, since clearly an IV is not used: `aes = AES.new( "a3c6794oiu876t54", AES.MODE_ECB, '' )` where the third parameter `''`, an empty string, is the IV. Perhaps you can try removing it entirely and see if that fixes it for you.

Can you please provide version numbers for your Python, pip, and Crypto packages?

capstan1, 2019-05-08:
Thanks for your quick answer!

what do you mean by 

> "Perhaps you can try removing it entirely and see if that fixes it for you."

```
Python 2.7.12
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
```

`crypto --version`  doesnt work....

kueblc, 2019-05-08:
Open `/home/eric/Desktop/androidtv/tuya-convert/scripts/smartconfig/multicast.py` and change line 14 from `aes = AES.new( "a3c6794oiu876t54", AES.MODE_ECB, '' )` to `aes = AES.new( "a3c6794oiu876t54", AES.MODE_ECB )`

linusbarth, 2019-05-13:
> Open /home/eric/Desktop/androidtv/tuya-convert/scripts/smartconfig/multicast.py and change line 14 from aes = AES.new( "a3c6794oiu876t54", AES.MODE_ECB, '' ) to aes = AES.new( "a3c6794oiu876t54", AES.MODE_ECB )

I had the same problem and followed your advice and two new problems arised. To solve all of them I needed to replace the lines 14-15 by: 
`key = b'a3c6794oiu876t54'
aes = AES.new( key, AES.MODE_ECB )
encrypt = lambda data : aes.encrypt( pad(data, 16).encode('UTF-8') )`

And now it worked and I could follow up with the `curl http://10.42.42.42/flash3` command and the "sonoff-xxxx"-accesspoint appeared. Thank you very much.

kueblc, 2019-05-14:
Glad it worked for you @linusbarth. It looks to me like you are using Python <= 3.4, which may explain this issue.

@capstan1 Have you resolved your issue? You can try upgrading your Python installation or try the code modification. Let us know what worked for you.

kueblc, 2019-10-04:
Implemented suggestion by @linusbarth in #291, thanks @linusbarth!

closed 2019-10-04 by kueblc