ct-Open-Source/tuya-convert #192: Tuya curtain switch won't flash

HA-TB303, 2019-05-09:
Device:
https://www.aliexpress.com/item/Tuya-Smart-Life-WiFi-Curtain-Switch-for-Electric-Motorized-Curtain-Blind-Roller-Shutter-Google-Home-Amazon/32963064005.html

Issue:

Hi tried flashing a Tuya Wifi Curtain switch (brandless) but it won't work. I have used the same setup (raspberry pi 3+ using the internal wifi) several times before with success.

Is this a device with new firmware which can not be flashed with tuya convert, or is there still hope? ;)

Logs: 

```
pi@raspberrypi:~/tuya-convert/scripts $ cat *.log
1557382733: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1557382733: Using default config.
1557382733: Opening ipv4 listen socket on port 1883.
1557382733: Opening ipv6 listen socket on port 1883.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Listening on port 80
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:15:c8:9f and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED a4:cf:12:a3:61:08
wlan0: STA a4:cf:12:a3:61:08 RADIUS: starting accounting session 033E5D2F101C8A6D
wlan0: STA a4:cf:12:a3:61:08 WPA: pairwise key handshake completed (RSN)
wlan0: STA 70:70:0d:0f:99:e5 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 70:70:0d:0f:99:e5
wlan0: STA 70:70:0d:0f:99:e5 RADIUS: starting accounting session 28C06D7793334D7E
wlan0: STA 70:70:0d:0f:99:e5 WPA: pairwise key handshake completed (RSN)
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED a4:cf:12:a3:61:08
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: disassociated
wlan0: STA a4:cf:12:a3:61:08 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED a4:cf:12:a3:61:08
wlan0: STA a4:cf:12:a3:61:08 RADIUS: starting accounting session 6C10E6FFFA89C6D5
wlan0: STA a4:cf:12:a3:61:08 WPA: pairwise key handshake completed (RSN)
````

HA-TB303, 2019-05-10:
When attempting the flash with the device in pairing mode, after a few seconds the devices LED stops flashing and then turns off. The device does not seem to reboot, but it does leave the pairing mode.

When I put it in pairing mode without attempting the flash, it stays in pairing mode so there is some communication going on.

WagnerBeccard, 2019-05-11:
looks like mine 
``` 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:23:63:06 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 14:30:c6:3f:69:ec
wlan0: STA 14:30:c6:3f:69:ec RADIUS: starting accounting session DA93BEEE0BAFD932
wlan0: STA 14:30:c6:3f:69:ec WPA: pairwise key handshake completed (RSN)
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 14:30:c6:3f:69:ec
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: associated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: associated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: associated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:94:d1:52:a1
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:94:d1:52:a1
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: associated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:94:d1:52:a1
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:94:d1:52:a1
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: disassociated
wlan0: STA 60:01:94:d1:52:a1 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 60:01:94:d1:52:a1
wlan0: STA 60:01:94:d1:52:a1 RADIUS: starting accounting session FE261F04550B051C
wlan0: STA 60:01:94:d1:52:a1 WPA: pairwise key handshake completed (RSN)
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 14:30:c6:3f:69:ec
wlan0: STA 14:30:c6:3f:69:ec RADIUS: starting accounting session E5B3468A892AF34B
wlan0: STA 14:30:c6:3f:69:ec WPA: pairwise key handshake completed (RSN)
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 14:30:c6:3f:69:ec
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
wlan0: STA 14:30:c6:3f:69:ec IEEE 802.11: disassociated
``` 
but then

 ``` 
curl http://10.42.42.37
curl: (7) Failed to connect to 10.42.42.37 port 80: Connection refused ```

DoktorHeisenberg, 2019-05-13:
Have the exact same problem. LED is blinking fast, device (Gosund SP1) connects to the AP, LED switches off. Another SP1 flashed without any problems using the same live environment, so I guess that some devices already have a fixed firmware while others doesn't.

juliogonzalez, 2019-05-13:
Same here with a Tellur TLL331011 bulb I received today (Wifi module 1.0.0/ECU module 1.0.0). 
Despite the version, I guess the firmware is fixed as well.

However if I run:
> nc -u -l -k -p 6666
> nc -u -l -k -p 6667

as suggested at https://github.com/ct-Open-Source/tuya-convert/issues/165#issuecomment-480084833, then can see the bulb output in JSON format (and luckily I can still control it locally using port 6668)

If I can provide more data, including the logs, if needed.

Couldn't see anything particular, on pairing mode, I can't see anything but the bulb getting IP address, when it gets out of pairing mode, it makes the requests, including:
```
URI:/gw.json?a=tuya.device.dynamic.config.get&gwId=XXXXXXXXXXXXXXXXXXX&t=555&v=1.0&sign=YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
WARN: unknown request: tuya.device.dynamic.config.get (/gw.json?a=tuya.device.dynamic.config.get&gwId=XXXXXXXXXXXXXXXXXXX&t=555&v=1.0&sign=YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY)
[32m[I 190513 19:59:53 web:2246][m 200 POST /gw.json?a=tuya.device.dynamic.config.get&gwId=XXXXXXXXXXXXXXXXXXX&t=555&v=1.0&sign=YYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY (10.42.42.40) 7.11ms
```
Maybe this is a duplicate of #185 and #165?

kueblc, 2019-05-14:
@HA-TB303 @Underknowledge as @DoktorHeisenberg mentioned, you may be running a patched firmware. You can verify this running the commands like @juliogonzalez. Stay tuned to see if we can release a fix, or flash it via serial.

@juliogonzalez That's good news you are getting JSON output, so I'd say this is a different issue entirely. Some devices have broken update mechanisms, but we'll need to do a little more digging to be sure. Please open a new issue with all the details to keep things organized.

haselchen, 2019-05-17:
Same issue here with a Digoo SP 202 Plug (WIFI 1.1.1   ,  MCU  1.1.1)
Serial Flash is not possible. The circuit board is too well secured. So i pray to god, that you´ll find a solution for OTA Flash.

drock1985, 2019-05-18:
Same here currently with a Merkury light bulb.... just can't get it to flash. I'm beginning to think they have been patched.

lucmbahmoute, 2019-05-27:
> Same here currently with a Merkury light bulb.... just can't get it to flash. I'm beginning to think they have been patched.

Same here just returned mine. I am starting to think if there are still old stocks...

davericher, 2019-05-30:
and to think I went to Walmart and bought other stuff while picking up these bulbs, ahh well at least I got 3 in before the patches, sorry Walmart, our relationship is over

davericher, 2019-05-30:
5* I returned the two Merkury flood light syle rgbws, and picked up two more of the regular bulb style rgbws, there still good to go

lucmbahmoute, 2019-05-30:
> 5* I returned the two Merkury flood light style rgbws, and picked up two more of the regular bulb style rgbws, there still good to go
> 
> 7* I forgot the light boxes (3d printed lithophane boxes) i have some of these in

You mean white + colour ones? Were you able to flash successfully? Maybe I should try picking up more

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc