ct-Open-Source/tuya-convert #194: Unable to flash Tellur TLL331011 bulb

juliogonzalez, 2019-05-14:
Device type: Bulb
Model: Tellur TLL331011 10W RGB
Wifi module: 1.0.0 (according to Tuyalife application).
ECU module: 1.0.0 (according to Tuyalife application).
Manual: https://www.tellur.com/wp-content/uploads/2018/08/TLL331011-EN-User-Manual-WiFi-Smart-Bulb-E27-10W-whitewarmRGB-dimmer.pdf

It seems tuya-convert is unable to detect the device when it's on pairing mode and with a telephone connected to tuya-convert wifi (to get fast blinking the user needs to turn ON-OFF-ON-OFF-ON).

It seems the fast blinking mode is AZ mode. I tried AP as well, but without luck.

However it seems tuya-convert can see the bulb when it's on normal mode (requests to Tuya cloud are present at the log).

Logs (I masked IDs and signatures): 
[logs.tar.gz](https://github.com/ct-Open-Source/tuya-convert/files/3179931/logs.tar.gz)

As suggested at https://github.com/ct-Open-Source/tuya-convert/issues/165#issuecomment-480084833, I ran:

> nc -u -l -k -p 6666
> nc -u -l -k -p 6667

And in both cases I can see the bulb sending json data (including ID and Key it seems!)

If I can help with any other log or test, just least me know :-)

juliogonzalez, 2019-05-14:
BTW, it seems today I can only see the bulb at 6666. Maybe yesterday I was running netcat using port 6666 twice at the same time.

Anyway, here is the (masked) data:
```
U��{"ip":"XXX.XXX.XXX.XXX","gwId":"XXXXXXXXXXXX","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"XXXXXXXXXX","version":"3.1"}r�o�U
```

kueblc, 2019-05-14:
Try changing line 83 in `fake-registration-server.py`
from
`answer = b'{"result":{"auto":3,"fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))`
to
`answer = b'{"result":{"auto":true,"type":0,"size":"%d","fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_len,file_md5.encode('utf-8'))`
and let me know how it goes

juliogonzalez, 2019-05-14:
Done (I commented line 83, and added line 84):
```
pi@raspberrypi:~/tuya-convert/scripts $ grep -n 'answer = ' fake-registration-server.py 
83:            #answer = b'{"result":{"auto":3,"fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))
84:            answer = b'{"result":{"auto":true,"type":0,"size":"%d","fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_len,file_md5.encode('utf-8'))
```
After connecting the phone to the wifi network, configuring the bulb to learn mode (fast blinking), and hitting enter, all I can see at the screen is:
```
======================================================
Starting pairing procedure in screen
RTNETLINK answers: File exists
~/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
.....................................
```
And at the logs:
```
==> smarthack-smartconfig.log <==
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```
And the bulb keeps blinking for some minutes after (I guess) giving and stop blinking even without being connected to any network.

I guess I should not worry about:
```
RTNETLINK answers: File exists
```
As it is just the result of `sudo ip route add 255.255.255.255 dev $WLAN`

I verified that anyway the route is there:
```
pi@raspberrypi:~/tuya-convert $ ip route|grep wlan
10.42.42.0/24 dev wlan0 proto kernel scope link src 10.42.42.1 
169.254.0.0/16 dev wlan0 proto kernel scope link src 169.254.133.178 metric 303 
255.255.255.255 dev wlan0 scope link
```

juliogonzalez, 2019-05-14:
Funny fact: After the bulb stops blinking, it just started using the old config, as I can see it connected to my network, requesting IP via DHCP and again sending the JSON data to the network.

kueblc, 2019-09-28:
Closing due to inactivity. If you're still having issues, try the latest `tuya-convert` and open a new issue.

closed 2019-09-28 by kueblc