ct-Open-Source/tuya-convert #195: Support Blitzwolf SHP2 V2.4

JBS5, 2019-05-15:
Just received my Blitzwolf SHP2 which is a V2.4 and is not found when running the script, because the firmware has the HTTPS update I guess (https://github.com/ct-Open-Source/tuya-convert/issues/160#issuecomment-479057062). Is there anything to say about supporting this new revision? It would be great if soldering is not needed with this new version.

 [image]

soxhi8, 2019-05-16:
Same here, Gosund SP1-variant though, but same PCB. Firmware version according to Smart Life is 1.06.
Would a dump help?

soxhi8, 2019-05-18:
Dumped the firmware of my 2.4 (Gosund SP1).
Version number is 1.06.

[1.06_Gosund.bin.zip](https://github.com/ct-Open-Source/tuya-convert/files/3193462/1.06_Gosund.bin.zip)

Ixtalo, 2019-05-28:
I can confirm that.
New v2.4 20181101 devices (bought 05/2019) have a new firmware with version 1.06; old working versions were 1.05.
New devices with new firmware v1.06 use HTTPS (e.g., https://a3.tuyaus.com/gw.json, found in firmware) and try to connect to port 443 (instead of 80).

It does not help to simply reconfigure the fake-registration-server.py to port 443.
Then it will connect (can be verified by tcpdump), but the HTTPS TLS handshake seems to fail.

@soxhi8: The dumped Blitzwolf SHP2 firmware is mostly identical to "1.06_Gosund.bin.zip".

Old firmware:
```
2nd boot version : 1.4(b1)
Oct  8 2018 11:43:27 5.33 2.1 3.1 
OS SDK ver: 1.4.2(78f3caf) compiled @ Oct 23 2017 13:45:35
1.0.5 Oct  8 2018 11:43:27 5.33 2.1 3.1 
```

New firmware:
```
2nd boot version : 1.7(5d6f877)
OS SDK ver: 2.0.0(e8c5810) compiled @ Jan 25 2019 14:26:04
1.0.6   Mar  2 2019 14:44:54    5.42    2.2 3.3 
sign    https://a3.tuyacn.com/gw.json   a   %u  https://a3.tuyaeu.com/gw.json   https://a3.tuyaus.com/gw.
encrypt ÿÿÿÿÿÿ  tuya_tls.c  
extended master secret  master secret   key expansion   unknown SSLv3.0 TLSv1.0 TLSv1.1 TLSv1.2
TLS-PSK-WITH-AES-128-CBC-SHA256
```

Similar issues from the last days:
#197 
#199 
#201

Niek, 2019-06-04:
See discussion in this topic: https://github.com/ct-Open-Source/tuya-convert/issues/166

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc