ct-Open-Source/tuya-convert #197: Teckin SP22 not flashable anymore with update 1.0.7

sebastianklein96, 2019-05-17:
I'm hereby reporting another device that can not be OTA-flashed anymore after the newest Tuya update.

Device: Teckin SP22
WIFI Firmware v1.0.7
MCU Firmware v1.0.7

Tuya connect is running on a RPi3 with a fresh install of Raspbian. An Android smartphone is connected for Smartconfig to work properly.

With the newest update applied, the device briefly establishes a connection with the vtrust-flash WiFi network broadcasted by my RPi3 but then immediately disassociates.

I have attached the four log files from tuya-connect as well as a pcap file containing all traffic captured on wlan0 during the process. The pcap file and the log files are from the same flashing attempt.

Summary of the information extracted from logs and pcap as far as I can understand it:
SP22 is referenced as Espressi_c2:eb:b5. Its full MAC address is 60:01:94:C2:EB:B5.
My phone's (OnePlus 6) MAC address is 64:A2:F9:82:FE:F3.

- the interface wlan0 is created and vtrust-flash is broadcasted
- the RPi3 has IP 10.42.42.1
- the phone associates, gets IP 10.42.42.22
- the Espressi_c2:eb:b5 broadcasts LLC
- 4-Way-Handshake between RPi3 and Espressi_c2:eb:b5 is completed in frames 1537-1540
- Espressi_c2:eb:b5 is assigned IP 10.42.42.38
- 10.42.42.38 starts DNS request for a3.tuyaus.com in frame 1566 (0xa80b A a3.tuyaus.com)
- 10.42.42.1 responds
- 10.42.42.38 tries to send TCP packet from port 56662 to 443 (SSL) in frame 1568
- 10.42.42.38 starts another DNS request for a3.tuyaus.com in frame 1597 (0x6d1d A a3.tuyaus.com)
- 10.42.42.1 responds again
- 10.42.42.38 tries to send another TCP packet from port 29202 to 443 (SSL) in frame 1599
- One more ARP reply by Espressi_c2:eb:b5 is recored in frame 1602
- Afterwards: silence. No Espressi_c2:eb:b5 is to be seen anymore.

I followed the instructions bit by bit and in my opinion the dnsmasq blocking is working. When I try to visit a3.tuyaus.com from the connected phone, I'm just greeted with dnsmasq's hello world page. This seems to prove the theory that Tuya devices are now trying to communicate via SSL and are disassociating immediately when an invalid packet is received.

If more information is required I'm happy to deliver. Sadly I'm not able to solder to the ESP8266 since the case is glued.

sebastianklein96, 2019-05-17:
I have just ordered another fresh batch of six SP22s. **(For everyone looking to buy some: get the three pack on Amazon DE right now. They're 8€ a pop with a 10€ coupon code that is offered directly on the product page until 21st of May)** When they arrive I'll try to flash them with Tuya Convert directly out of the box using only devices that are not connected to the internet in any way to prevent any firmware upgrades or successfull DNS requests.

I'm expecting to get devices with a firmware version that is still flashable. I'll still capture all network traffic generated in the procedure and provide it + logs as above. 

I'll also provide the backed up bin files and maybe I'll sacrifice one of the already updated plugs to see if I can extract the newer firmware.

probonopd, 2019-05-17:
I have received exactly this thee pack from Amazon DE today and have flashed them semi-successfully before they had any chance to connect to the Internet. I have backups of the original firmware in case anyone needs it.

sebastianklein96, 2019-05-17:
I've foolishly made the mistake of connecting them to the app first to see if they work. All four I had... Obviously they got the update immediately. 

We should probably both update this issue as we gather new information since finding someone with the same hardware seems to be quite rare.

probonopd, 2019-05-17:
> I've foolishly made the mistake of connecting them to the app first to see if they work. All four I had... Obviously they got the update immediately.

Hence, I hope https://github.com/ct-Open-Source/tuya-convert/pull/198 will get merged.

sebastianklein96, 2019-05-18:
I have received the new batch of SP22s this morning and am happy to report that flashing them out of the box with the provided Tasmota 6.5.0 bin works like a charm. Afterwards I connected them to my Wifi and configured them for the Teckin (52) template. Power meter and switching works, only the reported voltage has an offset.

Attached you'll find log files, the network traffic captured on wlan0 during the flash and the original bin files.

I was not able to pry open one of the devices with the new Tuya firmware. They're so well glued together that the ESP8266 would be destroyed trying to open them. If at all, they can be opened from the back. But there's a lip in the injection mold that stops one from inserting any prying tools further than 1mm. So the only option would be sawing them open, which I don't have the appropriate tools for.

[device-info-6f77.txt](https://github.com/ct-Open-Source/tuya-convert/files/3194081/device-info-6f77.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3194082/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3194083/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3194084/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3194085/smarthack-wifi.log)
[pcap and bin.zip](https://github.com/ct-Open-Source/tuya-convert/files/3194086/pcap.and.bin.zip)

probonopd, 2019-05-18:
Do the red and blue LEDs behave correctly for your SP22s with the Teckin (52) template @sebastianklein96?

haselchen, 2019-05-18:
@sebastianklein96

Step by Step Instructions for Noobs? 😊

sebastianklein96, 2019-05-18:
> 
> 
> Do the red and blue LEDs behave correctly for your SP22s with the Teckin (52) template @sebastianklein96?

Yes, they do. I didn't change anything except for calibrating the power measurements and it works like a charm. Did you flash the Tasmota bin that was delivered with tuya-convert? Tuya-convert ships with Sonoff Basic. Maybe replace the sonoff-basic.bin in ~/tuya-convert/files/ with the sonoff.bin from the newest Tasmota release. 

> 
> 
> @sebastianklein96
> 
> Step by Step Instructions for Noobs? 😊

Well the instructions are already in the README of this project. I set up a fresh install of Raspbian on my RPi3, cloned the repository, installed the dependencies using the install script.
Afterwards I ran the ./start_flash script, connected a laptop to the vtrust-flash wifi network, plugged in the SP22, waited about five seconds for it to boot and then held the button for about 7 seconds until the led started to flash blue. Then I hit enter in the flash script. Took about 10 seconds until the plug was recognized and the firmware dump began. Afterwards I just did the curl http://10.42.42.42/flash2 and curl http://10.42.42.42/flash3 . A new wifi popped up from the plug and I connected to it with the laptop, navigated to 192.168.4.1 and put in my wifi information.

probonopd, 2019-05-18:
@sebastianklein96 it is all working now for me; I had not entered my MQTT server details yet and this is why it was flashing blue. Now the LEDs are working properly.

haselchen, 2019-05-18:
@sebastianklein96

Whats the different between this way and the normal way of tuya convert flash?

sebastianklein96, 2019-05-19:
> @sebastianklein96 it is all working now for me; I had not entered my MQTT server details yet and this is why it was flashing blue. Now the LEDs are working properly.

@probonopd Glad to hear that!

> Whats the different between this way and the normal way of tuya convert flash?
> I bought a Digoo Plug . But the WiFi Version is 1.1.1
> So the normal way of Flashing with Tuya Convert fails.

@haselchen I literally followed the instructions provided in the Tuya Convert Readme. The description that I gave earlier is just me following those instructions.

In which way does the flash of your Digoo plug fail? Did you update its firmware or connect it to the internet before attempting the flash?

haselchen, 2019-05-19:
I wanted to flash the plug as always. Step by step following the instructions as always.
The leds are blinking , ready for flashing. So i hit Enter.  After ca. 3seconds. the Plug makes "klick" and the leds are off. Plug is back to normal mode.
I think it is because of the new firmware. (WIFI 1.1.1)
Every owner of a tuya device describes the current failure during flashing.

Why does it work for you with the firmware over 1.0.5?

(Can we also treat the problem in German?)

probonopd, 2019-05-19:
It only works if you do the flash directly with factory-new devices that have never accessed the Internet or the app, because they update themselves to a new firmware from the factory after which they can no longer be flashed with tuya-convert.

Es funktioniert nur direkt mit fabrikneuen Geräten, die noch nie mit dem Internet oder der App verbunden waren, weil sie sich sonst von selbst auf eine neue Firmware des Herstellers aktualisieren, mit der sie nicht mehr mit tuya-convert geflasht werden können.

sebastianklein96, 2019-05-19:
As probonopd said, it only works if the plugs didn't have any chance to receive an update. The devices I used for updating all did not have an internet connection and therefor the plugs could be flashed since they had no way to update.

-German-
Wie Probonopd es gesagt hat. Du musst auf jeden Fall verhindern, dass die Geräte irgend eine Internetverbindung haben. Ich tippe mal, dass der dnsmasq Filter, der mit Tuya Convert ausgeliefert wird, nicht mehr greift, weil die Geräte jetzt über SSL kommunizieren. Auch, wenn der Tipp unethisch ist: falls du den Stecker bei Amazon gekauft hast, melde eine Retoure an und kauf den selben Stecker noch mal. Vielleicht hast du Glück und der neue Stecker kommt mit der alten Firmware. Dann kannst du mit Geräten ohne Internetzugang flashen.

haselchen, 2019-05-19:
I bought the double socket at Banggood. I think that only devices with new firmware are delivered.  
I've connected it to the app now.
In your opinion, is there a possibility that despite connecting to Tuya, a flash (if a solution was found) is still possible?

-German-
Hab die Doppelsteckdose bei Banggood gekauft. Ich denke, dass nur noch Geräte mit neuer Firmware ausgeliefert werden. Weil ich die Dose benutzen will, habe ich sie natürlich jetzt mit der App verbunden.
Besteht deines Erachtens die Möglichkeit, dass trotz Verbindung zu Tuya, ein Flash (wenn eine Lösung gefunden wurde) noch möglich ist?

sebastianklein96, 2019-05-19:
With the current version of Tuya Convert, I don't think it's possible. The developers are already looking into ways to circumvent the new security measures though. But I don't have any insight on how long that will take. It's probably gonna be a while since they literally have to emulate Tuya's SSL responses as far as I understand. I think your safest bet would be to get rid of the plugs you have now if that is still possible (don't know banggoods return policy). 

-German-
Mit der aktuellen Version von Tuya Convert wird das wohl nichts. Allerdings suchen die Entwickler wohl schon Wege, die neuen Sicherheitsmaßnahmen von Tuya zu umgehen. Keine Ahnung, ob und wann das was wird. Ich würde aber sagen, dass es nen Moment dauern wird, SSL-Verschlüsselung zu reverse engineeren. Wenn du kannst, sieh zu dass du die Stecker wieder los wirst und dir Ersatz besorgst, von dem du weißt dass er flashbar ist oder den du einfach zurück geben kannst. Wie gesagt, aktuell ist das Dreierpack SP22 auf Amazon noch für 25€ zu haben und die die ich bekommen habe, waren flashbar (vorausgesetzt, man gibt ihnen keine Möglichkeit, vor dem Flash ins Internet zu kommen).

eterpstra, 2019-05-20:
I have received 4 x SP111 Go-sund sockets.  The sockets havent connected to the internet yet. Alltough i have succefully flashed 1 device, the other 3 are not working :(

Same as @haselchen  (The leds are blinking , ready for flashing. So i hit Enter. After ca. 3seconds. the Plug makes "klick" and the leds are off. Plug is back to normal mode.)

I don't know how one 1 plug was flashed succes, the other three are not working.

ngdio, 2019-05-20:
Can't flash Teckin FL41 either. Same issue

eterpstra, 2019-05-21:
Is there a way to check the firmware without using the app?

sebastianklein96, 2019-05-21:
@eterpstra Asking the same question in three seperate issues isn't exactly good etiquette.

Answer:
https://github.com/ct-Open-Source/tuya-convert/issues/160#issuecomment-494515604

hazardlabs, 2019-05-24:
Same issue with brand new Teckin SP27's from Amazon UK, I can only assume they are now coming from the factory patched?

Xan8, 2019-05-24:
I just bought 4 teckin sp22 and i can't flash them. I didn't connected them to the internet, I just plug them in and when tuya converter is searching (. . . . .) the blinking led of the plug stops blinking. Shoud I return them (Amazon) or do u thing this will be solved soon (less than 1 month)?

sebastianklein96, 2019-05-25:
@Xan8 just keep an eye on this repository and see if some development is done. If not, just return them before the end of your return period.

eterpstra, 2019-05-28:
@sebastianklein96 But soldering is still possible right?

sebastianklein96, 2019-05-28:
@eterpstra I don't know anything about that. But if I had to guess, I'd say it should still be possible since I don't know of a way to "write-protect" an esp8266's flash memory. Just guessing though.

baldfox, 2019-06-18:
Any way to set up an alert so that we would know when a newer version of Tuya convert is uploaded? I have a bricked Teckin SP22 at the moment, but don't want to crack it open due to it being used with mains power etc... (don't feel comfortable trying to seal and glue it all back together afterwards). This one was upgraded accidentally to the latest firmware, and now nothing recognises it.

kueblc, 2019-06-18:
@baldfox You can click the "Watch" button at the top of the page to get notifications about development activity. You can also try [codetheweb/tuyapi](https://github.com/codetheweb/tuyapi) if your goal is local control, next best thing to custom firmware.

baldfox, 2019-06-19:
> @baldfox You can click the "Watch" button at the top of the page to get notifications about development activity. You can also try [codetheweb/tuyapi](https://github.com/codetheweb/tuyapi) if your goal is local control, next best thing to custom firmware.

@kueblc Thanks a lot. I've never really used github before. Appreciate your help!

thirug010, 2019-07-06:
@baldfox, 
Please take look at the solution I am working on, it provides the virtually tasmota features on tuya devices using tuyapi with flashing it. [Virtual Tasmota for Tuya Api Devices](https://github.com/thirug010/Virtual-Tasmota-for-Tuya-api-devices/blob/master/README.md). 

[image]

Thanks 
Thiru

tjamesdp, 2019-07-12:
> With the current version of Tuya Convert, I don't think it's possible. The developers are already looking into ways to circumvent the new security measures though. But I don't have any insight on [...]

Would it be possible to implement sslstrip in to the tuya-convert software to strip out the https transactions and downgrade it to http? https://github.com/moxie0/sslstrip

baldfox, 2019-07-15:
> @baldfox,
> Please take look at the solution I am working on, it provides the virtually tasmota features on tuya devices using tuyapi with flashing it. [Virtual Tasmota for Tuya Api [...]

Thanks for this. I looked at it. I really don't want to keep the tuya/smartlife firmware on there though at the moment. I don't want any of my smarthome products phoning home. I guess I'll need to wait for now.

baldfox, 2019-07-15:
I thought I'd have another attempt at this. If i hold down the teckin sp22 for 5 seconds it blinks rapidly. If I then hold the button again for a further 7 seconds, it flashes blue slowly. It will then throw up an SSID of smartlife-XXXX. I then try and connect to that but I can't get any further. Anyone know the default webpage to try and administer it again? At this junction I think i'd rather just revert the SP22 to default and try and use it with Tuya until the SSL is broken and I can go OTA again.

kueblc, 2019-07-15:
> I don't want any of my smarthome products phoning home

You can try blocking `*.tuya(us|eu|cn).com` at your router and use [codetheweb/tuyapi](https://github.com/codetheweb/tuyapi/) to control your devices locally.

> If i hold down the teckin sp22 for 5 seconds it blinks rapidly. If I then hold the button again for a further 7 seconds, it flashes blue slowly. It will then throw up an SSID of smartlife-XXXX.

`tuya-convert` only works with the first mode, known as EZ config. It does not work with AP config.

baldfox, 2019-07-16:
> `tuya-convert` only works with the first mode, known as EZ config. It does not work with AP config.

Thanks for that. It saves me some time. Unfortunately in AP mode the snart-life app doesn't conclude it's registration so it's well and truly fubarred I think.

timbo-lino, 2019-08-28:
Hi, 

I just bought some TECKIN SP22 from Amazon. Thought it would work out.
Did a fresh Rasperry Pi Setup and got stock here

file:///home/tim/Bilder/Screenshots/Screenshot_20190828_100242.png

It seems that I cant flash them.

Is there a workaround/ a solution already out?
Will there be a solution in the future?
Should I just send them back?

What would you guys do?

baldfox, 2019-08-28:
I had the same issue. The only thing I can suggest is that we all write to Teckin and ask them for a work around. If there are enough of us it might work. We can highlight the success of Shelly and their approach to open source software as an example of how well they could do.

thirug010, 2019-08-29:
as of now we have only few alternative options ,

1. use tuya-api based control without flashing device with tasmota firware ([Refer here](https://github.com/thirug010/Virtual-Tasmota-for-Tuya-api-devices/blob/master/README.md) or [here ](https://github.com/codetheweb/tuyapi) )

2. use soldering iron to hard wire flashing with tasmota _(You might need to put MCU in reset mode along with putting the esp in flashing mode to avoid the serial data interruptions from MCU mostly for dimmers)_

3. use shelly or sonoff diy devices for easy flashing support

Thanks 
Thiru

timbo-lino, 2019-08-29:
> @timbo-lino,
> 
> as of now we have only few alternatives ,
> 
>     1. use tuya-api based control without flashing device with tasmota firware ([Refer here](https://github.com/thirug010/Virtual-Tasmota-for-Tuya-api-devices/blob/master/README.md) or [here ](https://github.com/codetheweb/tuyapi) )
> 
>     2. use soldering iron to hard wire flashing with tasmota _(You might need to put MCU in reset mode along with putting the esp in flashing mode to avoid the serial data interruptions from MCU mostly for dimmers)_
> 
>     3. use shelly or sonoff diy devices for easy flashing support
> 
> 
> Thanks
> Thiru

Thanks for pointing out all the options.

Tuya-api based control seems to be an overkill for me, because as far as I see, I need to run a service 24/7 to use it.

I think the SP22`s are going back to Amazon and I will replace it with a shelly or sonoff.

pandit77, 2019-09-20:
TECKIN SP22 Users check: https://github.com/ct-Open-Source/tuya-convert/issues/273

I just sucessfully flashed four SP22 Plugs

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc