ct-Open-Source/tuya-convert #199: Gosund WP3 Endless ...........  Won't Flash

cheechie, 2019-05-20:
Just purchased a Gosund model- WP3.  It has the 1.0.5 firmware https://www.amazon.com/Assistant-Required-Enabled-Control-Gosund/dp/B072ZX8RTZ

I have tried flashing with both Kali Linux in Virtual Box and with a Raspberry Pi B3.  
When I put the plug into pairing mode and then press enter to flash.  The plug stops blinking fast but it never programs.  In both cases I get the exact same error.  

Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
........................................................................................................

Here are my logs.

**smarthack-wifi.log**

^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 60:01:94:c8:bd:3b
wlan0: AP-STA-DISCONNECTED c8:38:90:13:35:bf
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlan0 with hwaddr d8:3d:4c:94:35:f5 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA c8:38:90:13:35:bf IEEE 802.11: authenticated
wlan0: STA c8:38:90:13:35:bf IEEE 802.11: associated (aid 1)
wlan0: AP-STA-CONNECTED c8:38:70:13:35:df
wlan0: STA c8:38:90:13:35:bf RADIUS: starting accounting session 44D22CF0B138FFAC
wlan0: STA c8:38:90:13:35:bf WPA: pairwise key handshake completed (RSN)
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: authenticated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: associated (aid 2)
wlan0: AP-STA-CONNECTED 60:01:94:c8:bd:3b
wlan0: STA 60:01:94:c8:bd:3b RADIUS: starting accounting session CB46AEC709694B4F
wlan0: STA 60:01:94:c8:bd:3b WPA: pairwise key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED 60:01:94:c9:bd:3a
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: authenticated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: associated (aid 2)
wlan0: AP-STA-CONNECTED 60:01:94:c8:bd:3b
wlan0: STA 60:01:94:c8:bd:3b RADIUS: starting accounting session 31D0DEBEC5B93FA6
wlan0: STA 60:01:94:c8:bd:3b WPA: pairwise key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED60:01:94:c8:bd:3b
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: authenticated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: associated (aid 2)
wlan0: AP-STA-CONNECTED 60:01:94:c8:bd:3b
wlan0: STA 60:01:94:c8:bd:3b RADIUS: starting accounting session 31D0DEBEC5B93FA6
wlan0: STA 60:01:94:c8:bd:3b WPA: pairwise key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED 60:01:94:c8:bd:3b
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

**smarthack-web.log**

^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/lib/python3/dist-packages/tornado/platform/asyncio.py", line 132, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
[32m[I 190520 10:22:01 web:2162][m 200 GET / (10.42.42.24) 0.93ms
[33m[W 190520 10:22:01 web:2162][m 404 GET /favicon.ico (10.42.42.24) 2.73ms

probonopd, 2019-05-21:
Did you flash the device out of the factory-new box, or did the device have contact to the app and/or the internet before you tried to flashed it?

cheechie, 2019-05-21:
I flashed it out of the factory-new box, it was never connected to the cloud.

eterpstra, 2019-05-21:
How can i check what the firmare version is?

sebastianklein96, 2019-05-21:
@eterpstra Asking the same question in three seperate issues isn't exactly good etiquette.

Answer:
https://github.com/ct-Open-Source/tuya-convert/issues/160#issuecomment-494515604

jfractalj, 2019-05-22:
I'm having the same issue with the same model device.

cheechie, 2019-05-22:
> I'm having the same issue with the same model device. I am running the script from an Ubuntu 18.04 VM.

Did you flash it right out of the box?

jfractalj, 2019-05-22:
Yes I did. I've had it sitting in the box for over a month. I also noticed that the resolv.conf is not providing DHCP. Do you know if that is expected behavior?

cheechie, 2019-05-22:
Where is resolv.conf located, I don't see it.

jfractalj, 2019-05-22:
sorry, I was typing without thinking. The dnsmasq process was not running. A symptom is not being able to get an IP via DHCP on your secondary device (smartphone, etc). Are you having the same issue?

cheechie, 2019-05-22:
My 2nd device (I have used both iphone and android) has access to the internet when connecting to the vtrust-flash network from my wifi dongle.  I never looked or check their ip's

jfractalj, 2019-05-22:
Yeah, the vtrust-flash network doesn't route out to the internet. They are routing out through their mobile networks to get there. For me, the process did not work because dnsmasq wasn't working properly. I had a conflict with systemd-resolved. I fixed it by running:
**sudo systemctl stop systemd-resolved**
I then reran the scripts. The process then completed. This also worked with a plug that has been connected to the app.

cheechie, 2019-05-22:
Both devices where in airplane mode and only connected to the vtrust-flash network.  Is this vtrust-flash network not supposed to route out to the internet?

That's good to hear it worked with the one that was connected to the app.

jfractalj, 2019-05-22:
That's my understanding, but I am probably wrong. 

If you run the command **sudo dnsmasq** on the machine you are running the script on, what is the output?

cheechie, 2019-05-22:
I'm not at that machine right now, I will try that command when I'm back.

sebastianklein96, 2019-05-22:
After looking through the Tuya files I'm under the impression that the vtrust-wifi network should indeed forward traffic to the internet.
If you have a look in ~/tuya-convert/scripts/setup_ap.sh, line 62-70, it reads as follows:

```
echo "Configuring AP interface..."
sudo ifconfig $WLAN up 10.42.42.1 netmask 255.255.255.0
echo "Applying iptables rules..."
sudo iptables --flush
sudo iptables --table nat --flush
sudo iptables --delete-chain
sudo iptables --table nat --delete-chain
sudo iptables --table nat --append POSTROUTING --out-interface $ETH -j MASQUERADE
sudo iptables --append FORWARD --in-interface $WLAN -j ACCEPT
```

To my understanding, the line `sudo iptables --table nat --append POSTROUTING --out-interface $ETH -j MASQUERADE` forwards all outgoing traffic from the WLAN interface (vtrust-flash) to $ETH, which would be your ethernet interface connected to the internet. The line following that routes all incoming traffic from your ethernet interface to your wifi interface.

This setup file is called in start_flash.sh, line 39, so it takes effect immediately when starting the flash process. It also makes sense to allow internet access, since tuya-convert allows for the user to grab a 3rd party bin file via http directly through the curl commands while tuya-convert is running.

In that case, assuming that the dnsmasq config running behind tuya-convert isn't catching the traffic to the tuya servers anymore, you would be providing an internet connection to your device by connecting it to vtrust-flash.

eterpstra, 2019-05-22:
@jfractalj I ran that comment and I am getting the following output:

dnsmasq: failed to bind DHCP server socket: Address already in use

jfractalj, 2019-05-22:
@sebastianklein96  yes you are correct of course. I made an offhanded comment about how I thought it worked while troubleshooting my problem instead of looking. My apologies.

@eterpstra it looks like you are having the same issue I was having. If you run **sudo netstat -tulpn | grep :53** it should reveal the process that is blocking dnsmasq.

cheechie, 2019-05-22:
when I run "sudo dnsmasq" I get the same error
dnsmasq: failed to bind DHCP server socket: Address already in use

> @eterpstra it looks like you are having the same issue I was having. If you run **sudo netstat -tulpn | grep :53** it should reveal the process that is blocking dnsmasq.

root@kali:~# sudo netstat -tulpn | grep :53
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      2043/dnsmasq        
tcp6      0      0 :::53                       :::*                       LISTEN      2043/dnsmasq        
udp       0      0 0.0.0.0:53              0.0.0.0:*                               2043/dnsmasq        
udp6     0      0 :::53                       :::*                                        2043/dnsmasq

jfractalj, 2019-05-22:
Looks like you already have dnsmasq running. Is the script running in the background? If not, you can kill that process with **kill -9 2043** and try running the script again to see if you have better luck.

cheechie, 2019-05-22:
> kill that process with **kill -9 2043** and try running the script again to see if you have better luck.

Tried that and no luck.  Weird that it does it on the Raspberry Pi and Virtual Box Kali.

jfractalj, 2019-05-22:
So you may not be having the same issue as me after all. The symptoms look the same as the device not being able to get an IP address, but outside of that I am not sure.  Sorry I haven't been more helpful.

eterpstra, 2019-05-22:
I also tried and kill the processes with the mentioned method but when I put the plugs in pairing mode and start flashing it stops after 2 seconds and then a red light shows up

jfractalj, 2019-05-22:
That's what happened with me. I ran Wireshark on the middlebox machine to inspect traffic and realized that there was no traffic at all passing from the outlet even though the script logs showed that the device joined and disconnected from the AP. This led me to believe that the issue I was having may be a general connectivity issue. This led me down the path of testing the ap script and finding my DHCP issue. I hope the steps I took to troubleshoot help.

cheechie, 2019-05-22:
> I also tried and kill the processes with the mentioned method but when I put the plugs in pairing mode and start flashing it stops after 2 seconds and then a red light shows up

That exactly what mine does with the following error. 

wlan0: AP-STA-DISCONNECTED 60:01:94:c9:bd:3a
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: disassociated
wlan0: STA 60:01:94:c8:bd:3b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

RichardUUU, 2019-05-24:
Same problem here. Fresh install of rPi and tuya-convert. Brand new out-of-the-box Gosund model- WP3.

jfractalj, 2019-05-24:
I have one more device that I can test with, but it has been connected to the internet and is a few months old. If it helps, I can try to replicate your issue to see if there's anything I can do to help. Did you use the prebuilt rPi image?

eterpstra, 2019-05-24:
I don't think that is the problem anymore. The devices which are now available are flashed with the latest firmware unfortunately.

RichardUUU, 2019-05-24:
I also couldn't flash a Topgreener switch. Bummer if I can't use these.

jfractalj, 2019-05-24:
@eterpstra If that were the case, wouldn't the devices I have connected to the internet also have this firmware? I am able to flash those still.

eterpstra, 2019-05-24:
@jfractalj well I am not sure about that. The fact is that the last couple of days people started new topics with different devices which are not flashable anymore. And for the past few months, everything looked fine for "old" models.

jfractalj, 2019-05-24:
@eterpstra Either way, I have a new on on the way that arrives tomorrow. I'll play with it then and see if I encounter the same issue.

cheechie, 2019-05-24:
> @eterpstra Either way, I have a new on on the way that arrives tomorrow. I'll play with it then and see if I encounter the same issue.

I am interested to see what you find out.

jfractalj, 2019-05-26:
The new device worked for me. After flashing I did the curl http://10.42.42.42/undo. The firmware module states it is a 1.0.4 firmware vs the 1.0.5 stated above. It also states no new firmware updates are available. Does anyone have a copy of the 1.0.5 .bin?

pritpaul, 2019-05-26:
I believe I successfully flashed a WP3 ordered from Amazon on 5/24.  It appeared to go through and flash fine, but the sonoff-XXXX AP only appeared for ~30 seconds and I haven't gotten it to reappear since.  That's probably a separate issue though.

jfractalj, 2019-05-28:
I have one more arriving tomorrow ordered from the original link in this thread. If I am still unable to reproduce the issue, if anyone is interested, I am willing to swap a converted WP3 for a 1.0.5 or non-working WP3 in order to help troubleshoot.

jfractalj, 2019-05-28:
I received a new device and it does have the 1.0.5 firmware. I am able to reproduce the issue.

cheechie, 2019-05-29:
Well I hope there is a way around 1.0.5.

cheechie, 2019-05-29:
I hope there is a way around this!

RichardUUU, 2019-05-29:
I'm perversely glad to know there is a good reason for my inability to flash. Now I'm wondering what are the odds this can be overcome? The answer will affect some purchasing decisions. Thoughts anyone?

kymnyth, 2019-06-12:
I bought a pack of the WP3s and WP5s from amazon at the same time.  The WP3s would not flash but the WP5s would.  I ordered another pack of WP5s and by the time I got those, they too had been updated and did not flash. :(

thwe1966, 2019-06-21:
Hi first,

I am new here and have the same problem: the flash process will not start.

Is there now a workarround to be able to flash the devices with the new firmware?

(To my knowledge, it is probably due to the new firmware ... or ???)

CaptClaude, 2019-06-28:
Data Point: Bought 4 Gosund WP3s from Amazon in the US just a week ago. After setting up and trying two different Raspberry Pi 3b's and installing `git clone -b create-ap https://github.com/kueblc/tuya-convert.git` on the second Pi I was still not seeing `vtrust-flash`. Finally I killed `wpa_supplicant` and was able to flash all 4 without issue.  Unfortunately I appear to have fat-fingered the initial WiFi configuration after flashing and all it would do was slow blink blue/red.
Because the whole process interested me, I did a careful job of creating empty log files and capturing them with `tail -f` for two attempts, one with wpa_supplicant (failed) and one without wpa_supplicant (which succeeded). Doing so makes it clear exactly what gets written to the log files for each of the two cases. Note that some log files are not written to. The point is to show what successful and unsuccessful AP activation logs contain.

Logs with wpa_supplicant running:

> ==> ./scripts/smarthack-mqtt.log <==
> 
> ==> ./scripts/smarthack-smartconfig.log <==
> 
> ==> ./scripts/smarthack-web.log <==
> 
> ==> ./scripts/smarthack-wifi.log <==
> Backing up NetworkManager.cfg...
> Restarting NetworkManager...
> Backing up /etc/dnsmasq.conf...
> Writing dnsmasq config file...
> Creating new /etc/dnsmasq.conf...
> Writing hostapd config file...
> Configuring AP interface...
> Applying iptables rules...
> Starting DNSMASQ server...
> Starting AP on wlan0 in screen terminal...
> Configuration file: /etc/hostapd/hostapd.conf
> wlan0: Could not connect to kernel driver
> Using interface wlan0 with hwaddr b8:27:eb:24:d5:c1 and ssid "vtrust-flash"
> wlan0: interface state UNINITIALIZED->ENABLED
> wlan0: AP-ENABLED 
> 
> ==> ./scripts/smarthack-web.log <==
> Listening on port 80
> 
> ==> ./scripts/smarthack-mqtt.log <==
> 1561755781: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
> 1561755781: Using default config.
> 1561755781: Opening ipv4 listen socket on port 1883.
> 1561755781: Opening ipv6 listen socket on port 1883.

Log files without wpa_supplicant running:
(copy/paste complete with some special characters)

> ==> ./scripts/smarthack-mqtt.log <==
> 
> ==> ./scripts/smarthack-smartconfig.log <==
> 
> ==> ./scripts/smarthack-web.log <==
> 
> ==> ./scripts/smarthack-wifi.log <==
> wlan0: INTERFACE-DISABLED 
> ^Cwlan0: interface state ENABLED->DISABLED
> wlan0: AP-DISABLED 
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> 
> ==> ./scripts/smarthack-web.log <==
> ^CTraceback (most recent call last):
>   File "./fake-registration-server.py", line 129, in <module>
>     main()
>   File "./fake-registration-server.py", line 125, in main
>     tornado.ioloop.IOLoop.current().start()
>   File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
>     self.asyncio_loop.run_forever()
>   File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
>     self._run_once()
>   File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
>     event_list = self._selector.select(timeout)
>   File "/usr/lib/python3.5/selectors.py", line 445, in select
>     fd_event_list = self._epoll.poll(timeout, max_ev)
> KeyboardInterrupt
> 
> ==> ./scripts/smarthack-mqtt.log <==
> ^C1561756084: mosquitto version 1.4.10 terminating
[... 120 lines cut ...]
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> 
> ==> ./scripts/smarthack-web.log <==
> [32m[I 190628 16:09:26 web:2246][m 206 GET /files/upgrade.bin (10.42.42.34) 56.48ms
> [32m[I 190628 16:09:28 web:2246][m 206 GET /files/upgrade.bin (10.42.42.34) 1620.45ms
> 
> ==> ./scripts/smarthack-mqtt.log <==
> 1561756166: Received PUBLISH from 75070011600194b6ad71 (d0, q0, r0, m0, 'smart/device/out/75070011600194b6ad71', ... (80 bytes))
> 1561756167: Received PUBLISH from 75070011600194b6ad71 (d0, q0, r0, m0, 'smart/device/out/75070011600194b6ad71', ... (81 bytes))
> 
> ==> ./scripts/smarthack-wifi.log <==
> wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: associated
> wlan0: AP-STA-CONNECTED 60:01:94:b6:ad:71
> wlan0: STA 60:01:94:b6:ad:71 RADIUS: starting accounting session 20999E1CBA87CCE9
> wlan0: STA 60:01:94:b6:ad:71 WPA: pairwise key handshake completed (RSN)
> 

Prior to doing this the RPi 3b was painstakingly updated following [this guide](http://www.thesmarthomehookup.com/tuya-convert-walk-through-february-2019/).

In answer to the question "how do I recover my WiFi after using an RPi as a `vtrust-flash` server, just reboot.

thwe1966, 2019-07-01:
Hello,

tried again, even with a blank 'wpa_suplicant' -> no success!

Details:
- Device: Teckin smart plug (eu)
   [https://www.amazon.de/gp/product/B07HHFKWJJ/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1](url)
- Raspberry Pi 3 B with Strech lite, completely new
- tuya-convert installed and started with 'start_flash'
   (wlan 'vtrust-flash' is established, connection with a smartphone works)
- Teckin smart-plug is in pairing mode (led is flashing fast)
- After pressing 'Enter' -> 'starting pairing procedure ... waiting for the upgraded device to appear ...'
- Led on the Teckin goes out briefly, then constantly on
- flush program can apparently not appear correctly / completely the device

[image]

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3344075/smarthack-mqtt.log)

[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3344076/smarthack-smartconfig.log)

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3344078/smarthack-web.log)

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3344080/smarthack-wifi.log)

Is a flashing of these devices not possible anymore?

CaptClaude, 2019-07-01:
I now have a WP3 that behaves the same way, although I `sudo killall wpa_supplicant` prior to running `./start_flash.sh`.  I make the WP3 blink fast (blue), start the process, the script runs for a few seconds and the WP3 turns red (so something is happening) but the script just puts dots on the screen and never progresses. The only suspicious thing in any of the log files is in `smarthack-mqtt.log`:

> 1561943563: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
> 1561943563: Using default config.
> 1561943563: Opening ipv4 listen socket on port 1883.
> 1561943563: Error: Address already in use

And I don't really know whether it _**is**_ suspicious.  In the log for a successful flash, I see this:

> 1561754370: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
> 1561754370: Using default config.
> 1561754370: Opening ipv4 listen socket on port 1883.
> 1561754370: Opening ipv6 listen socket on port 1883.
> 1561754498: New connection from 10.42.42.12 on port 1883.
> 1561754498: New client connected from 10.42.42.12 as 05835352dc4f228bf8dc (c1, k60, u'05835352dc4f228bf8dc').
> 1561754498: Sending CONNACK to 05835352dc4f228bf8dc (0, 0)
> 1561754498: Received SUBSCRIBE from 05835352dc4f228bf8dc
> 1561754498:     smart/device/in/05835352dc4f228bf8dc (QoS 0)
> 1561754498: 05835352dc4f228bf8dc 0 smart/device/in/05835352dc4f228bf8dc
> 1561754498: Sending SUBACK to 05835352dc4f228bf8dc

I also connected my smartphone to `vtrust-flash` and will try again with something else connected to it.  There are some other hints in other posts that I saw last night that I will try as well.

kevinduongcalabrio, 2019-07-02:
> Data Point: Bought 4 Gosund WP3s from Amazon in the US just a week ago. After setting up and trying two different Raspberry Pi 3b's and installing `git clone -b create-ap https://github.com/kueblc/tuya-convert.git` on the second Pi I was still not seeing `vtrust-flash`. Finally I killed `wpa_supplicant` and was able to flash all 4 without issue. Unfortunately I appear to have fat-fingered the initial WiFi configuration after flashing and all it would do was slow blink blue/red.
> Because the whole process interested me, I did a careful job of creating empty log files and capturing them with `tail -f` for two attempts, one with wpa_supplicant (failed) and one without wpa_supplicant (which succeeded). Doing so makes it clear exactly what gets written to the log files for each of the two cases. Note that some log files are not written to. The point is to show what successful and unsuccessful AP activation logs contain.
> 
> Logs with wpa_supplicant running:
> 
> > ==> ./scripts/smarthack-mqtt.log <==
> > ==> ./scripts/smarthack-smartconfig.log <==
> > ==> ./scripts/smarthack-web.log <==
> > ==> ./scripts/smarthack-wifi.log <==
> > Backing up NetworkManager.cfg...
> > Restarting NetworkManager...
> > Backing up /etc/dnsmasq.conf...
> > Writing dnsmasq config file...
> > Creating new /etc/dnsmasq.conf...
> > Writing hostapd config file...
> > Configuring AP interface...
> > Applying iptables rules...
> > Starting DNSMASQ server...
> > Starting AP on wlan0 in screen terminal...
> > Configuration file: /etc/hostapd/hostapd.conf
> > wlan0: Could not connect to kernel driver
> > Using interface wlan0 with hwaddr b8:27:eb:24:d5:c1 and ssid "vtrust-flash"
> > wlan0: interface state UNINITIALIZED->ENABLED
> > wlan0: AP-ENABLED
> > ==> ./scripts/smarthack-web.log <==
> > Listening on port 80
> > ==> ./scripts/smarthack-mqtt.log <==
> > 1561755781: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
> > 1561755781: Using default config.
> > 1561755781: Opening ipv4 listen socket on port 1883.
> > 1561755781: Opening ipv6 listen socket on port 1883.
> 
> Log files without wpa_supplicant running:
> (copy/paste complete with some special characters)
> 
> > ==> ./scripts/smarthack-mqtt.log <==
> > ==> ./scripts/smarthack-smartconfig.log <==
> > ==> ./scripts/smarthack-web.log <==
> > ==> ./scripts/smarthack-wifi.log <==
> > wlan0: INTERFACE-DISABLED
> > ^Cwlan0: interface state ENABLED->DISABLED
> > wlan0: AP-DISABLED
> > nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> > ==> ./scripts/smarthack-web.log <==
> > ^CTraceback (most recent call last):
> > File "./fake-registration-server.py", line 129, in 
> > main()
> > File "./fake-registration-server.py", line 125, in main
> > tornado.ioloop.IOLoop.current().start()
> > File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
> > self.asyncio_loop.run_forever()
> > File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
> > self._run_once()
> > File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
> > event_list = self._selector.select(timeout)
> > File "/usr/lib/python3.5/selectors.py", line 445, in select
> > fd_event_list = self._epoll.poll(timeout, max_ev)
> > KeyboardInterrupt
> > ==> ./scripts/smarthack-mqtt.log <==
> > ^C1561756084: mosquitto version 1.4.10 terminating
> > ==> ./scripts/smarthack-wifi.log <==
> > Backing up NetworkManager.cfg...
> > Restarting NetworkManager...
> > Backing up /etc/dnsmasq.conf...
> > Writing dnsmasq config file...
> > Creating new /etc/dnsmasq.conf...
> > Writing hostapd config file...
> > Configuring AP interface...
> > Applying iptables rules...
> > Starting DNSMASQ server...
[... 82 lines cut ...]
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: disassociated
> > ==> ./scripts/smarthack-web.log <==
> > �[32m[I 190628 16:09:26 web:2246]�[m� 206 GET /files/upgrade.bin (10.42.42.34) 56.48ms
> > �[32m[I 190628 16:09:28 web:2246]�[m� 206 GET /files/upgrade.bin (10.42.42.34) 1620.45ms
> > ==> ./scripts/smarthack-mqtt.log <==
> > 1561756166: Received PUBLISH from 75070011600194b6ad71 (d0, q0, r0, m0, 'smart/device/out/75070011600194b6ad71', ... (80 bytes))
> > 1561756167: Received PUBLISH from 75070011600194b6ad71 (d0, q0, r0, m0, 'smart/device/out/75070011600194b6ad71', ... (81 bytes))
> > ==> ./scripts/smarthack-wifi.log <==
> > wlan0: STA 60:01:94:b6:ad:71 IEEE 802.11: associated
> > wlan0: AP-STA-CONNECTED 60:01:94:b6:ad:71
> > wlan0: STA 60:01:94:b6:ad:71 RADIUS: starting accounting session 20999E1CBA87CCE9
> > wlan0: STA 60:01:94:b6:ad:71 WPA: pairwise key handshake completed (RSN)
> 
> Prior to doing this the RPi 3b was painstakingly updated following [this guide](http://www.thesmarthomehookup.com/tuya-convert-walk-through-february-2019/).
> 
> In answer to the question "how do I recover my WiFi after using an RPi as a `vtrust-flash` server, just reboot.

Do you know which firmware version your GoSunds were running on when you received them or are running on now?

CaptClaude, 2019-07-02:
I got an email with the following question from @kevinduongcalabrio whose comment may not yet have percolated to the surface:
>Do you know which firmware version your GoSunds were running on when you received them or are running on now?

Answer: One is running whatever version of Gosund software it came with (I do not have the happyhome app or whatever it is called) and the other three are running the latest Tasmota because they successfully flashed.

wesmininger, 2019-07-06:
I bought a 4 pack of gosund WP3 from amazon July 1 2019
Using the tuya-convert instructions I was able to get one of them flashed. The other 3 are making me go insane with not working. None have eve been connected to any app or wifi and they have only been plugged in while trying to flash

Ive even unplugged my normal router and setup an old one as a totally new network with the pi for AP wired in and still cant get the other 3 to flash.

RpiB3 (not +, tho I tried 2 of those as well)
Rpi lite Stretch image 9 April 2019 release (new buster image was just released so I had to dig for the old version) Ive started with a fresh image nearly every attempt and always done update and upgrades

I can join the vtrust-flash wifi network with another pi3b (to simulate the phone, or whatever) get an ip of 10.42.42.26 with this device.
put the device (ive tried this over with all 3 of the ones i havent managed to flash) in fast blue flash mode.

no matter how long or how fast i press Enter to start the flash process, the blue light flashes for a bit, then turns solid red and stays that way. The light only turns red after I start the flashing process

In the wifi logs I can see that the device joins the AP. 

If I hadnt managed to get 1 of the 4 pack working I would give up expecting that they were maybe updated, but these have never been connected to the internet.

Im about to just chop them open and flash with FTDI.

CaptClaude, 2019-07-08:
@wesmininger Check out this closed issue #207  about opening the WP3 for flashing. My experience was 3 Good, 1 No Flash. I will pry my no-flash open and do the same.  Post your process and results.

graphite-elegance, 2019-08-06:
Has anyone noticed that these aren't using the "tuya" app but now the "Go Smart" app that is specifically for the Gosund switches? Does that mean the Tuya Convert needs a fork for the Gosund "Go Smart" information possibly??

kueblc, 2019-08-06:
From the README
> A Chinese company named Tuya offers a free-to-brand turnkey smart home solution to anyone. Using their offer is dead-simple, since everything can be done by clicking through the Tuya web page, from choosing your pre-designed products or pre-programmed wifi-modules (mostly ESP8266) to building your own app. In the end, this has resulted in as they claim over 11 000 devices 'made' by over 10 000 vendors using Tuyas firmware and cloud services.

Most products have their own brand on the app, it is the same library under the hood.

Tech4Ever, 2019-08-06:
Thanks @kueblc for giving me the hint that there are other with my problem:

Yesterday I received 4 Gosund SP1 EU devices an Ihave the exact same problems reported above:

When I start the flashing procedure, after around 5-10 seconds I can here the device switch and the blinking blue LED turns red or turns off. I don´t get any error or success message on my raspi.

I have no solution, maybe they are not flashable anymore as some others guessed.

TimelessNL, 2019-08-24:
@kueblc In #249 you mentioned that if the device does not send HTTP GET requests it may have the patched firmware. But are there any other signs?

I bough myself some [LSC Smart Connect](https://www.action.com/nl-nl/brand/lsc-smart-connect/) bulbs sold by [Action](https://www.action.com/nl-nl/lscsmartconnect/).
for example this [E14 bulb](https://www.mikrocontroller.net/topic/479796?goto=5950544#5950544) which is based around the ESP8285 and SM2135E.
When running _./start_flash.sh_ it does indeed connect to the AP after the initial smart-config procedure. **But neither does it request a DHCP lease nor is it accessible on its 10.42.42.42 static address.**
Or is there any knowledge if they completely changed the IP address?

_Unfortunately opening the bulb is a one way trip as is seen on the picture below, and even after opening the TX/RX pins are not easily accessible._
[image]

TimelessNL, 2019-08-24:
**Another small update:**
I had to reset my bulb a couple of times and during that process I noticed that if you apply power 4 times it actually starts its own little AP called SmartLife-XXXX where the X's are replaced by the last 4 characters of its MAC address.

Maybe someone else also noticed this as well?

henkiejan1, 2019-08-30:
> @kueblc In #249 you mentioned that if the device does not send HTTP GET requests it may have the patched firmware. But are there any other signs?
> 
> I bough myself some [LSC Smart Connect](https://www.action.com/nl-nl/brand/lsc-smart-connect/) bulbs sold by [Action](https://www.action.com/nl-nl/lscsmartconnect/).
> for example this [E14 bulb](https://www.mikrocontroller.net/topic/479796?goto=5950544#5950544) which is based around the ESP8285 and SM2135E.
> When running _./start_flash.sh_ it does indeed connect to the AP after the initial smart-config procedure. **But neither does it request a DHCP lease nor is it accessible on its 10.42.42.42 static address.**
> Or is there any knowledge if they completely changed the IP address?
> 
> _Unfortunately opening the bulb is a one way trip as is seen on the picture below, and even after opening the TX/RX pins are not easily accessible._
> [image]
> 
> **EDIT:**
> Never mind, after trying an Ubuntu 16.04 install the tuya device did indeed request a DHCP address. No idea why it didn't on 18.04 even while my my phone did get a DHCP lease...

Did you have any process with this? I have buyed a socket but when i start to flash i only get dots on the screen. The led on the socket flash quickly and i connect my smartphone to the vtrust ssid. Should i delete the device from the LSC app? How did you check that the device get a ip?

TimelessNL, 2019-08-30:
Yeah that are the signs. Those dots don't actually mean much. The background process like wifi smartconfig does it's job and gets the bulb to connect to the wifi hotspot. But that's all, then its sending https requests that cannot be intercepted (yet). I took a brief look at the source code from tuya but did not find anything interesting just yet.

Those LSC bubs are produced past 04-2019 so they contain the fix already. And even if they where OTA-able only the filament style bulbs are worth the afford since they don't use the unkown I2C chip to drive the leds like the RGBWW do.

henkiejan1, 2019-08-30:
i has first buyed the socket, pir sensor and a remote control (never seen a remote control with a wifi chip. It´s a realtek one. But also has upgraded the socket with the latest firmware. So i think i could never flash it.

TimelessNL, 2019-08-31:
@henkiejan1 please visit [this forum](https://www.mikrocontroller.net/topic/479796) this will give you some more information about these LCS products. If you don't speak German you could use google translate.

idxman01, 2019-09-05:
it sounds like this is related to firmware, but in case it helps:

I flashed a gosund WP3 this week loaded with 1.04.  All 3 were previously connected to the Tuya app instead of smartlife or the new gosund app..  Thankfully it didn't upgrade firmware.

it wasn't smooth and I had to restart it several times...  I mainly needed to stop _and_ disable systemd-resolve, several other services, kill previous SCREEN processes and run the following bits from scripts/setup_ap.sh which were left hanging...

I'm about to flash the other two and will send any other quirks that crop up.

Manual cleanup:

```shell
if test -d /etc/NetworkManager; then
        sudo rm /etc/NetworkManager/NetworkManager.conf > /dev/null 2>&1
        sudo mv /etc/NetworkManager/NetworkManager.conf.backup /etc/NetworkManager/NetworkManager.conf
        sudo service network-manager restart
fi
sudo /etc/init.d/dnsmasq stop > /dev/null 2>&1
sudo pkill dnsmasq
sudo rm /etc/dnsmasq.conf > /dev/null 2>&1
sudo mv /etc/dnsmasq.conf.backup /etc/dnsmasq.conf > /dev/null 2>&1
sudo rm /etc/dnsmasq.hosts > /dev/null 2>&1
sudo iptables --flush
sudo iptables --flush -t nat
sudo iptables --delete-chain
sudo iptables --table nat --delete-chain
```

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc

daninfuchs, 2020-05-28:
I know this is (partially) irrelevant, but I wanted to mention that I just the other day (~May 20 2020) ordered a four-pack of WP3 switches from Amazon ( https://www.amazon.com/Gosund-Compatible-Required-appliances-Certified/dp/B079MFTYMV/ ) 

Straight out of the box I installed Tuya-convert (git pull xyz etc) and could not flash any of the four UNTIL applying the fix mentioned here by user @jfractalj;

> Yeah, the vtrust-flash network doesn't route out to the internet. They are routing out through their mobile networks to get there. For me, the process did not work because dnsmasq wasn't working [...]

After the fix, it worked on both units I tested, BUT it was unable to pull the full firmware due to a VERY VERY slow download kicking a timeout (after three runs through the script; each time, it reached ~50% in spits and spurts of a few hundred bytes at a time) -- after which I finally gave up and told it to flash Tasmota, but I do not have the original binary backed up anymore. (I don't care; if aftermarket firmware doesn't run it's not getting used.)

From there it seems to have worked fine. I have four in the box; I'll flash two now, if anyone can tell me how to *safely* check which firmware version they're on I'll do that next.

[19 more replies]