ct-Open-Source/tuya-convert #200: OTA Flash not work for Fcmila Gu10 RGBW

fpries, 2019-05-21:
Hello together,

the issue from my side, the device has the IP 10.42.42.**12** and did not get the ip 10.42.42.**42**, what maybe the result that the script can't  find the device and makes without stopping dot after dot …...

But step by step:

1. What i buy on JOOM "Belive-Gu10-RGBW-WiFi bulb".
http://www.joom.com/de/products/5cb7c9ea28fc710101dd83cd

2. What i got was the following "Fcmila Smart Spotlight".
[image]
  picture_1: the package box
[image]
  picture_2: the package box - socket Gu10 - 220V - multicolor
[image]
  picture_3: the package box - China lable
[image]
  picture_4: device - smart bulb RGBW 6W
[image]
  picture_5: device - LED board - FL-M4 V2.4
[image]
  picture_6: device - supply board
[image]
  picture_7: device - overview
[image]
  picture_8: assimilation mit supply

3. I tried to flash the device with a Raspberry Pi 3B+. 
    device is in quick flashing mode 
[image]
  picture_9: rpi_tuya_convert

4. I could collect the following log details:
4.a file mqtt.log
[20190521_smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3203839/20190521_smarthack-mqtt.log)
4.b file smartconfig.log
[20190521_smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3203844/20190521_smarthack-smartconfig.log)
4.c file web.log
[20190521_smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3203850/20190521_smarthack-web.log)
4.d file WiFi.log
[20190521_smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3203851/20190521_smarthack-wifi.log)

5. WirelessNetworkMonitor
[image]
  picture_10: WN-Monitor

This showes that the device get IP 10.42.42.**12**

6. SmartLife-8E2A as a Access Point 
    device in slow flashing mode  
[image]
  picture_11: device in AP-Mode

I hope i give the right infos, someone can help to understood, what is going wrong and what could be the next steps ... 

best regards,
fpries

probonopd, 2019-05-21:
Have you connected the light bulb to the app/the internet before flashing?

fpries, 2019-05-21:
i did not use a smartlife app before, should i?

probonopd, 2019-05-21:
No. Beware. Connecting the bulb to the app or to the internet will likely update its firmware and make it unflashable.

dicko72, 2019-06-04:
Any update flashing these fcmila bulbs. I have the same issue.

fpries, 2019-06-06:
Hello dicko72:

No, Sorry. At the moment it looks like that no one can help us with this bulbs.

The only thing i can report, that i try to do this:   
(https://community.openhab.org/t/step-by-step-guide-for-adding-tuya-bulbs-wi-fi-smart-led-smart-life-app-to-oh2-using-tuya-mqtt-js-by-agentk/59371)
But in that case, i do not get/find the tuya local key.

kueblc, 2019-06-11:
> the issue from my side, the device has the IP 10.42.42.**12** and did not get the ip 10.42.42.**42**, what maybe the result that the script can't find the device and makes without stopping dot after dot …...

This means your device is not running the intermediate firmware; which as others stated is likely due to your firmware being patched. If you successfully install the intermediate firmware, the device will take `10.42.42.42`. In the meantime, stay tuned until we can release a workaround.

closed 2019-06-11 by kueblc