ct-Open-Source/tuya-convert #201: Unable to flash Vansware Smart Plug 16A

jokerigno, 2019-05-22:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

I followed the guide step by step but I'm unable to flash the smart plug " Vansware Smart Plug 16A"

Here are the logs:

**smarthack-smartconfig.log**

> Put Device in Learn Mode! Sending SmartConfig Packets now
> Sending SSID                  vtrust-flash
> Sending wifiPassword          flashmeifyoucan
> SmartConfig in progress
> ..........
> SmartConfig complete.

**smarthack-wifi.log**

> wlan0: STA c4:0b:cb:b1:86:41 IEEE 802.11: associated
> wlan0: AP-STA-CONNECTED c4:0b:cb:b1:86:41
> wlan0: STA c4:0b:cb:b1:86:41 RADIUS: starting accounting session 328ABA38D4C3C2BA
> wlan0: STA c4:0b:cb:b1:86:41 WPA: pairwise key handshake completed (RSN)
> wlan0: STA ec:fa:bc:2e:05:ef IEEE 802.11: associated
> wlan0: AP-STA-CONNECTED ec:fa:bc:2e:05:ef
> wlan0: STA ec:fa:bc:2e:05:ef RADIUS: starting accounting session 6933A25181C63499
> wlan0: STA ec:fa:bc:2e:05:ef WPA: pairwise key handshake completed (RSN)
> wlan0: STA ec:fa:bc:2e:05:ef IEEE 802.11: disassociated
> wlan0: AP-STA-DISCONNECTED ec:fa:bc:2e:05:ef
> wlan0: STA ec:fa:bc:2e:05:ef IEEE 802.11: associated
> wlan0: STA ec:fa:bc:2e:05:ef IEEE 802.11: disassociated
> wlan0: STA ec:fa:bc:2e:05:ef IEEE 802.11: disassociated
> wlan0: STA ec:fa:bc:2e:05:ef IEEE 802.11: associated
> wlan0: AP-STA-CONNECTED ec:fa:bc:2e:05:ef
> wlan0: STA ec:fa:bc:2e:05:ef RADIUS: starting accounting session 593B237115FD445A
> wlan0: STA ec:fa:bc:2e:05:ef WPA: pairwise key handshake completed (RSN)

**smarthack-web.log**

> [32m[I 190522 23:26:33 web:2246][m 200 GET / (10.42.42.28) 6.71ms
> 
> 
> URI:/gw.json?a=s.gw.token.get&gwId=04657807ecfabc2e05ef&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":6,"source":"ap","path":"broadcast"}}&t=7&v=3.0&sign=fc6231094170cd0bc90432401f73a475
> Answer s.gw.token.get
> [32m[I 190522 23:26:51 web:2246][m 200 POST /gw.json?a=s.gw.token.get&gwId=04657807ecfabc2e05ef&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":6,"source":"ap","path":"broadcast"}}&t=7&v=3.0&sign=fc6231094170cd0bc90432401f73a475 (10.42.42.12) 5.22ms
> 
> 
> URI:/gw.json?a=s.gw.dev.pk.active&gwId=04657807ecfabc2e05ef&other={"token":"00000000"}&t=7&v=3.0&sign=67668049ec8d1b07c3db954f8bd61758
> Answer s.gw.dev.pk.active
> READ GW ID 04657807ecfabc2e05ef
> TRIGGER UPGRADE IN 10 SECONDS
> Trigger upgrade in 10 seconds
> [32m[I 190522 23:26:53 web:2246][m 200 POST /gw.json?a=s.gw.dev.pk.active&gwId=04657807ecfabc2e05ef&other={"token":"00000000"}&t=7&v=3.0&sign=67668049ec8d1b07c3db954f8bd61758 (10.42.42.12) 69.01ms
> 4f79211dd2ebbc3bbed6fe3d8655d626
> b'2.1d2ebbc3bbed6fe3dt5inCMDHpcJKQpxfOe2RbRQqa/FbZKiZbDHIeCgoHdMDbX+JetHVB1suC/DvI8xUJsm5uMafEvnzLLz/1MQ4v8ChQQPUmyAJ9TPx4QM0ANXdg4SXZL9kLEC9aqKXAOwI'
> 
> 
> URI:/gw.json?a=s.gw.update&gwId=04657807ecfabc2e05ef&t=11&v=2.0&sign=06a1556271d4bdf97360b2407bd2dd34
> Answer s.gw.update
> [32m[I 190522 23:27:05 web:2246][m 200 POST /gw.json?a=s.gw.update&gwId=04657807ecfabc2e05ef&t=11&v=2.0&sign=06a1556271d4bdf97360b2407bd2dd34 (10.42.42.12) 7.38ms
> 
> 
> URI:/gw.json?a=s.gw.dev.update&gwId=04657807ecfabc2e05ef&t=7&v=2.0&sign=bca3b442129358ea91aaff3c9267a134
> Answer s.gw.update
> [32m[I 190522 23:27:05 web:2246][m 200 POST /gw.json?a=s.gw.dev.update&gwId=04657807ecfabc2e05ef&t=7&v=2.0&sign=bca3b442129358ea91aaff3c9267a134 (10.42.42.12) 10.80ms
> 
> 
> URI:/gw.json?a=atop.online.debug.log&gwId=04657807ecfabc2e05ef&t=10&sign=3cf48cecd0ab66c275631e45c183b58f
> Answer atop.online.debug.log
> [32m[I 190522 23:27:09 web:2246][m 200 POST /gw.json?a=atop.online.debug.log&gwId=04657807ecfabc2e05ef&t=10&sign=3cf48cecd0ab66c275631e45c183b58f (10.42.42.12) 15.43ms
> 
> 
> URI:/gw.json?a=tuya.device.upgrade.silent.get&gwId=04657807ecfabc2e05ef&t=7&v=4.1&sign=cb6b54ee05cdf4e1aa2a36cd0500c51e
> Answer s.gw.upgrade
> b'{"result":{"auto":3,"fileSize":"506587","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"dc92883ebc44f64253bfb004433fc929"},"t":100,"e":false,"success":true}'
> [32m[I 190522 23:27:09 web:2246][m 200 POST /gw.json?a=tuya.device.upgrade.silent.get&gwId=04657807ecfabc2e05ef&t=7&v=4.1&sign=cb6b54ee05cdf4e1aa2a36cd0500c51e (10.42.42.12) 14.90ms
> 
> 
> URI:/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=275&sign=6719cc8f3ea55db5b64f09cfaa65ddf2
> WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=275&sign=6719cc8f3ea55db5b64f09cfaa65ddf2)
> [32m[I 190522 23:30:04 web:2246][m 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=275&sign=6719cc8f3ea55db5b64f09cfaa65ddf2 (10.42.42.12) 16.18ms
> 
> 
> URI:/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=290&sign=45f5021ac01494818196594fbae6cfbf
> WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=290&sign=45f5021ac01494818196594fbae6cfbf)
> [32m[I 190522 23:30:19 web:2246][m 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=290&sign=45f5021ac01494818196594fbae6cfbf (10.42.42.12) 7.84ms
> 
> 
> URI:/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=305&sign=d8f84312526f98eb3a18aef8964c0512
> WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=305&sign=d8f84312526f98eb3a18aef8964c0512)
> [32m[I 190522 23:30:34 web:2246][m 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=305&sign=d8f84312526f98eb3a18aef8964c0512 (10.42.42.12) 9.35ms
> 
> 
> URI:/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=320&sign=42d0909a0baaae64e22bc507e27a79c0
> WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=320&sign=42d0909a0baaae64e22bc507e27a79c0)
> [32m[I 190522 23:30:49 web:2246][m 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=04657807ecfabc2e05ef&t=320&sign=42d0909a0baaae64e22bc507e27a79c0 (10.42.42.12) 14.39ms
> 

**smarthack-mqtt.log**

> 1558564015: New connection from 10.42.42.12 on port 1883.
> 1558564015: New client connected from 10.42.42.12 as 04657807ecfabc2e05ef (c1, k30, u'04657807ecfabc2e05ef').
> 1558564015: Sending CONNACK to 04657807ecfabc2e05ef (0, 0)
> 1558564015: Received SUBSCRIBE from 04657807ecfabc2e05ef
> 1558564015:     smart/device/in/04657807ecfabc2e05ef (QoS 0)
> 1558564015: 04657807ecfabc2e05ef 0 smart/device/in/04657807ecfabc2e05ef
> 1558564015: Sending SUBACK to 04657807ecfabc2e05ef
> 1558564023: New connection from 127.0.0.1 on port 1883.
> 1558564023: New client connected from 127.0.0.1 as P1 (c1, k60).
> 1558564023: Sending CONNACK to P1 (0, 0)
> 1558564023: Received PUBLISH from P1 (d0, q0, r0, m0, 'smart/device/in/04657807ecfabc2e05ef', ... (147 bytes))
> 1558564023: Sending PUBLISH to 04657807ecfabc2e05ef (d0, q0, r0, m0, 'smart/device/in/04657807ecfabc2e05ef', ... (147 bytes))
> 1558564023: Socket error on client P1, disconnecting.
> 1558564024: Socket error on client 04657807ecfabc2e05ef, disconnecting.
> 1558564024: New connection from 10.42.42.12 on port 1883.
> 1558564024: New client connected from 10.42.42.12 as 04657807ecfabc2e05ef (c1, k30, u'04657807ecfabc2e05ef').
> 1558564024: Sending CONNACK to 04657807ecfabc2e05ef (0, 0)
> 1558564024: Received SUBSCRIBE from 04657807ecfabc2e05ef
> 1558564024:     smart/device/in/04657807ecfabc2e05ef (QoS 0)
> 1558564024: 04657807ecfabc2e05ef 0 smart/device/in/04657807ecfabc2e05ef
> 1558564024: Sending SUBACK to 04657807ecfabc2e05ef

I'm stuck at this point:

**Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!**

sebastianklein96, 2019-05-23:
You seem to have firmware loaded on your device that has been patched against the exploit Tuya Convert uses to flash. Have you updated your device's firmware  / connected it to the internet before / connected it to any sort of app before trying to flash with Tuya Convert?

You can get more information in these issues since they all deal with the same issue:
https://github.com/ct-Open-Source/tuya-convert/issues/160
https://github.com/ct-Open-Source/tuya-convert/issues/197
https://github.com/ct-Open-Source/tuya-convert/issues/199

jokerigno, 2019-05-23:
Nope. I just plugged in and put in configuration mode. I already am aware of this issue (I'm following #160) but was hoping that this devices were still usable.

kueblc, 2019-07-02:
Hi @jokerigno,

Actually taking a closer look at your logs now, I see that the device did connect and make HTTP requests. It should still be possible to flash this device OTA if you are still interested.

jokerigno, 2019-07-03:
Yes I'm interested! What do you suggest? To try with the default procedure or I have to make some change?

kueblc, 2019-07-03:
Try the `enhance-api` branch with `git clone -b enhance-api https://github.com/kueblc/tuya-convert.git`

jokerigno, 2019-07-09:
I will try and let you know.

jokerigno, 2019-07-13:
It worked!

Please noticed that I had to change also the smartphone used to connect to wifi. Don't ask me why but my mi6 did't the job so i used my tablet.

closed 2019-07-13 by jokerigno

kueblc, 2019-07-15:
Awesome, thank you for testing @jokerigno

vigne, 2019-08-09:
@kueblc FYI, your branch was also the only way to update the Luminea NX-4462-675 which is listed under supported devices. Main brach failed also with " Socket error on client P1, disconnecting." like @jokerigno 's

jeanfrancoissantos, 2019-09-17:
Hi! This branch is the only way to flash Glode LED Bullbs from Costco (CA). Thanks!

kueblc, 2019-09-17:
Great, thank you all for the feedback. I will try to get this into the main branch shortly.

Jujonet, 2020-07-15:
Maybe this is such a huge refloat, but I couldn’t be more grateful.
With kueblc enhanced branch I could flash a couple of Teckin SP22 I was suspecting could have been connected to internet before as refundé to flash with ruta-convert (I got them 2nd hand) and also a Blitzwolf SHP8 I thought came bricked from China....

woomiez, 2021-12-16:
I've got a few of the Vansware plugs as well, but this branch doesn't seem to work for me. Is it possible that it's not fully compatible with the current Raspbian (bullseye) distro? I don't see the vtrust-flash wifi appear, and there's a not very helpful error from the network driver in the log (RTNETLINK: file exists). It also seems to rely on iptables which isn't in Raspbian anymore by default (but i installed it separately), and python seems to be missing its "crypto" module, even though I apt'ed that as well already.