ct-Open-Source/tuya-convert #202: Unable to flash Teckin SP21

DBa2016, 2019-05-25:
Trying to flash factory-fresh Teckin SP21 (got a pack of 4 from Amazon.de)...

After starting "./start_flash.sh", and confirming everything (yes, my ipad connects to vtrust-flash), the flash process basically stalls.

The relevant log file snippets:
```
==> scripts/smarthack-wifi.log <==
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:29:60:39 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

==> scripts/smarthack-web.log <==
Listening on port 80

==> scripts/smarthack-mqtt.log <==
1558749874: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1558749874: Using default config.
1558749874: Opening ipv4 listen socket on port 1883.
1558749874: Opening ipv6 listen socket on port 1883.

==> scripts/smarthack-wifi.log <==
wlan0: STA <won't tell>a IEEE 802.11: associated
wlan0: AP-STA-CONNECTED <won't tell>
wlan0: STA <won't tell> RADIUS: starting accounting session 536267C8330F22CD
wlan0: STA <won't tell> WPA: pairwise key handshake completed (RSN)

==> scripts/smarthack-smartconfig.log <==
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.

==> scripts/smarthack-wifi.log <==
wlan0: STA dc:4f:22:5a:be:eb IEEE 802.11: associated
wlan0: AP-STA-CONNECTED dc:4f:22:5a:be:eb
wlan0: STA dc:4f:22:5a:be:eb RADIUS: starting accounting session 026314598D6512EC
wlan0: STA dc:4f:22:5a:be:eb WPA: pairwise key handshake completed (RSN)

```

I connected one of these 4 devices to Smart Life app and see MCU and Wifi FW version of 1.0.2 both.

Already flashed 4 smart plugs (from different manufacturers), just cannot get the Teckins running.

Any suggestions?

sebastianklein96, 2019-05-25:
Most likely the firware is too recent on those devices out of the box, which would mean that they cannot be flashed with Tuya Convert at the moment.

The firmware numbers for different devices of the same manufacturer don't always mean the same. So if you expect to be able to flash those SP21's because they have a lower firmware version than for example the SP22's that I reported not flashable with 1.0.7, you might be mistaken.

The one connected to the Tuya app will now have the newest firmware for sure and therefore definitely cannot be flashed anymore.

Haschmaradi, 2019-06-11:
I read some place that one should not connect an iOS device to vtrust-flash, because then flashing may not work properly. I connected vtrust-flash to an iMac and could flash all 4 SP21 with no problem. I got them from Amazon about 6 weeks ago. I could even flash the one that was already connected to the Tuya app, and used for 5 weeks, without any problems....

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc