ct-Open-Source/tuya-convert #204: Unable to flash Costco/Feit A19 bulb

cwhits, 2019-05-28:
**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
Feit BPA800 bulb from Costco.

The bulb flashes quickly (EZ mode) then turns solid once the SmartConfig packets are sent. It connects to vtrust-flash but nothing happens after that.

::0e is my phone, and ::dd is the light bulb.
```
==> scripts/smarthack-wifi.log <==
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlan0 with hwaddr 80:1f:02:94:3c:62 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 58:cb:52:46:38:0e IEEE 802.11: authenticated
wlan0: STA 58:cb:52:46:38:0e IEEE 802.11: associated (aid 1)
wlan0: AP-STA-CONNECTED 58:cb:52:46:38:0e
wlan0: STA 58:cb:52:46:38:0e RADIUS: starting accounting session 94377C8B439AFE73
wlan0: STA 58:cb:52:46:38:0e WPA: pairwise key handshake completed (RSN)

==> scripts/smarthack-web.log <==
Listening on port 80

==> scripts/smarthack-mqtt.log <==
1559086465: mosquitto version 1.5.7 starting
1559086465: Using default config.
1559086465: Opening ipv4 listen socket on port 1883.
1559086465: Opening ipv6 listen socket on port 1883.

==> scripts/smarthack-smartconfig.log <==
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.

==> scripts/smarthack-wifi.log <==
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: authenticated
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: associated (aid 2)
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: authenticated
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: associated (aid 2)
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH b4:e6:2d:67:21:dd
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: authenticated
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: associated (aid 2)
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH b4:e6:2d:67:21:dd
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: deauthenticated due to local deauth request
Station b4:e6:2d:67:21:dd trying to disassociate, but it is not associated
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: authenticated
wlan0: STA b4:e6:2d:67:21:dd IEEE 802.11: associated (aid 2)
wlan0: AP-STA-CONNECTED b4:e6:2d:67:21:dd
wlan0: STA b4:e6:2d:67:21:dd RADIUS: starting accounting session 399DD934DA590355
wlan0: STA b4:e6:2d:67:21:dd WPA: pairwise key handshake completed (RSN)
```

cwhits, 2019-05-29:
it shows up in my arp table as 10.42.42.11 but I can't connect to it:
```root@kali:~/tuya-convert# arp -a
? (10.42.42.29) at 58:cb:52:46:38:0e [ether] on wlan0
? (10.42.42.11) at b4:e6:2d:67:3f:99 [ether] on wlan0
? (10.42.42.42) at <incomplete> on wlan0
_gateway (10.0.2.2) at 52:54:00:12:35:02 [ether] on eth0
root@kali:~/tuya-convert# curl http://10.42.42.11
curl: (7) Failed to connect to 10.42.42.11 port 80: Connection refused
```

cwhits, 2019-05-29:
running it through wireshark I can see that it's trying to connect to https://a3.tuyaus.com

RoulstoneR, 2019-06-05:
Hi, did you get to the bottom of this? I'm having the same issue.

cwhits, 2019-06-05:
yep! they have patched firmware and don't work with Tuya Convert any more.

RoulstoneR, 2019-06-05:
I did wonder if that was the issue. I'm assuming you haven't got a workaround? look like I need to return them.

e2m32, 2019-06-14:
I bought 5 at the local Menards. 4 worked like a charm, one appears to have the updated firmware.

kitchung, 2019-07-21:
Numbers under the wifi logo may be production date codes. 228-19-14 doesn't work for me. Can you'all post the codes for your working bulbs?

StealthChesnut, 2019-08-04:
228-28-42 doesn't work, model BPA800/RGBW/AG/2/UK.

I have managed to open the bulb (gently prying around the globe, it appears to be just stuck on with an adhesive), remove the LED board and then start tracing the pads on the ESP board. Has anyone found out what all 7 pads are?

cwhits, 2019-08-04:
I found a schematic but it didn't look like the board had a tx/rx on it

StealthChesnut, 2019-08-04:
On my bulb, the ESP board has 7 pads on the back. Holding the bulb with the light socket fitting at the bottom, the four vertical pads are (top to bottom) GND, 3.3V, ?, Pin15 GPIO0. 
There are 2 pads close to each other on a diagonal, top-right is Pin 25, GPIO3 RX and bottom-left is Pin 26 GPIO1 TX. There's also another pad on it's own, but I don't know what that is either.

The pins above are the pins on the ESP8266 chip. I checked these by continuity testing with a multimeter and needle probes, but that's not to say they'll be the same on every other bulb.

alexyao2015, 2019-08-05:
How hard was it to disassemble the bulb itself? I read that there is a resin that the entire bulb is potted in. Does it look like [this](https://fccid.io/SYW-A19RGBWAGT1/Internal-Photos/Internal-Photos-4018527)? Photo 2 shows the resin that I am afraid of and that looks to be really hard to remove. From my understanding, all these bulbs use [this](https://docs.tuya.com/en/hardware/WiFi-module/wifi-lc2v-module.html) module.

StealthChesnut, 2019-08-05:
Great link. The plastic dome on top can be pryed off, gently. It does take some force, but there are no catches or clips, just a soft glue.

The bulb is potted in resin as shown in photo 2. I don't think I can get the boards out of there, but I will be attaching wires to the pads shown in photo 10 with the boards in-situ. Rotate that photo 90 degrees clockwise to match the orientation I am working in.

alexyao2015, 2019-08-05:
So if you found the pins necessary to flash the board, were you able to successfully do it over UART? Any issues?

StealthChesnut, 2019-08-05:
Yes, successfully flashed. This thread also has some pictures and success and they concur the boards are likely similar: https://github.com/ct-Open-Source/tuya-convert/issues/99 

Step by step guide to flashing here: https://community.blynk.cc/t/how-to-backup-restore-official-firmware-on-any-espressif-esp8266-esp32/34309

Flashed the standard latest Tasmota (6.6.0) .bin using esptool. Hard reset between each esptool operation by unplugging/power off (so read-flash, reset, erase, reset, write-flash). I haven't reassembled yet, but the device is on my WLAN.

cwhits, 2019-08-05:
@StealthChesnut was your bulb not filled with epoxy?

alexyao2015, 2019-08-05:
Haven't opened mine up yet but from what I've read is that it's more of a rubbery resin that doesn't cover up the pads.

#99 says that they had trouble powering the 3.3v from the pads and had to connect directly to mains. Did you have that same problem? @StealthChesnut

StealthChesnut, 2019-08-05:
@cwhits There was some epoxy, but not enough to cover the pads. Most of the white  circuit board was uncovered. 

@alexyao2015 I was able to power from the pads. It's very tight for space though and I checked my connections for shorts before powering up.

I got the bulb open by prying with thin bladed screwdrivers, working my way around gently. It's mostly the glue holding the dome on. If you're careful it should come off if you just keep working round slowly.

I've been able to use the tasmota template but it only shows sliders for brightness and white temperature so far.

alexyao2015, 2019-08-05:
So [this](https://blakadder.github.io/templates/feit_rgb_a19.html) template didn't work for color? Maybe try going through each gpio one by one.

cwhits, 2019-08-06:
also don't forget:
>Note BGR color order - setoption37 54 in tasmota console to remap.

StealthChesnut, 2019-08-06:
I used that template, but on rebooting the web gui only showed two sliders,
one for colour temp and one for brightness. I ran out of time but will look
further into it and try the console command.

kueblc, 2019-08-06:
The Sonoff-Tasmota web GUI doesn't offer a color control, whether or not the device supports it. Control the color through MQTT or console.

alexyao2015, 2019-08-10:
I don't know if I'm totally missing something, but I removed the top cap but can't seem to figure out how to remove the led plate. There are no screws on mine like I saw in the picture. Anyone have tips/help?
[image]

alexyao2015, 2019-08-11:
Well I finally pried the plate off and suffice it to say they really don't want you to get in there.
[image]

StealthChesnut, 2019-08-11:
That's a completely different LED layout to mine, and what happened with
the boards?! Not sure how you'd go about getting that off other than
complete disassembly.

Is that a Costco/ Felt bulb or another brand?

alexyao2015, 2019-08-11:
So I purchased the 100 watt version from my local Menards. Same feit brand and same price and almost the same model number as the 60 watt version. Appears that the inside is completely different. Finally got the plate off without damaging anything and discovered that the module is also different and covered in a playdough like putty. Luckily it was moldable so I could scoop it out. The module looks like an esp12e so I think I'll have to solder wires to get it to flash.

alexyao2015, 2019-08-13:
So after a lot of pain, I got tasmota on it and everything lights up expect for the cool LEDs. So either I broke something while soldering the wires or I can't find the GPIO pins. I can choose to either flash back the tuya firmware to see if the hardware itself was damaged but if it wasn't I have to go through the trouble of resoldering and flashing again.
Any ideas on how to find the GPIO pins so I can avoid having to go through it again?

rlowens, 2019-09-08:
> On my bulb, the ESP board has 7 pads on the back. Holding the bulb with the light socket fitting at the bottom, the four vertical pads are (top to bottom) GND, 3.3V, ?, Pin15 GPIO0.
> There are 2 pads [...]

Here's the test pads labeled to be clear on the TYLC2V module
[image]
(I used these 2 images from Tuya and played with contrast to trace the pins, and they agree with StealthChesnut meter-testing. https://images.tuyacn.com/smart_res/developer_default/TYLC2V-1.png https://images.tuyacn.com/smart_res/developer_default/TYLC2V-2.png )

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc

DSuds, 2019-10-03:
OTA flashing works with the latest branch and my bulbs that have an FCC id ending in GT1 rather than GT1.  I read somewhere they changed driver chips from the SM16726P but I'll take a peek as I try to get a template working.

goodfore, 2019-10-03:
@DSuds Thats good news. Can you confirm that latest branch works for Feit bulbs? I was following the other thread, it seems to be referencing another bulb. Also, my bulb's FCC id also ends with GT1.

cwhits, 2019-10-03:
@goodfore I bought fiet bulbs from Costco yesterday and it worked

DSuds, 2019-10-03:
Well I can confirm both model 1 and 2 can be flashed. The GT2 is significantly different inside however with a SM726EC.

kueblc, 2019-10-03:
SM726EB (used by Merkury Innovations Color A21 Bulb) is protocol compatible with the SM16716. Haven't heard of the SM726EC but I'd guess that it is also compatible.

DSuds, 2019-10-03:
Probably fine but I dove in depth first here... now that I have disassembled bulbs I was serial flashing and now OTA flashing I should probably figure out how to control them!
Looks like the pin out may be different at they don't seem to respond the same as the first version.
GT2 LED board:
[image]

kueblc, 2019-10-03:
Using Tasmota? Try module SYF05 and then set GPIO13 to SM16716 PWR.

mhightower83, 2019-10-03:
@DSuds There is a similar discussion on the SYM-A19RGBWAGT2 here 
https://github.com/arendst/Sonoff-Tasmota/issues/6534#issuecomment-537771525
@kueblc They have moved around CLK(GPIO14) and DAT(GPIO12). PWR is still on GPIO13. Both GPIO4 and GPIO5 are now PWMs. They also changed the way the two PWM channels handle the Warm/Cold/Brightness. One PWM only controls the blend between Cold and Warm LEDs. The other PWN controls the brightness. Do you have any idea what Tasmota options fit this usage?

DSuds, 2019-10-03:
There was a fair bit of work that went into handling the two whites and rgb levels so there may need to be some similar work in tasmota to manage this new control style. I'll dig in when back at the computer! I wish there was an easier way to tell the models apart though - my eyes aren't that great anymore!

jtmoderate876, 2019-11-22:
> that have an FCC id ending in GT1 rather than GT1. I read somewhere they changed driver chips from t

@DSuds What did you mean by using GT1 twice in "ending it GT1 rather than GT1"