ct-Open-Source/tuya-convert #205: Digoo SP202 nothing happens

Klagio, 2019-05-29:
Nothing happens....: 

I connect my phone to the newly created vtrust-flash AP, I set the plug to start flashing, I press Enter.... and all I get are dots. Below are the log files.

smarthack web
```

Traceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 123, in main
    app.listen(options.port)
  File "/usr/local/lib/python3.5/dist-packages/tornado/web.py", line 2112, in listen
    server.listen(port, address)
  File "/usr/local/lib/python3.5/dist-packages/tornado/tcpserver.py", line 151, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python3.5/dist-packages/tornado/netutil.py", line 174, in bind_sockets
    sock.bind(sockaddr)
OSError: [Errno 98] Address already in use
Traceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 123, in main
    app.listen(options.port)
  File "/usr/local/lib/python3.5/dist-packages/tornado/web.py", line 2112, in listen
    server.listen(port, address)
  File "/usr/local/lib/python3.5/dist-packages/tornado/tcpserver.py", line 151, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python3.5/dist-packages/tornado/netutil.py", line 174, in bind_sockets
    sock.bind(sockaddr)
OSError: [Errno 98] Address already in use

```
```

Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:9b:b6:98 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 4c:49:e3:52:fb:88 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 4c:49:e3:52:fb:88
wlan0: STA 4c:49:e3:52:fb:88 RADIUS: starting accounting session 616DA8C976FB8AE2
wlan0: STA 4c:49:e3:52:fb:88 WPA: pairwise key handshake completed (RSN)
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 4c:49:e3:52:fb:88
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:9b:b6:98 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 4c:49:e3:52:fb:88 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 4c:49:e3:52:fb:88
wlan0: STA 4c:49:e3:52:fb:88 RADIUS: starting accounting session 6C0E00A7381A57A2
wlan0: STA 4c:49:e3:52:fb:88 WPA: pairwise key handshake completed (RSN)

```

```

Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.

```

ezraholm50, 2019-09-02:
exactly the same here. Ubuntu with Wifi dongle.

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc