ct-Open-Source/tuya-convert #206: Cannot flash Zoozee SA102

ltagliamonte, 2019-05-29:
Great project guys!!
I got some Zoozee SA102 from Amazon and I'd like to flash the Tasmota firmware on them.

I see from this page https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices that they should be supported.

I'm using my rasp-py the guest wifi gets created and I can successfully connect my phone but the smart plug doesn't connect to it, I can see it using: `screen -r smarthack-wifi`

I'm able to connect the plug to the smart life app, i also tried the factory reset from the app but it didn't work as well.
Is it possible that they changed the wifi network name or pwd the plug is looking for? Can't find any other info on the web.

sebastianklein96, 2019-05-29:
> I'm able to connect the plug to the smart life app, i also tried the factory reset from the app but it didn't work as well.

That's most likely the problem. By connecting the plug to the app you gave it internet access, which in return enables the plug to update its firmware. As you can see in most of the recent issues, updating the firmware means you cannot use Tuya Convert on the plug.

ltagliamonte, 2019-05-29:
i tried without connecting the plug to smart life but it didn't work.
I have other plugs not used i'm testing those.

ltagliamonte, 2019-05-30:
I successfully flashed my outlets.
After a long debugging session and compiling and reinstalling python 3.7.2 with openssl on my rasp-pi I was able to use the tool.
Since python 3 isn't still installed by default and for some distro isn't packaged either I would suggest to try to run the tornato webserver in docker container; it will make this tool way way more easy to use for noobs.
Teckin US preset works.

closed 2019-05-30 by ltagliamonte

nemesis03, 2019-06-03:
I am currently working on a similar problem. I have two Koogeek KLSP2 (which look similar to BlitzWolf SHP5). The case is glued so I cannot easily open them. I use a raspberry pi zero as server.
Flashing with tuya-convert failed, because these plugs already have a new firmware and try to communicate with HTTPS (checked with tcpdump and wireshark)

I activated HTTPS in the fake-registration-server.py by changing the main function:
```
def main():
    parse_command_line()
    app = tornado.web.Application(
        [
            (r"/", MainHandler),
            (r"/gw.json", JSONHandler),
            ('/files/(.*)', FilesHandler, {'path': str('../files/')}),
        ],
        #template_path=os.path.join(os.path.dirname(__file__), "templates"),
        #static_path=os.path.join(os.path.dirname(__file__), "static"),
        debug=options.debug,
    )
    ssl_options={'certfile':'certificate.crt','keyfile':'privateKey.key'}
    srv=tornado.httpserver.HTTPServer(app, xheaders=True, ssl_options=ssl_options)
    srv.listen(443)
    print("Listening")
    tornado.ioloop.IOLoop.current().start()
```
HTTPS is working, I can access the fake registration server with Firefox, but the plug cannot communicate with the server, because there is "no shared cipher".
I can see in Wireshark that the plug is offering two cipher suites:
```
Cipher Suite: TLS_PSK_WITH_AES_128_CBC_SHA256 (0x00ae)
Cipher Suite: TLS_EMPTY_RENEGOTIATION_INFO_SCSV (0x00ff)
```
I then tried to add the firest cipher suite into the fake registration server by changing the ssl_options to 
`   ssl_options={'certfile':'certificate.crt','keyfile':'privateKey.key','ciphers':'PSK-AES128-CBC-SHA256'}`
(when using a misspelled cipher I get an error "No cipher can be selected", so the above cipher seams to be the correct spelling and is installed on the system)
Unfortunately the "no shared cipher" error persists.

Can you or someone else give a hint how to overcome this cipher problem.
How did you set up your python server?

Maybe the PSK is missing. Did you find out the PSK and how did you add it to the script?

floodwayprintco, 2019-09-02:
I think I am having the same problem. SA102 plug from Zoozee, never connected to app - but tuya-convert just hangs and never finds the plug.

Any suggestions to get this to work? 

I don't see any big issues in my log besides the mqtt.log where it says Cannot exec 'mosquitto' no such file or directory.

Any advice would be appreciated. Are my plugs even flashable? This threads makes it seem that they are.

GrantsPi, 2019-10-01:
I too have not been successful with the ZooZee SA102.  Tuya-Convert seems to connect but then times out after 10 or 15 minutes.  Then I’m not able to get the switches to respond to anything.  It appears that they are ‘bricked’

hume-github, 2019-10-23:
I can confirm that the "new" SA102s (purchased in the last thirty days, otherwise identical) seem to be reacting badly to tuya-convert.  I have a set of three that I purchased late July which I was able to get flashed without too much trouble (one bricking).  I purchased another set looking to do the same procedure, and now two are dead.

They smartconfig just fine, I see them connect to the mosquitto, register with the fake-registration-server, and then they request upgrade.bin.  They pull it down, and then that's the last I hear from them.  So I've got three working fine, and two more from the newer batch bricked.

I *do* have a USB UART and I'm working on recovering them via the serial console, but I'm new to that sort of thing and I'm not being overly successful in connecting (and I don't want to abuse this spot for help).

It should be noted that on the Amazon page for the SA102, there's a person commenting that they were having similar problems as this... but then they let their SA102 connect via SmartLife app, and then forced the SA102 to factory-reset via that app.  After that they were able to tuya-convert successfully.

Since I only have two "virgin" SA102s left I want to get serial recovery working before testing that, though.

kueblc, 2019-10-23:
@GrantsPi @hume-github thanks for reporting your experience. It would be really helpful for debugging if you could post your logs and backups from devices you believe may be bricked. In most cases it is found that they are not actually bricked and just stuck waiting on the next step of the process.

hume-github, 2019-10-23:
@kueblc : I'm glad to say that seems to be the case with the second of the two of the new batch.  I've managed to recover that one... it took some plugging/unplugging of power but it eventually showed up on the vflash SSID and let me flash it.

Unfortunately I've lost the logs from the *first* dead one.  I noticed that sometimes it will flash the LED when plugged in but it seems to have stopped even doing that.  I'm going to work with that one some more and see if I can coax it out.

hume-github, 2019-10-24:
@kueblc : Here's from plug 3:

hostapd (94:65:2d:6c:c9:74 is my phone, sitting on 10.42.42.10...)
```
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlx00ad2442d767 with hwaddr 00:ad:24:42:d7:67 and ssid "vtrust-flash"
wlx00ad2442d767: interface state UNINITIALIZED->ENABLED
wlx00ad2442d767: AP-ENABLED
wlx00ad2442d767: STA 94:65:2d:6c:c9:74 IEEE 802.11: associated
wlx00ad2442d767: AP-STA-CONNECTED 94:65:2d:6c:c9:74
wlx00ad2442d767: STA 94:65:2d:6c:c9:74 RADIUS: starting accounting session 29CC1200AEA2DAE7
wlx00ad2442d767: STA 94:65:2d:6c:c9:74 WPA: pairwise key handshake completed (RSN)
wlx00ad2442d767: STA cc:50:e3:03:ec:4d IEEE 802.11: associated
wlx00ad2442d767: AP-STA-CONNECTED cc:50:e3:03:ec:4d
wlx00ad2442d767: STA cc:50:e3:03:ec:4d RADIUS: starting accounting session 1F0A8ED8EB116437
wlx00ad2442d767: STA cc:50:e3:03:ec:4d WPA: pairwise key handshake completed (RSN)
wlx00ad2442d767: STA cc:50:e3:03:ec:4d IEEE 802.11: disassociated
wlx00ad2442d767: AP-STA-DISCONNECTED cc:50:e3:03:ec:4d
```

dnsmasq:

```
dnsmasq: started, version 2.79 cachesize 150
dnsmasq: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
dnsmasq-dhcp: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
dnsmasq: using nameserver 1.1.1.1#53
dnsmasq: using nameserver 9.9.9.9#53
dnsmasq: read /etc/hosts - 7 addresses
dnsmasq-dhcp: DHCPDISCOVER(wlx00ad2442d767) cc:50:e3:03:ec:4d
dnsmasq-dhcp: DHCPOFFER(wlx00ad2442d767) 10.42.42.34 cc:50:e3:03:ec:4d
dnsmasq-dhcp: DHCPDISCOVER(wlx00ad2442d767) cc:50:e3:03:ec:4d
dnsmasq-dhcp: DHCPOFFER(wlx00ad2442d767) 10.42.42.34 cc:50:e3:03:ec:4d
dnsmasq-dhcp: DHCPREQUEST(wlx00ad2442d767) 10.42.42.34 cc:50:e3:03:ec:4d
dnsmasq-dhcp: DHCPACK(wlx00ad2442d767) 10.42.42.34 cc:50:e3:03:ec:4d ESP_03EC4D
```

mosquitto:
```
1571876318: New connection from 10.42.42.34 on port 1883.
1571876318: New client connected from 10.42.42.34 as 88020554cc50e303ec4d (c1, k30, u'88020554cc50e303ec4d').
1571876318: Sending CONNACK to 88020554cc50e303ec4d (0, 0)
1571876318: Received SUBSCRIBE from 88020554cc50e303ec4d
1571876318:     smart/device/in/88020554cc50e303ec4d (QoS 0)
1571876318: 88020554cc50e303ec4d 0 smart/device/in/88020554cc50e303ec4d
1571876318: Sending SUBACK to 88020554cc50e303ec4d
1571876326: New connection from 127.0.0.1 on port 1883.
1571876326: New client connected from 127.0.0.1 as P1 (c1, k60).
1571876326: Sending CONNACK to P1 (0, 0)
1571876326: Received PUBLISH from P1 (d0, q0, r0, m0, 'smart/device/in/88020554cc50e303ec4d', ... (147 bytes))
1571876326: Sending PUBLISH to 88020554cc50e303ec4d (d0, q0, r0, m0, 'smart/device/in/88020554cc50e303ec4d', ... (147 bytes))
1571876326: Socket error on client P1, disconnecting.
1571876330: Received PUBLISH from 88020554cc50e303ec4d (d0, q0, r0, m0, 'smart/device/out/88020554cc50e303ec4d', ... (80 bytes))
1571876330: Received PUBLISH from 88020554cc50e303ec4d (d0, q0, r0, m0, 'smart/device/out/88020554cc50e303ec4d', ... (81 bytes))
1571876375: Client 88020554cc50e303ec4d has exceeded timeout, disconnecting.
1571876375: Socket error on client 88020554cc50e303ec4d, disconnecting.
```

and finally fake-registration-server.py:
```
URI:/gw.json?a=s.gw.token.get&gwId=88020554cc50e303ec4d&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":1,"source":"ap","path":"broadcast"}}&t=8&v=3.0&sign=88adf3a3f0d2a9fb3f72b48cddcc4e54
Answer s.gw.token.get
[I 191023 21:18:36 web:2246] 200 POST /gw.json?a=s.gw.token.get&gwId=88020554cc50e303ec4d&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":1,"source":"ap","path":"broadcast"}}&t=8&v=3.0&sign=88adf3a3f0d2a9fb3f72b48cddcc4e54 (10.42.42.34) 0.36ms

URI:/gw.json?a=s.gw.dev.pk.active&gwId=88020554cc50e303ec4d&other={"token":"00000000"}&t=7&v=3.0&sign=ee02a13cde810517b05a3d8539c972e0
Answer s.gw.dev.pk.active
READ GW ID 88020554cc50e303ec4d
TRIGGER UPGRADE IN 10 SECONDS
[I 191023 21:18:36 web:2246] 200 POST /gw.json?a=s.gw.dev.pk.active&gwId=88020554cc50e303ec4d&other={"token":"00000000"}&t=7&v=3.0&sign=ee02a13cde810517b05a3d8539c972e0 (10.42.42.34) 4.01ms
Trigger upgrade in 10 seconds
[... 9 lines cut ...]
URI:/gw.json?a=tuya.device.upgrade.silent.get&gwId=88020554cc50e303ec4d&t=7&v=4.1&sign=3225d5b7fe628e5901b715874cc7d650
Answer s.gw.upgrade
b'{"result":{"auto":3,"fileSize":"506587","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"dc92883ebc44f64253bfb004433fc929"},"t":100,"e":false,"success":true}'
[I 191023 21:18:43 web:2246] 200 POST /gw.json?a=tuya.device.upgrade.silent.get&gwId=88020554cc50e303ec4d&t=7&v=4.1&sign=3225d5b7fe628e5901b715874cc7d650 (10.42.42.34) 3.36ms
806fa555c93d1e06ca801e4674f83c8c
b'2.1c93d1e06ca801e46t5inCMDHpcJKQpxfOe2RbciptWt0ALftwtMURiz+L1KCLiUBrJr/im+tQk6mXAhZJsm5uMafEvnzLLz/1MQ4v1MYMNAWWbftQSc9rFnOMg7dg4SXZL9kLEC9aqKXAOwI'

URI:/gw.json?a=s.gw.upgrade&gwId=88020554cc50e303ec4d&t=103&sign=7addf228f09434fa4b579da806f23ab2
Answer s.gw.upgrade
b'{"result":{"auto":3,"fileSize":"506587","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"dc92883ebc44f64253bfb004433fc929"},"t":100,"e":false,"success":true}'
[I 191023 21:18:46 web:2246] 200 POST /gw.json?a=s.gw.upgrade&gwId=88020554cc50e303ec4d&t=103&sign=7addf228f09434fa4b579da806f23ab2 (10.42.42.34) 5.54ms

URI:/gw.json?a=s.gw.upgrade.updatestatus&gwId=88020554cc50e303ec4d&t=100&sign=8ece87ec5dbb3edf65d1fbb4052314a1
[... cut ...]
After this, the smartplug audibly "clicked" (which I'm assuming is the reboot) and is apparently dead.  It hasn't tried to re-associate with the AP after about five minutes.

This is identical behaviour to the first two plugs, where plug-2 eventually came back after some plugging/unplugging and unmeasured button-mashing.  plug-1 is still dead.

I've left plug 3 in the wall, untouched, in case you have some suggestion as to where to go from here, rather than the random stuff I've been trying.  Will the button commands mentioned at https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage do anything here?

kueblc, 2019-10-24:
From your `fake-registration-server` log I can see that you are running an old version of `tuya-convert`. And it looks like it took the `upgrade.bin` so it should be running the intermediate firmware. If that's the case, it won't respond to any button presses.

Have you tried physically power cycling the device ie. unplug and plug back in? It will first try to connect to `vtrust-flash` and if it can't it will spawn an AP called `vtrust-recovery`.

hume-github, 2019-10-24:
Unplugging and plugging (both immediately and after a few minutes' wait) produces no reaction.  No vtrust-recovery appears and there's no attempt to connect to the vtrust-flash AP.

Dead plug 1 will flash its LED five times on plug-in, but then just sits there.  Dead plug 3 flashes once, and then just sits there.  Or... they did.  I just tried that now and neither one is doing so.

I've pulled a fresh tuya-convert, not that it's been needed yet.  :)

sreknob, 2019-12-03:
Just commenting that I've had the same behaviour in 3 of 8 SA102s with the most recent version of tuya-convert using a raspi B+
5 of them flashed fine, with some hangs in the backup of the flash (which I was able to pull from another terminal window or browser) but otherwise worked well.
The other three appear to get stuck somewhere during setup of the intermediate flash. They showed being programmed, LED went from flashing, to flashing red, then relay clicked off and LED went out, never to be seen again.... no recovery SSID or joining flash SSID while tailing wifi log.
Interestingly, sometimes the LEDs exhibit different behaviour, sometimes flashing fast, sometimes slow and most often not at all when powered up.
@hume-github - found a good way to open these guys up?

hume-github, 2019-12-03:
@sreknob : I've seen others dremel the back off (the necessary solder points are on the back of the unit, directly under the plastic).  I spent more time (and blood) running a knife around the seam on the back and gently prying the cover off.  Then you have to desolder the mains power to slide the logic board off to get access to the solder points underneath.

I was able to flash it using esptool then after some trial/error with the USB link.  The guides I've seen recommend fully erasing the flash before flashing in the firmware, and I can confirm that doing so helps.

Of the four I managed to get working, three either failed or were simply too unreliable to trust.  The one I opened and flashed via serial was one of the failed.  I intend to open it back up and re-flash it again with Tasmota 7.  OTA updates seemed to cause most of my failures... only one died outright, but the others were crashing/rebooting with exceptions roughly once an hour.  One hung/crashed and cut power to the fridge (it was an experiment).  I'm hoping when I open them this time I'll be able to rig up a proper connector so I can plug into them more easily next time.

Fortunately despite all the fuss I'm still having fun.  But for anyone who just wants it to work, the ZooZees are a really bad choice.  :)

sreknob, 2019-12-04:
Thanks @hume-github for the detailed reply, much appreciated. If you do eventually disassemble one again, it would be amazing if you took some photos to document it. Can’t find anything specifically for the sa102 on google. Guess we’ll see who gets around to it first :-)
Looking at the similar Teckin one (SP22, I think), I wonder if a few very well placed drill holes and some pogo pins would do the job.
As for reliability, I’ve put esphome on my units and OTA reflashed them around 50 times without issues once they were converted. I’ll report back how they fair after I’ve run them in a little.

hume-github, 2019-12-04:
@sreknob : Check out https://github.com/arendst/Tasmota/issues/4175 ... that's a Teckin, but the guts of the SA102 are identical.

sreknob, 2019-12-04:
> @sreknob : Check out [arendst/Tasmota#4175](https://github.com/arendst/Tasmota/issues/4175) ... that's a Teckin, but the guts of the SA102 are identical.

Exactly what I was looking for - many thanks!