ct-Open-Source/tuya-convert #207: Gosund WP3 Wiring for Flashing?

cheechie, 2019-05-30:
Does anyone know what the wiring should be for the gosund WP3.  Do I have to hold GPIOZero to ground when I plug in the usbflasher to get this into programing mode?

[image]
[image]

fyrmangeoff, 2019-05-31:
I just got some of these from Amazon. Was trying to OTA flash them not having any luck.  How did you get them apart?

cheechie, 2019-05-31:
I just used some plumbers pliers and squeezed around the edge to put the seal loose.  I put some thin cardboard in between the pliers and the switch to not damage it.  It came apart really clean.  

[image]

0xBiscuit, 2019-06-03:
Yes. I pulled the v5 firmware this way today. I had to remove the chip. You'll find a tx and rx pin along with 3v and ground. You need to pull the IO0 pin low in order to read or flash the firmware.

cheechie, 2019-06-03:
I connected to the pins and flashed it with esphome.  It functions correctly.

kueblc, 2019-06-11:
Glad to hear it; thanks @0xBiscuit for your input

closed 2019-06-11 by kueblc

kymnyth, 2019-06-12:
Anybody care to share photos and instructions for flashing?  New to this and flashed some shelly's and did some TUYAConvert flashes but this will be my first attempt at soldering on connections for flashing.

cheechie, 2019-06-20:
After opening it with the pliers like described above.  I used a small saw to cut across the bottom of the plug to remove a portion of the underside.  That gave me access to the pins to solder to.  Then after it was programmed, I used glue (crazy glue and epoxy) to seal it back up and close the gap from the saw.  I did not take pictures of the process. 

[image]

aage21, 2019-08-04:
Do you have any pics of the pins soldered?

cheechie, 2019-08-06:
No I do not.  Sorry

funkybluemonkey3, 2020-06-27:
So which side corresponds to the points that need to be soldered, top or bottom? 
[image]

cheechie, 2020-06-28:
> So which side corresponds to the points that need to be soldered, top or bottom?
> [image]

I honestly don't remember. I think I used a multimeter to check for continuity to make sure I was connecting to the proper pins.

kiprock, 2021-01-19:
Hey @cheechie did you have to pull the entire chip out? I connected to the pins and supplied voltage +3 from my pi but esptool won't connect to it. Thanks in advance for your response!!!

cheechie, 2021-01-19:
> Hey @cheechie did you have to pull the entire chip out? I connected to the pins and supplied voltage +3 from my pi but esptool won't connect to it. Thanks in advance for your response!!!

I didn't pull the chip out.  I believe i solder to the pins.

kiprock, 2021-01-19:
Thanks! Did you power it externally or from the host plug?

cheechie, 2021-01-19:
> Thanks! Did you power it externally or from the host plug?

I powered with the programmer. Not the 120v

kiprock, 2021-01-19:
For anyone else in this situation, I flashed mine using a pi and esptool. I had to use the host power (120V) instead of the 3v from the pi. I did not have to remove the chip. Thanks @cheechie for responding to my questions!

Scope666, 2024-05-05:
Does anybody know what the pins are for this version?  I think it's a v2 but not positive.  It's definitely flashable because I've had both Tasmota and ESP Home on it.

[image]