ct-Open-Source/tuya-convert #209: Connects to wifi but doesn't transfer anything

sheppy99, 2019-06-01:
I've got a brand new 2 gang WiFi switch that I'm trying to flash. it connects to tuya-convert but nothing transfers. From my phone after connecting I get "Hello World" on a web browser from http://10.42.42.1 and I can send and receive mqtt messages.
smarthack-wifi.log suggests that the switch is connected to the wifi, but apart from the packets from my phone there is no activity reported in smarthack-mqtt.log and smarthack-web.log. smarthack-smartconfig.log reports smartconfig complete. If I press and hold the first button on the WiFi switch the LED comes on permanently and I have another network called "SmartLife-EA67"
What am I missing? What file is the WiFi switch looking for and where should it be? If seems like it isn't requesting an update, do I need to download the SmartLife app before it will try and update?

sheppy99, 2019-06-03:
I tried this again a day later and the device got a different ip address which still isn't 10.42.42.42 - should I edit the flash script to flash this new ip or will it likely brick the device?

SwiCago, 2019-06-04:
I bought 10 Tuya AI lights, most went right away. 2 required trying the procedure over and over with reboots. Eventually they took the correct IP. And 3 still have not taken the correct IP. Those that don't take hold sit on IP 10.42.42.28 
Would be nice if the script could let you know it failed, instead of waiting for the silly dots

sheppy99, 2019-06-04:
That's interesting! In the end I found out what the pinout on the ESP was and flashed with a cable which took a fraction of the time it did to try and get this working. It would be good to work out what is going wrong though for others who aren't good with a soldering iron and magnifying glass 
> I bought 10 Tuya AI lights, most went right away. 2 required trying the procedure over and over with reboots. Eventually they took the correct IP. And 3 still have not taken the correct IP. Those [...]

SwiCago, 2019-06-05:
After letting the unit marinate a few hours, I decided to give them another try  right after I posted. 2 of the 3 flashed. Now I just have one left. It seems to me as though it is a timing issue or something.

kueblc, 2019-06-11:
Glad you got your firmware installed via soldering. To answer a few questions before I close this:

> If I press and hold the first button on the WiFi switch the LED comes on permanently and I have another network called "SmartLife-EA67"

This is AP config mode, where it blinks slowly. `tuya-convert` needs the device to be in EZ config mode, where it blinks quickly.

> Looking at the ip address range in setup_ap.sh it runs from 10.42.42.10 to 10.42.42.40, and the script start_flash.sh is referring to 10.42.42.42 - where does the device get 10.42.42.42 from, is it hard coded into the original tuya firmware?

`10.42.42.42` is hardcoded in the intermediate firmware, which is loaded by the Tuya firmware and allows us to flash custom firmware. This is how we know the device is running the intermediate rather than the Tuya or Tasmota firmware. No need to edit the script if you don't have this IP address.

Closing; feel free to open a new issue with any new concern.

closed 2019-06-11 by kueblc

sheppy99, 2019-06-11:
I tried putting it into fast blink mode and that didn't work either. It did connect to the wifi according to the logs but nothing else happened, no website activity and no mqtt activity.

kueblc, 2019-06-12:
It sounds likely that these devices have the patched firmware. If you have a dump of the firmware we can verify that and contribute to the research efforts.

sheppy99, 2019-06-12:
Happy to have a go at that with the next batch, how do I do it?

kueblc, 2019-06-12:
You can use the esptool ie `esptool.py read_flash 0x00000 0x100000 fwbackup.bin`

sheppy99, 2019-06-12:
Ok, I’ll do that. What should I do with it once I have it?

kueblc, 2019-06-12:
You can post it here or if you'd prefer you can email it to me directly. If you've never connected the device to your WiFi network there isn't any sensitive information contained within the binary.

sheppy99, 2019-06-19:
I have a new switch connected to serial but I only have the windows version of the program. Can you point me to what program to use on Windows 10 and what command to use please. Normally to upload I simply start the flasher that comes with ESPEasy, select com3 and the file I want to upload which is usually sonoff.bin.
I did try it in python under windows but it didn't work

sheppy99, 2019-07-01:
[010719_Tuya_Firmware_Backup.zip](https://github.com/ct-Open-Source/tuya-convert/files/3343703/010719_Tuya_Firmware_Backup.zip)