ct-Open-Source/tuya-convert #210: I flashed this per the guide but now the wifi light is solid blue and won't reset.  I logged into the tasmota page and set my wifi SSID and my password but it doesn't show up  in my devices.  Is there a way to force this plug back into flash mode so I can attempt it again or is it just bricked? Tuya Smart Plug 10AL250V I am sorry I have no idea what firmware this unit came with.

droidforever, 2019-06-06:
[###](url
[logfiles.zip](https://github.com/ct-Open-Source/tuya-convert/files/3259818/logfiles.zip)
) Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

droidforever, 2019-06-06:
I flashed this per the guide but now the wifi light is solid blue and won't reset.  I logged into the tasmota page and set my wifi SSID and my password but it doesn't show up  in my devices.  Is there a way to force this plug back into flash mode so I can attempt it again or is it just bricked? Tuya Smart Plug 10AL250V I am sorry I have no idea what firmware this unit came with.

droidforever, 2019-06-06:
I flashed this per the guide but now the wifi light is solid blue and won't reset. I logged into the tasmota page and set my wifi SSID and my password but it doesn't show up in my devices. Is there a way to force this plug back into flash mode so I can attempt it again or is it just bricked? Tuya Smart Plug 10AL250V I am sorry I have no idea what firmware this unit came with. Blue Solid Light not connecting to network
[logfiles.zip](https://github.com/ct-Open-Source/tuya-convert/files/3259848/logfiles.zip)

kueblc, 2019-06-11:
This device is running Sonoff Tasmota; it is not bricked. Sounds like you may have accidentally entered the wrong WiFi credentials. Please refer to the support channels for Sonoff Tasmota.

closed 2019-06-11 by kueblc