ct-Open-Source/tuya-convert #211: Gosund WP3 Flash failing

kymnyth, 2019-06-06:
Successfullly flashed other devices using tuya-convert.  
These are the steps I followed:
Accepted risk notice.
Connected phone to vtrust-flash
Set device to pairing mode, and I see fast flashing blue led on power button.
Press enter.
Fast flashing LED turns to slow flashing for a sec and then solid red 
Never makes it beyond dots flashing across screen.
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3263431/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3263432/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3263433/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3263434/smarthack-smartconfig.log)

RichardUUU, 2019-06-07:
Join the party: https://github.com/ct-Open-Source/tuya-convert/issues/199

kueblc, 2019-06-11:
This is a known issue, let's keep the discussion in one place if we can.

closed 2019-06-11 by kueblc