ct-Open-Source/tuya-convert #212: Universal Infrared remote cant flash

jamietang68, 2019-06-07:
Hi,

I followed the steps and previously able to flash the devices few months back (universal remote control). Same with this just different branding : https://www.amazon.com/YTF-Conditioner-Controller-Compatible-Smartphone/dp/B07H46LJZK

I just bought a few new 1, but it doesnt seems to respond. How do i know if the firmware cant be flashed anymore? 

I'm not sure how to check the firmware version. 

the logs below:

**smarthack-mqtt.log**
pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-mqtt.log
1559774655: Opening ipv4 listen socket on port 1883.
1559774655: Opening ipv6 listen socket on port 1883.
1559775620: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1559775620: Using default config.
1559775620: Opening ipv4 listen socket on port 1883.
1559775620: Opening ipv6 listen socket on port 1883.
1559776009: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1559776009: Using default config.
1559776009: Opening ipv4 listen socket on port 1883.
1559776009: Opening ipv6 listen socket on port 1883.

**smarthack-web.log**
pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-web.log
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80

**smarthack-wifi.log**
pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-wifi.log
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:dd:80:08 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA 04:4b:ed:79:f4:35 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 04:4b:ed:79:f4:35
wlan0: STA 04:4b:ed:79:f4:35 RADIUS: starting accounting session 886C15EAECDC8FAB
wlan0: STA 04:4b:ed:79:f4:35 WPA: pairwise key handshake completed (RSN)
wlan0: STA 04:4b:ed:79:f4:35 WPA: group key handshake completed (RSN)

**smarthack-smartconfig.log**
pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-smartconfig.log
Sending wifiPassword flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID vtrust-flash
Sending wifiPassword flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.

jamietang68, 2019-06-07:
solve by following this guide. seems like just need to update nodejs to latest version.

closed 2019-06-07 by jamietang68

kueblc, 2019-06-11:
> solve by following this guide. seems like just need to update nodejs to latest version.

Glad you got it working, but I'd like to add that we don't use `nodejs` so it was probably an unrelated issue.