ct-Open-Source/tuya-convert #213: Gosund SP1 (SP1-C_V2.4 20181101) is using https instead of http

toca2k, 2019-06-07:
Hi,

i'm trying to flash some Gosund SP1 (SP1-C_V2.4 20181101) . Looks like the Smartconfig part works fine. The device gets an IP, 10.42.42.32 in my case. As far as i understood the next step should be a http request to http://10.42.42.1. But in my case I see a connect to Port 443 via https :(

I tried to fake the https server with mitmproxy (sudo mitmproxy -R http://10.42.42.1 -p 443) but as I expected the Device is not accepting the certificate :(

Can anybody verify this?
Any ideas how to fix it?

jfractalj, 2019-06-08:
I have seen the same thing with the WP3. I don't think there has been a OTA fix released yet, but the device can probably be flashed manually.

kueblc, 2019-07-02:
Duplicate of #160 #166 #197 #199 #201

closed 2019-07-02 by kueblc

BigDi, 2019-07-07:
I tried it and works!

wolfgangroller, 2019-07-26:
> I tried it and works!

Please tell us what exactly you did, that it works now.

Thanks

BigDi, 2019-07-27:
I take the last two pull requests and merge it into the current release.