ct-Open-Source/tuya-convert #215: Can't Flash Any Switches / Plugs / Bulbs in my house

folkg, 2019-06-13:
I have a number of smart switches (normal and dimmer) as well as two smart plugs and two smart bulbs. I am unable to flash any of them with Tuya-Convert. I don't know if I just got unlucky and all of them have the latest firmware, or if there is something wrong with my set up. I am using a raspberry pi 3B+ on Raspbian Stretch. Could someone help me take a look at my logs and see what is going on?

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3288262/smarthack-web.log)

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3288263/smarthack-wifi.log)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3288264/smarthack-mqtt.log)

[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3288265/smarthack-smartconfig.log)

kueblc, 2019-06-14:
It looks like your device never connected, so there's still hope you aren't running the TLS firmware.

I'm seeing a lot of
```
iptables v1.6.0: can't initialize iptables table `filter': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
```

tobiwashere, 2019-06-14:
Have you tried an

sudo apt-get update
sudo apt-get upgrade
reboot

to make sure you are running the latest version of all tools?

folkg, 2019-06-15:
Thank you both for the suggestions! I think I have stopped the iptables issue. I had to do an rpi-update and reboot.

I successfully flashed 2 of my bulbs and 1 smart plug. Strangely though, I cannot flash the other identical smart plug, and I can't flash any of my wall switches. Would you mind taking another quick look and see if anything sticks out?

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3292587/smarthack-wifi.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3292588/smarthack-smartconfig.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3292589/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3292590/smarthack-web.log)

closed 2019-06-18 by folkg