ct-Open-Source/tuya-convert #216: Gosund SP111 unable to flash

andrasg, 2019-06-14:
I have several Gosund SP111 bought from amazon.de and trying to tuya-convert them. I am using a raspberry using a brand new raspbian stretch image. Have it up-to-date, have executed `sudo apt-get update` and `sudo apt-get upgrade` and it shows nothing to upgrade.

Steps I am taking:
1. launch `start_flash.sh`
2. connect my android tablet to the `vtrust-flash` AP
3. put the device into pairing (blue led flashing)
4. smartconnect succeeds
5. device clicks and led turns red
6. dots just keep on being displayed forever (10+ mins) with the screen showing `Waiting for the upgraded device to appear`

smarthack-mqtt.log:
```
1560542934: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1560542934: Using default config.
1560542934: Opening ipv4 listen socket on port 1883.
1560542934: Opening ipv6 listen socket on port 1883.
```

smarthack-smartconfig.log:
```
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```

smarthack-wifi.log:
```
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:4f:65:d2 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: associated
wlan0: AP-STA-CONNECTED bc:dd:c2:23:8c:4c
wlan0: STA bc:dd:c2:23:8c:4c RADIUS: starting accounting session 96F07BE53FA360C6
wlan0: STA bc:dd:c2:23:8c:4c WPA: pairwise key handshake completed (RSN)
wlan0: STA 08:fd:0e:f0:16:01 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 08:fd:0e:f0:16:01
wlan0: STA 08:fd:0e:f0:16:01 RADIUS: starting accounting session 87DBEAC35F7D5CC0
wlan0: STA 08:fd:0e:f0:16:01 WPA: pairwise key handshake completed (RSN)
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED bc:dd:c2:23:8c:4c
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: associated
wlan0: AP-STA-CONNECTED bc:dd:c2:23:8c:4c
wlan0: STA bc:dd:c2:23:8c:4c RADIUS: starting accounting session 83017C80662B81B6
wlan0: STA bc:dd:c2:23:8c:4c WPA: pairwise key handshake completed (RSN)
wlan0: STA 08:fd:0e:f0:16:01 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 08:fd:0e:f0:16:01
wlan0: STA 08:fd:0e:f0:16:01 IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED bc:dd:c2:23:8c:4c
wlan0: STA bc:dd:c2:23:8c:4c IEEE 802.11: disassociated
```

smarthack-web.log:
```
Listening on port 80
```

Am I unlucky and got some device with new, TLS firmware?

kueblc, 2019-06-14:
Yes unfortunately, judging by the fact that the device appears on the wifi but makes no requests, it looks like you are running the TLS firmware.

andrasg, 2019-06-14:
Thanks for confirming.

closed 2019-07-02 by kueblc

andrasg, 2019-11-15:
Is there any update to the SP111 with tuya-convert 2.0? The homepage leads me to believe that there is a new way now to flash devices even with TLS firmware?

kueblc, 2019-11-15:
Yes, you can now flash TLS devices as of #273