ct-Open-Source/tuya-convert #217: Cant flash Tuya TC-US

haianhlc99, 2019-06-17:
Hello I'm flashing Tuya TC-US 2 gang and it stopped here
here is my logs

smarthack-mqtt
`1560774309: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1560774309: Using default config.
1560774309: Opening ipv4 listen socket on port 1883.
1560774309: Opening ipv6 listen socket on port 1883.
^C1560775106: mosquitto version 1.4.10 terminating
1560775116: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1560775116: Using default config.
1560775116: Opening ipv4 listen socket on port 1883.
1560775116: Opening ipv6 listen socket on port 1883.
^C1560775316: mosquitto version 1.4.10 terminating
1560775321: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1560775321: Using default config.
1560775321: Opening ipv4 listen socket on port 1883.
1560775321: Opening ipv6 listen socket on port 1883.
^C1560776097: mosquitto version 1.4.10 terminating
1560776102: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1560776102: Using default config.
1560776102: Opening ipv4 listen socket on port 1883.
1560776102: Opening ipv6 listen socket on port 1883.
^C1560776607: mosquitto version 1.4.10 terminating
1560776611: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1560776611: Using default config.
1560776611: Opening ipv4 listen socket on port 1883.
1560776611: Opening ipv6 listen socket on port 1883.
`

smarthack-smartconfig
`Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
`
smarthack-web
`Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
ImportError: No module named 'tornado'
Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
[... 301 lines cut ...]
wlan0: STA bc:dd:c2:28:ea:6d RADIUS: starting accounting session C52960B39CED9027
wlan0: STA bc:dd:c2:28:ea:6d WPA: pairwise key handshake completed (RSN)
wlan0: STA 40:9c:28:a9:8f:92 IEEE 802.11: associated
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 40:9c:28:a9:8f:92
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 40:9c:28:a9:8f:92
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 40:9c:28:a9:8f:92
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 40:9c:28:a9:8f:92
wlan0: STA 40:9c:28:a9:8f:92 IEEE 802.11: disassociated
wlan0: STA 40:9c:28:a9:8f:92 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 40:9c:28:a9:8f:92
wlan0: STA 40:9c:28:a9:8f:92 RADIUS: starting accounting session F4F0CB59B19601AD
wlan0: STA 40:9c:28:a9:8f:92 WPA: pairwise key handshake completed (RSN)
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED bc:dd:c2:28:ea:6d
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
wlan0: STA bc:dd:c2:28:ea:6d IEEE 802.11: disassociated
`

kueblc, 2019-06-17:
Make sure your system is up to date and you've installed all the prerequisites. Skimming through the logs I see you don't have `tornado` installed and you need to update `iptables`.

kueblc, 2019-09-28:
Closing due to inactivity. If you're still having issues, try the latest `tuya-convert` and open a new issue.

closed 2019-09-28 by kueblc