ct-Open-Source/tuya-convert #219: How to restore Raspberry Pi original wlan0 configuration?  Or backout SoftAP configuration?

dachshund-digital, 2019-06-22:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

CaptClaude, 2019-06-29:
It was my experience that simply rebooting the Pi after running the flash scripts reverted it back to it's normal self. I may be wrong, but I don't think that the changes wrought by tuya-convert are permanent.

kueblc, 2019-07-02:
Try rebooting as @CaptClaude suggested, otherwise you may wish to do a clean install on your SD card.

closed 2019-07-02 by kueblc