ct-Open-Source/tuya-convert #220: Can't flash EXTSUD E14 Light bulb

TPISCAREL, 2019-06-23:
Hi,

I was be able to flash others Tuya devices except this light bulb. it is out of the box and never connected to Tuya android app.

ec:fa:bc:2c:d7:8e is my light bulb.

Here's the full log.
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3317728/smarthack-web.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3317729/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3317730/smarthack-smartconfig.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3317731/smarthack-wifi.log)

Thanks in advance.

kueblc, 2019-06-23:
Open `scripts/fake-registration-server.py` to [line 83](https://github.com/ct-Open-Source/tuya-convert/blob/master/scripts/fake-registration-server.py#L83) and replace
```
answer = b'{"result":{"auto":3,"fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))
```
with
````
answer = b'{"result":{"type":0,"auto":true,"size":"%d","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))
```
and let us know how it goes

TPISCAREL, 2019-06-23:
@kueblc it works!! I am  able to flash both of them.

Here's log results. Thank you so much. I was stuck for a week.

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3318505/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3318506/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3318507/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3318508/smarthack-smartconfig.log)

closed 2019-06-23 by TPISCAREL

kueblc, 2019-06-23:
> @kueblc it works!! I am able to flash both of them.

Thrilled to hear it. I'll plan on including this fix in the next release. Thanks for testing!