ct-Open-Source/tuya-convert #221: Unable to flash

filoor, 2019-06-25:
im unable to flash my switch 
type: tuya convert wifi licht 86 eu knop switch
link : https://nl.aliexpress.com/item/32996551323.html

Log:
`Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:5c:a1:95 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 64:a2:f9:0f:7a:26
wlan0: STA 64:a2:f9:0f:7a:26 RADIUS: starting accounting session 82C209AC03A8FF60
wlan0: STA 64:a2:f9:0f:7a:26 WPA: pairwise key handshake completed (RSN)
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ec:fa:bc:4b:88:8b
wlan0: STA ec:fa:bc:4b:88:8b RADIUS: starting accounting session E216751793294843
wlan0: STA ec:fa:bc:4b:88:8b WPA: pairwise key handshake completed (RSN)
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED ec:fa:bc:4b:88:8b
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ec:fa:bc:4b:88:8b
wlan0: STA ec:fa:bc:4b:88:8b RADIUS: starting accounting session B6A99F240E3AE89F
wlan0: STA ec:fa:bc:4b:88:8b WPA: pairwise key handshake completed (RSN)
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED ec:fa:bc:4b:88:8b
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: disassociated
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 64:a2:f9:0f:7a:26
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: disassociated
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 64:a2:f9:0f:7a:26
wlan0: STA 64:a2:f9:0f:7a:26 RADIUS: starting accounting session 64A92D1478C65BF2
wlan0: STA 64:a2:f9:0f:7a:26 WPA: pairwise key handshake completed (RSN)
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 64:a2:f9:0f:7a:26
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: disassociated
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 64:a2:f9:0f:7a:26
wlan0: STA 64:a2:f9:0f:7a:26 RADIUS: starting accounting session E159C1C15CDF0CD3
wlan0: STA 64:a2:f9:0f:7a:26 WPA: pairwise key handshake completed (RSN)
Backing up NetworkManager.cfg...
[... 135 lines cut ...]
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:5c:a1:95 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 64:a2:f9:0f:7a:26 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 64:a2:f9:0f:7a:26
wlan0: STA 64:a2:f9:0f:7a:26 RADIUS: starting accounting session 4DB23CC88E95D321
wlan0: STA 64:a2:f9:0f:7a:26 WPA: pairwise key handshake completed (RSN)
wlan0: STA ec:fa:bc:4b:88:8b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ec:fa:bc:4b:88:8b
wlan0: STA ec:fa:bc:4b:88:8b RADIUS: starting accounting session E2DA99517D18DC8C
wlan0: STA ec:fa:bc:4b:88:8b WPA: pairwise key handshake completed (RSN)
`

chrislawson, 2019-06-28:
> wlan0: Could not connect to kernel driver

try this

`sudo killall wpa_supplicant`

filoor, 2019-06-30:
I did a hardware Flash in the main time thanks will save this command for the next one

CaptClaude, 2019-07-01:
I see the 
> wlan0: Could not connect to kernel driver

error even after killing wpa_supplicant (and verifying that it is not running). I see it both on a successful flash and failed flash. I am assuming that the failure-to-flash is due to the new Tuya firmware that prevents tyya-convert from flashing.

kueblc, 2019-07-02:
Closing as issue has been resolved, please let us know if you still need help

closed 2019-07-02 by kueblc