ct-Open-Source/tuya-convert #222: Teckin SP20 won’t flash

garnern2, 2019-06-26:
Are you reporting a successful flash? No.

Do you have a question about configuring your newly flashed device? No.

Did you have issues getting a device to flash? Yes.

I have the Teckin SP20 plug, brand new. I tried to flash it before connecting to Smart Life. I read the other issue about a brute force approach and completing the process a second time, but how do we know anything is or has happened the first time? When do we stop the process in the sea of endless dots?

I made sure the device was in pairing mode and began the process. Now all I get are some light blue flickers when I initially connect power that stop after anywhere from 2 to 6 flashes. I have attempted connecting while the dots are running. I’ve attempted connecting my laptop to vtrust instead of my cellphone...and so on.

garnern2, 2019-06-29:
Is there any suggestion as to how I can completely reset the device? It seems bricked. What I don’t get is that one worked...I now have two that seem bricked.

baldfox, 2019-07-01:
i too am in a similar situation. I have a Teckin sp22 (eu version) that's bricked. Others flashed first time no problem. Would be interested in a way to reset them without forcing them open.

thwe1966, 2019-07-01:
Hello,

same problem, tryd even with a blank 'wpa_suplicant' -> no success!

Details:

- Device: Teckin SP21 smart plug (eu)
   https://www.amazon.de/gp/product/B07HHFKWJJ/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1
- Raspberry Pi 3 B with Strech lite, completely new
- tuya-convert installed and started with 'start_flash'
   (wlan 'vtrust-flash' is established, connection with a smartphone works)
- Teckin smart-plug is in pairing mode (led is flashing fast)
- After pressing 'Enter' -> 'starting pairing procedure ... waiting for the upgraded device to appear ...'
- Led on the Teckin goes out briefly, then constantly on
- flush program can apparently not appear correctly / completely the device

https://user-images.githubusercontent.com/52085361/60413209-5bfa1980-9bd4-11e9-8cd3-13bcf7f2f9a2.JPG

https://github.com/ct-Open-Source/tuya-convert/files/3344075/smarthack-mqtt.log

https://github.com/ct-Open-Source/tuya-convert/files/3344076/smarthack-smartconfig.log

https://github.com/ct-Open-Source/tuya-convert/files/3344078/smarthack-web.log

https://github.com/ct-Open-Source/tuya-convert/files/3344080/smarthack-wifi.log

garnern2, 2019-07-01:
I’ve torn mine open. Now what?

baldfox, 2019-07-02:
> I’ve torn mine open. Now what?

Get the soldering iron out!

jpcanovas, 2019-07-06:
I'm in the same situation with a Teckin SP22 (eu plug) I got from Amazon Spain two days ago. My mobile phone connects to the wifi network and i have a "listen to port 80" line in the smarthack-web.log. The plug connects to wifi, smarthconfig do its work (regarding its log), but the plug get out of the flashing mode after that (led permanent red).

Do you think there will be any update theat will enable to flash tuya without open it again?

tyjtyj, 2019-07-07:
@garnern2 , you device probaby did not flash at all. it still with stock tuya app. Did you try hold the button and plug the device(wait for fast flashing) and try connect via tuya app. 

Alternatively dont connect tuya app and unplug n plug then hold 5 second to fast blinking mode again

baldfox, 2019-07-10:
I thought the same initially but tuna app refuses to recognise it now. I am not able to do anything with the plug unfortunately. Any word on the next iteration on guys convert?

byvalkursstrappingbuttocks, 2019-07-16:
I cannot flash this either (SP20 US version).

I've tried three different wireless devices. Connected to "vrtrust-flash" with two different android devices and a laptop. I read about the power source potentially being a problem, so I've tried it plugged directly into the wall as well as into my UPS. I tried putting it in pairing mode before and after the dots appear. After the dots start nothing happens. Having it in pairing mode before the dots causes the plug to click, then switch to a solid red light. After that nothing.

```
Using interface wlan1 with hwaddr xx:xx:xx:xx:xx:xx and ssid "vtrust-flash"
wlan1: interface state UNINITIALIZED->ENABLED
wlan1: AP-ENABLED 
wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: authenticated
wlan1: STA yy:yy:yy:yy:yy:yy  IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED yy:yy:yy:yy:yy:yy 
wlan1: STA yy:yy:yy:yy:yy:yy  RADIUS: starting accounting session 5D2D1F09-00000000
wlan1: STA yy:yy:yy:yy:yy:yy  WPA: pairwise key handshake completed (RSN)
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
wlan1: STA zz:zz:zz:zz:zz:zz IEEE 802.11: authenticated
wlan1: STA zz:zz:zz:zz:zz:zz IEEE 802.11: associated (aid 2)
wlan1: AP-STA-CONNECTED zz:zz:zz:zz:zz:zz
wlan1: STA zz:zz:zz:zz:zz:zz RADIUS: starting accounting session 5D2D1F09-00000001
wlan1: STA zz:zz:zz:zz:zz:zz WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED zz:zz:zz:zz:zz:zz
wlan1: STA zz:zz:zz:zz:zz:zz IEEE 802.11: disassociated
wlan1: STA zz:zz:zz:zz:zz:zz IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

barqers, 2019-08-19:
Having the same issue as byvalkursstrappingbuttocks

SP20 as well CA version.

Anyone able to get this working?

benbamm, 2019-08-19:
SP21 here, bought from amazon.de last week. No way to get them flashed... Too bad :( hope there will be a solution soon.

RaptahJezus, 2019-08-24:
Unable to flash both SP20s I bought on Amazon on 8/24/2019.

I'm attempting to run tuya-convert on an RPi 3B, using my Note 9 as the other WiFi device. I put the device in pairing mode by plugging it in, and holding the power button for ~6 seconds until the blue LED flashes quickly. I then kick off the tuya-convert script, answer "yes", and connect my Note 9. I press enter, and it hangs on "Waiting for upgraded device to appear". 

I notice that after the first few dots, the SP20's LED stops blinking, and stays off. Here's my logs:

smarthack-mqtt.log

> 1566667511: mosquitto version 1.5.7 terminating
1566667551: mosquitto version 1.5.7 starting
1566667551: Using default config.
1566667551: Opening ipv4 listen socket on port 1883.
1566667551: Opening ipv6 listen socket on port 1883.

smarthack-smartconfig.log

> Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.

smarthack-web.log
> Listening on port 80

smarthack-wifi.log
> Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:30:44:53 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA 44:91:60:c8:3c:36 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 44:91:60:c8:3c:36
wlan0: STA 44:91:60:c8:3c:36 RADIUS: starting accounting session 46D462CDD8042B4A
wlan0: STA 44:91:60:c8:3c:36 WPA: pairwise key handshake completed (RSN)
wlan0: STA b4:e6:2d:75:e5:64 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED b4:e6:2d:75:e5:64
wlan0: STA b4:e6:2d:75:e5:64 RADIUS: starting accounting session 21B444AA338170B6
wlan0: STA b4:e6:2d:75:e5:64 WPA: pairwise key handshake completed (RSN)

According to my syslog, it seems to connect but then drop out of pairing mode
> Aug 24 18:35:10 raspberrypi hostapd: wlan0: STA 44:91:60:c8:3c:36 IEEE 802.11: associated
Aug 24 18:35:10 raspberrypi hostapd: wlan0: STA 44:91:60:c8:3c:36 RADIUS: starting accounting session 46D462CDD8042B4A
Aug 24 18:35:10 raspberrypi hostapd: wlan0: STA 44:91:60:c8:3c:36 WPA: pairwise key handshake completed (RSN)
Aug 24 18:35:10 raspberrypi dnsmasq-dhcp[7156]: DHCPREQUEST(wlan0) 10.42.42.25 44:91:60:c8:3c:36
Aug 24 18:35:10 raspberrypi dnsmasq-dhcp[7156]: DHCPACK(wlan0) 10.42.42.25 44:91:60:c8:3c:36 Note-9
Aug 24 18:35:25 raspberrypi hostapd: wlan0: STA b4:e6:2d:75:e5:64 IEEE 802.11: associated
Aug 24 18:35:25 raspberrypi hostapd: wlan0: STA b4:e6:2d:75:e5:64 RADIUS: starting accounting session 21B444AA338170B6
Aug 24 18:35:25 raspberrypi hostapd: wlan0: STA b4:e6:2d:75:e5:64 WPA: pairwise key handshake completed (RSN)
Aug 24 18:35:26 raspberrypi dnsmasq-dhcp[7156]: DHCPDISCOVER(wlan0) b4:e6:2d:75:e5:64
Aug 24 18:35:26 raspberrypi dnsmasq-dhcp[7156]: DHCPOFFER(wlan0) 10.42.42.26 b4:e6:2d:75:e5:64
Aug 24 18:35:26 raspberrypi dnsmasq-dhcp[7156]: DHCPREQUEST(wlan0) 10.42.42.26 b4:e6:2d:75:e5:64
Aug 24 18:35:26 raspberrypi dnsmasq-dhcp[7156]: DHCPACK(wlan0) 10.42.42.26 b4:e6:2d:75:e5:64 ESP_75E564

Is this because of the new firmware blocking tuya-convert?

HeyJonny, 2019-08-29:
A friend of mine bourght some plugs: It worked. At my side, I have the same issue: LED goes off aaaaand nothing ... 
Does anyone know how to find out, if it's the new version?
If it is the new version: Is there anyway to get the script running again or totally no hope?

dude2k, 2019-09-14:
Same here, I bought Gosund SP1 on Amazon last week. LED goes off and then nothing happens.

sirvictory444, 2019-09-17:
I bought 8 of these about 12 months. Ago.  September 2018.  I was able to get 4/8 working.  Followed the exact same procedure for every one.    I recently purchased 8 more and was not able to get any working.  From reading comments here and elsewhere it seems the newer versions are not working, probably due to a newer firmware?

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc