ct-Open-Source/tuya-convert #223: SSL could fix?

vergoossen, 2019-06-28:
I mentioned when you go to http://tuya.com you came into the webserver, but when you go to https://tuya.com it wont work.
I tried to edit the code to listen to port 443 and create self signed certificate, but at the end i got a python error like ssl.sslerror: [ssl] pem lib

Maybe someone else can try or help, maybe this could be the fix? Or am i thinking stupid?

wlatic, 2019-06-28:
Vergoossen,

Could you provide the full error you got?

In a past life I was a  web tech and can help see what the errors are and whats happening.

vergoossen, 2019-06-28:
I did something wrong, key en crt didn't match.
Now im getting:
SSL Error on 10 ('10.42.42.15', 26197): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852).

Maybe I need some CA bundle?

wlatic, 2019-06-28:
Check the ciphers in your config:

ssl_prefer_server_ciphers on;
ssl_ciphers ECDH+AESGCM:ECDH+AES256:ECDH+AES128:DH+3DES:!ADH:!AECDH:!MD5;

vergoossen, 2019-06-28:
In which config?

vergoossen, 2019-06-28:
In the config where i also declare my certs i think?
That doesnt work for now.

Im also curious how i can see which requests are made?

wlatic, 2019-06-28:
You could use: ssl_context.set_ciphers("ECDH+AESGCM:ECDH+AES256:ECDH+AES128:DH+3DES:!ADH:!AECDH:!MD5")
above your certificate addition line in 
ssl_context.load_cert_chain(certfile=certfile, keyfile=keyfile)

I'm guessing you made the changes to fake-registration-server.py

Its still unlikely this will work as the device itself will have to accept the certificate, which with it being self signed will provide a warning and drop the connection.

vergoossen, 2019-06-28:
Yes you're right, on my phone it gave a warning and accept it, but ofcourse the device wont accept te warning.
:(

kueblc, 2019-07-02:
Thanks for trying, it is a bit more involved as you have found. Closing this issue but feel free to continue using it for discussion.

closed 2019-07-02 by kueblc

madjam002, 2019-07-27:
@kueblc Are there any ongoing efforts to try and find ways around this in the latest firmware?

I’d like to help out but I am wondering if there is a Github issue/conversation somewhere where different methods have been attempted so we can all avoid duplicating effort.

Thanks