ct-Open-Source/tuya-convert #227: Scrolling text then locked up after connecting to vtrust wifi

qeggs, 2019-07-06:
As soon as I connect my smartphone or laptop to the vtrust wifi network, many lines of text scroll quickly by on the raspberry pi over 1-2 seconds. The last line of code says "...end trace...". At this point, the command prompt doesn't return, and the raspberry pi is unresponsively locked up.

Installed tuya convert on a clean copy of Raspbian Buster Lite. Had to apt-get update an and install git to proceed with tuya convert, but no other issues. Attempting to convert a MOES Wifi Dimmer Switch, but haven't gotten far enough to place it into pairing mode.

At least one video walkthrough of tuya convert indicates that scrolling lines of text should not appear after connecting my other devide to the vtrust SSID.

jpcanovas, 2019-07-06:
I was in the same situation with a Raspbian Buster Lite. Do `sudo apt install python-tornado`. After this, that error line won't appear (you will have a Listen to port 80 line). But for me and formmy new Teckin SP22 wasn't enough and I couldn't flash the plug. (I think this new plug includes the patch that blocks tuya-convert)

kueblc, 2019-07-22:
If this has been resolved please close the issue, otherwise please let us know what we can help with

qeggs, 2019-07-22:
I will try the solution above today or tomorrow and report back. Thanks.

closed 2019-09-28 by kueblc