ct-Open-Source/tuya-convert #228: WiFi not appearing ./start_flash.sh: line 4: screen: command not found

Klagio, 2019-07-08:
Installed on a pi3,  after launching the script there is no WiFi  created

pi@pioffice:~/tuya-convert $ ./start_flash.sh
./start_flash.sh: line 4: screen: command not found
./start_flash.sh: line 5: [: -gt: unary operator expected
./start_flash.sh: line 7: [: -eq: unary operator expected
sudo: screen: command not found
sudo: screen: command not found
sudo: screen: command not found
sudo: screen: command not found
~/tuya-convert/scripts ~/tuya-convert
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

Klagio, 2019-07-08:
```
pi@pioffice:~ $ lsusb
Bus 001 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

kueblc, 2019-07-08:
You need to run `install_prereq.sh` first

Klagio, 2019-07-09:
I did but diod not install all pre requisites, had to install some manually

kueblc, 2019-07-09:
Then this is where we need to debug. You should not need to install anything manually. Can you please provide details on what went wrong when installing prereqs?

closed 2019-07-22 by kueblc