ct-Open-Source/tuya-convert #229: How to check if power plug has new firmware (not possible to use tuya-convert) or old one?

Klagio, 2019-07-09:
I just bought a power plug, and tried to use tuya-convert.

When I launch 

`./start_flash.sh`

everything seems fine, exept that the dots are continuing and continuing, doing nothing. Does it means that the plug is shipped with the new firmware? I did not use the Smart Life app for fear of upgrading it automatically (the instruictions says to use Smart Life apop, so I think its a Tuya power plug)

Below the logs, let mne know if I need to give more logs

smarthack web log
```

[33m[W 190708 13:43:22 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 6.55ms
[33m[W 190708 13:43:24 web:2246][m 404 GET /api/333/config/wifi/ (127.0.0.1) 6.34ms
[33m[W 190708 13:43:24 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 8.10ms
[33m[W 190708 13:43:24 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 6.34ms
[33m[W 190708 13:43:24 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 6.40ms
[33m[W 190708 13:43:24 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 6.73ms
[33m[W 190708 13:43:24 web:2246][m 404 GET /api/1CFA908992/groups?_=1562573481149 (192.168.1.3) 6.95ms
[33m[W 190708 13:43:26 web:2246][m 404 GET /api/333/config/wifi/ (127.0.0.1) 6.70ms
[33m[W 190708 13:43:26 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 6.95ms
[33m[W 190708 13:43:26 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 6.86ms
[33m[W 190708 13:43:26 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 9.55ms
[33m[W 190708 13:43:26 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 9.73ms
[33m[W 190708 13:43:28 web:2246][m 404 GET /api/333/config/wifi/ (127.0.0.1) 8.21ms
[33m[W 190708 13:43:28 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 15.11ms
[33m[W 190708 13:43:28 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 12.47ms
[33m[W 190708 13:43:28 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 13.28ms
[33m[W 190708 13:43:28 web:2246][m 404 PUT /api/333/config/wifi/updated (127.0.0.1) 13.63ms

```
wifi log
```

wlan0: STA 68:57:2d:xxxx RADIUS: starting accounting session 65FF6273EA365263
wlan0: STA 68:57:2d:xxxx WPA: pairwise key handshake completed (RSN)
wlan0: STA 68:57:2d:xxxx IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 68:57:2d:25:96:dd
wlan0: STA 68:57:2d:xxxx IEEE 802.11: disassociated
wlan0: STA 68:57:2d:xxxx IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 68:57:2dxxxx

```

kueblc, 2019-07-09:
Please when opening a new issue use the template. What is the device you are trying to flash?

This does not look at all like a Tuya device from the web log, but it's strange that it would even connect at all if it wasn't.

Klagio, 2019-07-09:
Thatt's the device. On the instructions it says to use the app SMART LIFE, isn't the one of Tuya based products?

https://www.amazon.it/gp/product/B078BMQ8Q5/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

kueblc, 2019-07-09:
Can you capture network traffic using `tcpdump` or Wireshark while you attempt to flash?

Klagio, 2019-07-09:
I never did that, I can try

kueblc, 2019-07-09:
Do you have Phillips Hue related equipment? The API calls in your web log look like they are ZigBee related.

Klagio, 2019-07-09:
I have a Conbee stick on the same pi3. Shall I disconnect it?

kueblc, 2019-07-09:
Ahh yes, this is likely where the API calls are coming from.

Klagio, 2019-07-09:
No, I think is same

```

[33m[W 190709 18:42:15 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562676916711 (192.168.1.3) 5.40ms
[33m[W 190709 18:42:15 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 5.83ms
[33m[W 190709 18:42:16 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.62ms
[33m[W 190709 18:42:16 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.36ms
[33m[W 190709 18:42:16 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.36ms
[33m[W 190709 18:42:16 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.18ms
[33m[W 190709 18:42:17 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 5.81ms
[33m[W 190709 18:42:17 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.22ms
[33m[W 190709 18:42:18 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.53ms
[33m[W 190709 18:42:18 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.18ms
[33m[W 190709 18:42:18 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 8.07ms
[33m[W 190709 18:42:19 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562651273782 (192.168.1.3) 5.50ms
[33m[W 190709 18:42:19 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 5.49ms
[33m[W 190709 18:42:19 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.43ms
[33m[W 190709 18:42:19 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.16ms
[33m[W 190709 18:42:20 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.13ms
[33m[W 190709 18:42:20 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.66ms
[33m[W 190709 18:42:20 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562676916712 (192.168.1.3) 6.63ms
[33m[W 190709 18:42:21 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 6.28ms
[33m[W 190709 18:42:21 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.29ms
[33m[W 190709 18:42:21 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.59ms
[33m[W 190709 18:42:21 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.20ms
[33m[W 190709 18:42:21 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 7.40ms
[33m[W 190709 18:42:23 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 5.57ms
[33m[W 190709 18:42:23 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.18ms
[33m[W 190709 18:42:23 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.24ms
[33m[W 190709 18:42:23 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.28ms
[33m[W 190709 18:42:23 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.45ms
[33m[W 190709 18:42:24 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562651273783 (192.168.1.3) 5.73ms
[33m[W 190709 18:42:25 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562676916713 (192.168.1.3) 7.37ms
[33m[W 190709 18:42:26 web:2246][m 404 PUT /api/324/config/wifi/scanresult (127.0.0.1) 6.43ms
[33m[W 190709 18:42:27 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 5.53ms
[33m[W 190709 18:42:27 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.59ms
[33m[W 190709 18:42:27 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.13ms
[33m[W 190709 18:42:27 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.74ms
[33m[W 190709 18:42:27 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.50ms
[33m[W 190709 18:42:28 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 5.54ms
[33m[W 190709 18:42:29 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.24ms
[33m[W 190709 18:42:29 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562651273784 (192.168.1.3) 7.75ms
[33m[W 190709 18:42:29 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.61ms
[33m[W 190709 18:42:29 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.44ms
[33m[W 190709 18:42:29 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.23ms
[33m[W 190709 18:42:30 web:2246][m 404 GET /api/BDF58C3FEB/groups?_=1562676916714 (192.168.1.3) 5.26ms
[33m[W 190709 18:42:30 web:2246][m 404 GET /api/324/config/wifi/ (127.0.0.1) 6.04ms
[33m[W 190709 18:42:31 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.33ms
[33m[W 190709 18:42:31 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.55ms
[33m[W 190709 18:42:31 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.10ms
[33m[W 190709 18:42:31 web:2246][m 404 PUT /api/324/config/wifi/updated (127.0.0.1) 6.21ms

```

Klagio, 2019-07-09:
Please notet that 192.168.1.3 is the IP address of my PC, were I launched  a SSH connection to the pi3 that is having the tuya convert script and I do not recognize the mac (is it a mac?) BDF58C3FEB

3m[W 190709 18:42:30 web:2246]�[m� 404 GET /api/BDF58C3FEB/groups?_=1562676916714 (192.168.1.3) 5.26ms

kueblc, 2019-07-09:
That's too short to be MAC address.

kueblc, 2019-07-22:
Closing due to inactivity, please reopen if you have a network capture to share.

closed 2019-07-22 by kueblc