ct-Open-Source/tuya-convert #230: Flashing tuya zigbee Gateway

RomSand, 2019-07-10:
Hi,
 I was trying do dump their gateway flash to be able to root it at accedently erased it, maybe someone has this type of device and can dump his flash it's simple 16Mb SPI flash chip, reset it before dumping thanks.

kueblc, 2019-07-10:
Hi @RomSand 

Could you please clarify? It's not clear what you did and what you're asking for. Is this a tuya-convert issue?

kueblc, 2019-07-10:
If I understand correctly, you did not use tuya-convert (because it does not simply erase your flash) in which case this is not the right place to ask. This issue tracker is for issues with tuya-convert only.

I'm also not sure what you mean by "root" your Tuya device as it's not running *nix.

Lastly I should warn you that if you try using stock firmware from another device it will not work, as each device is assigned a unique auzKey.

RomSand, 2019-07-10:
I have Tuya smart gateway TYGWZ-01 it's based on ralink chip and runing Linux,
I tried to dump SPI flash that soldered on the board but accedently erased it. need a good dump so I can recover my device and continue my work... Thanks. 
I think there is account information in the gateway that's how it authenticates on the server that's why after receting the device it suposed to be erased... 
Thanks.

kueblc, 2019-07-10:
This issue tracker is for `tuya-convert` issues **only** so I kindly request that you move this to a more appropriate forum. I will however repeat my warning for emphasis, you cannot use the stock firmware from another device as each device is programmed with a unique `auzKey` at the factory, and thus you will not be able to authenticate.

closed 2019-07-10 by kueblc