ct-Open-Source/tuya-convert #233: Could this be made to work with ArchLinux?

markg85, 2019-07-14:
Hi,

I'm trying to give this a spin om some tuya switch, but i'm having trouble getting the scripts to work.

The current setup seems very much tailored towards Debian (or Ubuntu) which makes it a bit cumbersome to use on a different distribution.

Also, it blindly expects networkmanager (which i don't use) and is changing actual settings in them. It does make backup files, but you should **not** do that! If you do, you should advise to run this on a live image or perhaps a docker container to not screw up host configuration.

Cheers,
Mark

kueblc, 2019-07-14:
> Also, it blindly expects networkmanager (which i don't use) and is changing actual settings in them. It does make backup files, but you should **not** do that! If you do, you should advise to run this on a live image or perhaps a docker container to not screw up host configuration.

It is already advised to use on a blank Raspberry Pi for this very reason.

You can try the `create-ap` fork, which is what I use on Ubuntu.

`git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

kueblc, 2019-07-22:
Closing as it has been addressed. Feel free to open a new issue if you have any other concerns. You may also want to look at #172 where @yoarch discusses `tuya-convert` on Arch.

closed 2019-07-22 by kueblc