ct-Open-Source/tuya-convert #236: Gorund Sp1

Aruscha, 2019-07-18:
Hello,

I bought 4 Gosund Sp1 sockets on Prime Day, wanted to flash them today with Tuya-convert, unfortunately only the points appear in my case and there is no flash process, I left the whole time 20min but more than a thousand points did not happen what could be the reason?

Chaotic, 2019-07-18:
> As of January 28th, 2019, Tuya has started distributing a patch that prevents tuya-convert from completing successfully. It is up to the individual brands to adopt the patch, so some devices may be affected sooner than others. To ensure the best chance of success, do not connect your device with the official app as it may automatically update the device, preventing you from flashing with tuya-convert. Some devices are already being shipped with the update, in which case there is unfortunately no work around available at this time.

I'm in the same boat with the same plugs (through I got them before prime day.)

AS137430, 2019-07-19:
Your next option is local control through https://github.com/codetheweb/tuyapi or open up the device and flash Tasmota.

ranger81, 2019-07-19:
Same issue for me. I assume we need to manually flash them by soldering. I don't expect a solution very fast, if the https topic is the problem?!

kueblc, 2019-07-22:
Yes, you can use `tuyapi` to control them locally or open and solder to flash them manually. Closing as the question has been addressed.

> I left the whole time 20min but more than a thousand points

FYI save yourself some time, flashing takes less than a minute, leaving it run longer won't make a difference.

closed 2019-07-22 by kueblc