ct-Open-Source/tuya-convert #239: NX-SM200 gets wrong IP - 10.42.42.16

Osmodia666, 2019-07-21:
Hi, 
I commented the issue #67 https://github.com/ct-Open-Source/tuya-convert/issues/67,
with the problem, that the device doesn't get the specified IP address 10.42.42.42, it gets 10.42.42.16. 
Also, I did not connect a phone to the vtrust wifi, because if I do, the NX-SM200 can't connect. It only connects if I dont connect a android phone.

I edited the start_flash.sh and replaced every 10.42.42.42 with 10.42.42.16, but I get an error connection refused, so it doesn't work.

kueblc, 2019-07-22:
If you did not get the address 10.42.42.42 that means the device did not flash, it is still running the Tuya firmware. Don't edit the IP in the scripts.

kueblc, 2019-07-22:
Please follow the template when opening a new issue, we'll need to see your logs to help find the issue. Without looking the most likely is that you are running an incompatible firmware that can't be flashed OTA.

enkol, 2019-07-29:
For me tuya-convert worked perfectly with two OxaOxe NX-SM200 plugs today.

kueblc, 2019-07-29:
Great, thanks for confirming @enkol!

closed 2019-07-29 by kueblc