ct-Open-Source/tuya-convert #240: Unable to connect to vtrust-flash AP

langestefan, 2019-07-22:
Hi, I have been unsuccesful in connecting to the vtrust-flash AP after I run start_flash.sh. When connecting to the ap both my phone and my laptop report that there is an authentication problem and refuse to connect. 

My setup:
- Windows 10 PC running Ubuntu 19.04 in a VM
- My wifi dongle: [rtl5370](https://www.amazon.de/gp/product/B015TCA2EM/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
- My smart plug: [Gosund ](https://www.amazon.de/gp/product/B07B911Y6V/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) 

Smarthack-wifi.log:

> Backing up NetworkManager.cfg...
> Restarting NetworkManager...
> Backing up /etc/dnsmasq.conf...
> Writing dnsmasq config file...
> Creating new /etc/dnsmasq.conf...
> Writing hostapd config file...
> Configuring AP interface...
> Applying iptables rules...
> Starting DNSMASQ server...
> Starting AP on wlp0s12u2 in screen terminal...
> Configuration file: /etc/hostapd/hostapd.conf
> Using interface wlp0s12u2 with hwaddr d6:57:35:c8:89:d3 and ssid "vtrust-flash"
> wlp0s12u2: interface state UNINITIALIZED->ENABLED
> wlp0s12u2: AP-ENABLED 

What am I missing?

kueblc, 2019-07-22:
I don't have a solution, but just to confirm, you've double checked the password?

langestefan, 2019-07-22:
> I don't have a solution, but just to confirm, you've double checked the password?

Yes I have. I couldn't get it to work, so I switched over to a RPI. I managed to get an IP on my laptop while connected to the AP but my phone refused the connection again. 

However, now I'm stuck again. I can see in the logs that the Gosund connects to the AP and that the blue LED stops flashing. I can also see in the logs that "SmartConfig has completed". However, it still refuses to continue whatever it's trying to do so I only see more dots appearing in the terminal. Not sure what to do next...

kueblc, 2019-07-22:
Unfortunately this device is not compatible with `tuya-convert`. You could establish local control via `tuyapi` or try flashing the device manually, by opening the device and connecting over serial.

closed 2019-07-22 by kueblc

langestefan, 2019-07-22:
Interesting. I thought they were all the same?

kueblc, 2019-07-22:
From the readme

> As of January 28th, 2019, Tuya has started distributing a patch that prevents tuya-convert from completing successfully. It is up to the individual brands to adopt the patch, so some devices may be affected sooner than others. To ensure the best chance of success, do not connect your device with the official app as it may automatically update the device, preventing you from flashing with tuya-convert. Some devices are already being shipped with the update, in which case there is unfortunately no work around available at this time.

langestefan, 2019-07-24:
Okay. Guess that's the end of tuya-convert for me then. Back to good old soldering.

minax007, 2022-04-17:
Dear all, 

@Raumzeit shard following information, which has resolved my connection issue with vtrust-flash:

https://github.com/ct-Open-Source/tuya-convert/issues/892#issuecomment-1100248239