ct-Open-Source/tuya-convert #241: Broken HostAPD on Raspbian/Pi3 may cause wifi disconnect issues

phormix, 2019-07-23:
FYI,

This is NOT a bug with tuya-convert but more an issue with current versions of Raspbian and hostapd which prevent it from work as expected

It took a fair bit of Googling the underlying errors to find a solution., so I figured I'd post this up as a PSA in case people start throwing you complaints about the drops. I'm got an earlier model Pi3, so I'm not sure if this issue exists for other Pi versions.

Symptom:
  Hostapd will allow connection of the initial non-tuya device (i.e. your phone, tablet, etc) via the vtrust-flash network, however as soon as any other device (including that you are trying to flash) connects, the initial device is kicked out and hostapd will reject subsequent connections. The flash process thankfully doesn't start for me so at least it doesn't appear to be leading to bricked devices.

The steps outlined here have addressed the issue for me: 
  https://github.com/raspberrypi/firmware/issues/1117

kueblc, 2019-10-04:
Added to the [Troubleshooting wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting). Thanks for the contribution!

closed 2019-10-04 by kueblc

meingraham, 2019-10-04:
Colin,

I know that there were exchanges regarding the requirement to run Buster in order for the process to work when using an RPi3. The hostapd fix requires downgrading the RPi firmware itself (not the OS). Will this introduce conflicts between the newer OS vs older firmware?

Mike

kueblc, 2019-10-04:
I'm not familiar with the RPi environment, I personally use `tuya-convert` on my Ubuntu desktop. If anyone has an answer please feel free to update the wiki.

phormix, 2019-10-04:
I haven't run into any issues yet. But the only thing I use that particular
pi for is flashing

meingraham, 2019-10-04:
I ended up not needing to go this route. I decided to start from scratch and re-flashed my SD card with Buster and proceeded to clone Tuya-Convert and go through the process. I was able to flash a Gosund KS-602S that would not flash with Tuya-Convert v1. I'm now running the latest Tasmota (6.6.0.15) with the template for this device (it's in the template repo).

{"NAME":"Gosund KS-602S","GPIO":[17,56,0,255,0,0,0,0,0,0,21,0,158],"FLAG":0,"BASE":18}