ct-Open-Source/tuya-convert #242: Not possible to find my gosund sp111

MiguelESP89, 2019-07-24:
hey, 
I'm lost :D 
i want to flash my new gosund sp111. i installed all the things. my mobile phone is connceted to the wifi, the gosund is in pairing mode but my pi didn't find him. 

I have attached my logfiles. 

Thank you for your help

Miguel
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3428048/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3428049/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3428050/smarthack-web.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3428051/smarthack-smartconfig.log)

kueblc, 2019-07-24:
Duplicate of #216, unfortunately not compatible with `tuya-convert` at this time

closed 2019-07-24 by kueblc

ibeerens, 2019-09-05:
When will it be supported?