ct-Open-Source/tuya-convert #244: ASZKJ Smart Wifi RGBW bulb information

EverythingSmartHome, 2019-07-31:
Hi,

Sorry not sure if I am submitting this in the right place, just thought I'd drop some information on a bulb from ASZKJ, available on Amazon.co.uk.

This bulb came already on the latest firmware, unfortunately.

Flashing with esphome/tasmota is possible and pretty easy, its extremely easy to dismantle and all pins are located on the top side.

After some trial and error, I managed to get the GPIO's as follows:
Red: GPIO14
Green: GPIO12
Blue:GPIO13
White: GPIO4

I have a full flash dump I took of stock firmware if you would like me to upload? Please let me know.

Hope this helps someone with this bulb.

kueblc, 2019-07-31:
Hi @EverythingSmartHome,

Thanks for the report! Yes, firmware dumps are always appreciated.

EverythingSmartHome, 2019-08-14:
Sorry for the delay, please find the attached!
[ASZKJ firmware.zip](https://github.com/ct-Open-Source/tuya-convert/files/3501146/ASZKJ.firmware.zip)

kueblc, 2019-09-30:
Thanks again for your contributions!

For future visitors, we have TLS firmware support in the works #279.

closed 2019-09-30 by kueblc